jgyates/genmon #578: No issues, just thanks

jcleek, 2021-10-01:
So github offers no means to say thanks so I am creating an "issue" to tell my story. I have a pre-2008 17kw generator that I have been using the Verizon monitor to manage. I have had that for many years but last month I was notified they are turning the service off. I contacted Generac to see which of their new monitors would work and their answer was none. Great way to support customers. So I started looking and stumbled upon genmon. Being an IT guy and having 5 other PIs in the house doing various functions this was a project I could easily tackle. Once I saw the PI HAT from PintSize.Me it really was a no-brainer. The install was simple and once I was pointed in the right direction for advanced settings (who knew you could read the manual and get answers?) my generator started reporting much more information than the paid for monitor with its monthly service charge ever told me.

Thank you @jgyates  for your work and contribution to the community.

jgyates, 2021-10-01:
@jcleek 

Thanks for the kind feedback! I am glad everything is working well for you.

dave-parsons, 2021-11-07:
Useful, flexible, stable, free.   Love it.  Thank you.

closed 2021-12-06 by jgyates