jgyates/genmon #579: Did something change?

davehawthorneGitHub, 2021-10-01:
### Expected Behavior

Run log shows current data and graph shows green square when generator is switched states. I have just been working on getting this gen installed and the Pi / pintsize.me hat (and serial cable) so I don;t know if this is something I did or something that changed in the s/w (I think there was an update a couple weeks back?). 

**Also, I second the thanks on this s/w from one of the other posters...it is awesome!**

[image]

### Actual Behavior

Run log is out of date order and nothing is being added to calendar after Sept 12th
[image]

I have ran the generator in various states (auto-started with GenMon, manual start, and exercise) today with no log entries.

### Steps to Reproduce (including precondition)

Tried re-installing s/w on a new SD card, but other than installing an older backup (taken when I first installed GenMon, no other steps done.

### Screenshot or Pictures relating to the problem (if possible)

Added screen shots above

### Your Environment

- Generator Model: Generac 20KW Pre Nexus (2008)
- Genmon Logs: Submited
- Genmon Version: 1.17.02

jgyates, 2021-10-01:
Nothing has changed in the log parsing for quite some time.  Can you check the log entries on your controller and make sure they are the same or different? The controller will put log entries in the order they occur, not by date so if your date is off this can happen. Also, genmon does not sort them by date (or any value), it just displays them in the order the controller gives them.

davehawthorneGitHub, 2021-10-01:
Thanks, interesting.
Yes, it's in that order in the controller also. I have been trying to reset the maintenance and it just seems to ignore some things. The date is correct in the controller (and GenMon), maybe I have a controller issue. Will look into that more.

closed 2021-10-01 by davehawthorneGitHub

jgyates, 2024-10-04:
@davehawthorneGitHub, do you see anything like the behavior described in this thread: https://github.com/jgyates/genmon/issues/1191