jgyates/genmon #580: Custom Generator Controller such as Deep Sea Controller 7320MKII

Ntampata, 2021-10-04:
Hi jgyates, I am new on modbus staffs but there is a custom_controller.py file on the director genmonlib and comes the idea to add a DSE controller 7320 family file, like the existing files such as: _Generac_HPanel.py, Generac_evolution.py, etc_... 

My 1st question is: how can I start to setup the modbus registers for DSE 7320 controllers? 

As an example for read only registers (16 bits): the Battery Voltage is located at address 0x405. 

Assuming that,  slave id     : 0x01
                                 Fc                :0x03
                                 Start adrr: 0x405
                                  Num regs: 1 

Thanks for your attention.

jgyates, 2021-10-04:
Hi @Ntampata 

The custome_controller.py method is intended to be used with a JSON formatted file that describes your registers. This page has more info:

https://github.com/jgyates/genmon/wiki/Appendix-N-Genmon-Supporting-Other-Controller-Types

Also, here is an example of a working JSON formatted file.

https://github.com/jgyates/genmon/blob/master/data/controller/evo_lc.json

I am not sure what Fc 0x03 is in your above question. 

My suggestion would be to do this:

1) Make a copy of the file /data/controller/evo_lc.json, renaming it to something like "deapsea.json" and save it in the same directory on the genmon system.

2) modify the file /data/controller/deapsea.json replacing occurances of "000a" with "0405". "000a" is the battery voltage register on the Evolution controller so changing this to another 16 bit register with the address 0405 should work. We can change the decimal places if necessary in the JSON file.

3) Start genmon and go to the advanced page (double click the gear icon in the web interface

4) on the advanced page:
      set the slave ID ("Modbus slave address") to your slave ID
      select the file name you renamed above in the "Custom Controller Config File"
      set "Controller Type" to custom

5) Save your settings at the bottom of the advanced page, this will restart genmon

Assuming your data rate of your serial link is 9600,N,8,1 then it should read that data over the serial port. Wait about 15 seconds  and then look on the registers page (single click the gear icon) and see if you get any value for register 0x0405

Let me know if you have any questions.

Also, if we get this working would you be willing to add your custom JSON file to the repository?

Ntampata, 2021-10-04:
Hi jgyates, 

Thank you for your quick response 👍. 

Yes, I was thinking on make or copy and edit a Json file for this custom controller.

I will follow your recommendations,  and I will let you know.

 It would be grateful to contribute to this project, yes I will share it to the repository.

FC means: Function Code.

jgyates, 2021-10-04:
Also, I forgot to mention. There is a python program that may be useful that I put in the repository. The program /genmon/OtherApps/modbusdump.py can be used to read modbus registers over serial or serial over TCP.

For command line arguments, from the /genmon/OtherApps directory type:

      sudo python modbusdump.py -?

jgyates, 2023-07-12:
I have added the ability to allow custom controllers to be used with the SNMP add on. More info is [here](https://github.com/jgyates/genmon/wiki/Appendix-N-Genmon-Supporting-Other-Controller-Types#snmp-addon-program). Let me know if anyone needs any assistance in creating the JSON file to support SNMP your controller.