jgyates/genmon #581: Replacing working Genmon SD Card with new Genmon SD Card

eBoon123, 2021-10-05:
I have a currently working install of Genmon, but I am experiencing a loss of communications every 3 - 4 days that requires a reboot where I need to go outside and disconnect/reconnect the power to my Pi. Everything comes back up and works fine after the reboot. I wanted to rule out problems with the SD Card, so I created a brand new install of Genmon (following the step-by-step instructions in the Wiki) on a new SD Card. For some reason this install of Genmon does not work on my existing hardware setup - the serial connection does not seem to work. I have tried using both the default config and the config from my working SD Card, but in both cases, serial communication does not work from my new instance of Genmon on the new SD Card. When I simply swap in my old SD Card, everything works again. I think I am missing something very basic, but not sure what it might be. I have verified all the  Any ideas on what I might be doing wrong?

pi@raspberrypi:~/genmon/OtherApps $ sudo python serialconfig.py -c

Checking : BT service disabled: OK
Checking : Serial console command line removed: OK
Checking : Is enable UART in boot config: OK
Checking : Is BT overlay disabled: OK
Checking : Serial console service disabled: OK

Serial port settings are OK. A reboot is needed if changes were made.

pi@raspberrypi:~/genmon/OtherApps $ ls -l /dev/serial*
lrwxrwxrwx 1 root root 7 Oct  5 09:17 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root 5 Oct  5 09:17 /dev/serial1 -> ttyS0

Thank you!

skipfire, 2021-10-05:
Depending on how you installed I'd start with checking wires to make sure nothing came loose when you were moving the Pi around to get the SD card in and out.  Serial lines and ground.

eBoon123, 2021-10-05:
> You might try manually setting the serial configurations in raspi-config. Did you copy config files from your old install, or manually re-enter the configs?

The only thing I did in raspi-config was set new password and locale information. I tried using the default genmon config and restoring the config from my working SD Card - neither worked.

skipfire, 2021-10-05:
Have you done a loopback test?  It's a few steps to do it, https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting.

eBoon123, 2021-10-05:
> Have you done a loopback test? It's a few steps to do it, https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting.

I have not (mainly because I'm too lazy to disconnect everything and bring it in the house from where it is mounted). But clearly, from a hardware point of view, everything is working. Do you think a loop-back test would be necessary when everything works perfectly with the original SD Card?

skipfire, 2021-10-05:
If the hardware is fine, it must be configuration.  If the loopback test works then it shows the Pi's serial configuration is actually enabled and working.

If you do put the old SD card in again, try running `ls -l /dev/serial*` again to ensure you didn't have some different serial mapping on the previous install.