jgyates/genmon #591: Random invalid notifications regarding Transfer switch status

in2hockey04, 2021-10-19:
The genmon software has been randomly sending me invalid notifications regarding the transfer switch status.
This has been happening for about the last 3 months. 
See an example that just happened today below:

FRM:Genmon 
SUBJ:Transfer Switch Changed State Notice at 
MSG:Time: 02:34:29 PM
Date: Tue 19-Oct-2021

Power is being supplied by the generator.

FRM:Genmon 
SUBJ:Transfer Switch Changed State Notice at 
MSG:Time: 02:34:31 PM
Date: Tue 19-Oct-2021

Power is being supplied by the utility line.

- Generator Model: RG04524ANAX 
- Serial # 000000003004333374
- Genmon Logs: Submit via the About page: Submitted about 1500 PST
- Genmon Version: V1.17.04 (Was also happening with the previous 2 versions)

in2hockey04, 2021-10-20:
> Another thing to try is to increase the modbus timeout on the advanced page. I use serial over TCP (over Wifi) and have my modbus timeout set to 10 seconds. The default for direct serial is 3.

Is this where I would increase the timeout? If yes, it's interesting that it's set to 0 and not the default setting of 3. I've never changed that.
Do you recommend starting with 3 then or changing it right to 10?
Thanks!
Eddie

jgyates, 2021-10-20:
There is a minimum timeout of 3 that you can not change. This setting adds to that value.