jgyates/genmon #592: Generator shows as outputting 25kw during exercise

TopBanana123, 2021-10-31:
Not sure if I'm just not understanding the graphs, but when my generator exercises or if I just turn it on manually, it shows as outputting 25kw, which is the max output of the generator. Its putting out 0kw in reality, as there is zero load

Example here when it exercised 

[image]

jgyates, 2021-10-31:
How the software gets the values for power is dependant on the model. Some models have current transformers that the output is exposed over modbus. Other models have a hall effect sensor which is output over modbus. some models do not have either and you should disable this value. On some industrial models the controller outputs the power value that is calculated by the firmware.

This link has some know issues and info on the power reporting (see number 5 in the link)

https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues

This link has some background on the power usage and fuel monitoring that may be useful:

https://github.com/jgyates/genmon/wiki/Appendix-H-Monitoring-Fuel-and-Power-Usage

Also, all issues dealing with the display of output power have the tag "Power Display" so you can also search the genmon forum for closed issues with this label to get more details on this question as well.

Searching your past posts it looks like you have a Generac RG027. This model uses a hall effect sensor. The value read over modbus for evolution liquid cooled models is assumed to be some internal value that corresponds to power. The genmon software is probably the least accurate on evolution liquid cooled models due to what we have to work with over modbus for this controller.  The value read over modbus for liquid cooled models has been observed to change as the power changes but the output is not the actual power, but it is not the actual power output. You can either disable this feature as described in the link above or you can try to calibrate the way the software interprets this value (described in the link below). This thread describes the way to calibrate the softwares interpretation of the value read from the controller.

https://github.com/jgyates/genmon/issues/563

The numbers used in the "default" calibration are from a evolution liquid cooled diesel model (that is my generator).

the short answer is that getting the power from the controller varies depending on the model and in some models it is not possible or requires some calibration to get it to a workable situation. 

From the softwares perspective the goal of this feature was to try to monitor fuel usage, so power output was needed to calculate fuel consumption. When exercising, there is no load, however fuel is consumed so the software does not zero this value when exercising. Depending on the method the controller uses (CTs, hall effect sensor, or actual power reported over modbus) the results can vary quite a bit regarding the output power reported when exercising.

I hope this helps, let me know if you have other questions.

jgyates, 2021-11-06:
I am going to close this issue but feel free to post to this thread if you have any other questions or feedback.

closed 2021-11-06 by jgyates