jgyates/genmon #593: ATS fail detection?

skipfire, 2021-11-01:
Does the generator know if the transfer fails?  My ATS isn't transferring and the generator did not error and GenMon didn't show anything.  Any idea how difficult it would be to have GenMon notice that the transfer switch should be engaged but no significant power output has occurred after 5 minutes and then fire its own alert?

Because I know the helpful people here will also want to see if they can help me fix the ATS, I am pretty sure it is the electromagnet as I hear the relay click when it should, but no buzz or clank.  It successfully transferred last week for an outage, but didn't during my scheduled transfer test today.  Tomorrow I'm going to take the panels off to take a look and do a manual transfer test.

jgyates, 2021-11-01:
For the "dumb" type ATS (the ones typically used with Evolution and Nexus), my understanding is, no, there is nothing that tells the controller that the switch has a problem. I think the sequence of events is this:

1) Controller detects outage and starts engine if in auto mode, it waits for the warmup time (default is 5 seconds)
2) controller signals ATS via a relay 
3) ATS switch engages 

The controller knows that it has signaled the relay but does not know if the actual ATS is engaged.

@liltux may have more details as he know a lot more about this than I do.

griffths, 2021-11-02:
There are a set of dry contacts that can be used to confirm transfer switch position but of course that would require a physical connection to the Pi or some other way to get that information to Genmon

skipfire, 2021-11-02:
@jgyates what about GenMon detecting the possible failure through data?  If it is a transfer test, but output power doesn't measure above 500w for 5 minutes, some kind of notification or warning of a possible problem?

grsthegreat, 2021-11-02:
make sure both wires are plugged into the coils. the upper coil is the utility coil, the lower coil is the generator coil. i get service calls all the time about switches not transferring, and more times than not a wire fell off. may have to physically tighten the spade connector if its loose.

skipfire, 2021-11-06:
@grsthegreat is the purple wire the one you were thinking of?  Once I opened the switch it sure stood out.

liltux, 2021-11-06:
On the right side of the new switches it is standard to have an additional auxiliary contact that will change states.  That could be taken to the Rpi as in input.  This would be the best way to monitor switch state change.  Also all newer Gens all have CTs.  The switches that do not have an Aux contact it is easy to add.

grsthegreat, 2021-11-06:
yes. the wire slipped off of the coil. slightly recrimp connector and reinstall

skipfire, 2021-11-06:
Thanks @grsthegreat, tested again and it is working again.

grsthegreat, 2021-11-06:
good to hear. make sure none of the others are loose

skipfire, 2021-11-06:
> good to hear. make sure none of the others are loose

None of the others moved with a gentle tug, but I have my semi-annual service this month so I'll ask them to double check as well.  Thanks for the help.

jgyates, 2021-11-08:
I added this to the feature request list. I will update this thread when it has been added.

jgyates, 2021-11-08:
https://github.com/jgyates/genmon/blob/master/feature_requests.md

closed 2021-11-08 by jgyates