jgyates/genmon #594: Generator clock off when GenMon power is lost

dennis48755, 2021-11-03:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Generator clock keeps correct time

### Actual Behavior

Generator clock falls behind when GenMon Raspberry Pi is abruptly powered off

### Steps to Reproduce (including precondition)

Drop power to GenMon Raspberry  Pi

### Screenshot or Pictures relating to the problem (if possible)

Hello and thank you for your work on this project.

My GenMon setup has been running flawlessly for over three and a half years. I've never paid attention to my generator clock but notice when it does it's weekly test they are always at 2:00 pm as scheduled.

The past three days I've been working on replacing my 1500VA APC UPS with a 3000VA. During this time my GenMon Raspberry Pi has been powered off a number of times both intentionally & unintentionally.

Each time I've noticed the generator clock was off significantly, possibly (not sure) about the time GenMon was powered off.

The GenMon Raspberry Pi  clock appears to be set to the correct time.

### Your Environment

GenMon Raspberry Pi is located in a panel in my basement powered by an APC UPS. Communication to the generator via three wire serial connections over about 30 feet. Never see CRC errors or timeouts.

General Settings:
- Sync Daylight Savings Time: On
- Sync Time: Off

- Generator Model: G0064622
- Genmon Logs: Submitted
- Genmon Version: 1.17.04

jgyates, 2021-11-03:
Hi @dennis48755 

At little background on the clock: There are two separate clocks, one on the pi and one on the generator controller. The software can set the time on the generator clock, but only the hours and minutes, not seconds. The pi, by default if using raspbian as the operating system, will set the time over the internet and assuming that your time zone has been set correctly the time should match your local time.

To find out which time is off you  can go to https://www.time.gov/ and get the accurate time. The log into your pi via ssh and type

      date

This should give you an output of the time and date. The time will be in seconds and it should be within a few seconds of the time from https://www.time.gov/. If these two times are very close then just enabling the "Sync Time" should fix your problem.

The generator controller's time is not set over the internet (Evolution 2.0 can do this but Evolution 1.0 can not). This just a guess as to why you are seeing this, but, assuming that it is your generator time that is off, some clock circuits in electronic equipment can drift over time if the supplied power source is not at the frequency expected (e.g. 59Hz instead of 60Hz). There are ways to compensate for this but it takes extra hardware to do so. I am guessing the Evolution 1.0 controller does not have this extra compensation circuitry. 

Let me know how it goes.

dennis48755, 2021-11-04:
I'm frustrated by not knowing how best to respond. I don't write well and don't want to make the issue more confusing.

Please reread my OP, and note the mentioning of abrupt loss of power to the GenMon Raspberry Pi by my activities.

Trying to summarize, both clocks normally very accurate. When power is lost to the Pi somehow the generator clock is being changed even though Sync Clocks is OFF.

jgyates, 2021-11-04:
I think understand what you are saying. I know of no way that the software can change the time on the generator if you do not have the "Sync Time" enabled, even if the system power is abruptly shut off. These commands are sent over modbus with a checksum that if not correct will cause the packet to rejected by the controller. Sending random or invalid data to the controller is extremely unlikely.

If you think the controller time is off then set the "Sync Time" option. 

If you want to troubleshoot this further I would need more information. You do not give any specific times, so when you say the generator time is off I do not know by how much. It is expected to be off by up to on 1 minute due to the fact that the generator does not support setting the seconds, only the hours and minutes of the time of day. Knowing how much the time is different would help put this in perspective.

The front page of the web interface shows both the time on the pi (Monitor Time) and the time on the generator (Generator Time). Providing both would be helpful. Also providing the time of day that you captured the time on the web interface from https://www.time.gov/  would allow me to validate that the time on the pi is not incorrect.

jgyates, 2021-11-06:
@dennis48755 
I am going to close this issue however if you have any additional data or questions feel free to post to this thread.

closed 2021-11-06 by jgyates

dennis48755, 2021-11-06:
I've run some tests today all with the same results, just trying to get the best screen captures. The testing confirms (to me) GenMon is setting the generator clock ONCE when it boots with Sync Time = OFF in settings.

Using the timestamps from the bottom of the GenMon Status page...
I started with the Monitor Time & Generator in sync with in a minute or two.
[image]

I then disconnected the power to the GenMon Raspberry Pi.
[image]

A little over an hour later I disconnected my router so the Raspberry could not connect with a time server. And connected it to power.

When GenMon first came up the Status page showed the Monitor time a little more than an hour behind and the Generator time correct.
[image]

A short moment later the Generator time is changed to the hour behind time of the Raspberry.
[image]

Soon after plugging in the router, the Raspberry sets its time correctly. But NOT the Generator.
[image]

The generator remains with the incorrect time indefinitely until I manually set or sync it. 

Again Sync Time is off in the settings.
[image]

Thanks
Dennis

jgyates, 2021-11-07:
Hi @dennis48755 

I took a closer look at the code and you are correct. The software does set the time once even if sync time is not set. Here is the code in question:

https://github.com/jgyates/genmon/blob/master/genmon.py#L831

It has been a long time since I wrote that I had forgotten the logic and for some strange reason I thought this was not the case but it looks like it is. Thanks for clearing that up.

To address your original question, other than enabling Sync Time, I don't have a good option to resolve this issue as it is difficult for genmon to reliably know when the time was changed by another piece of software. The change to detect daylight savings time operates on the operating system knowing when daylight savings time occurs.

There are some add ons for the pi that provide a real time clock, but that seems like overkill.

Let me know if you have any other thoughts on this.