jgyates/genmon #595: Accessory HAT

skipfire, 2021-11-05:
Soliciting feedback... 
I'm currently designing a second HAT/pHAT for a DS18B20 temp sensor plus 2 SCT-013-000 CTs.  The DS18B20 will use pin 7 (one-wire).  I have a path to change the resistance if needed with a header to short built-in and add 1 or 2 through-hole resistors.  The CTs will connect through an MCP3008 ADC to SPI1 (pins 35, 38, 40), and I'm thinking pin 37 for the CS.  Any thoughts?

jgyates, 2021-11-05:
The SCT-013-000 is likely a good choice. How difficult would it be to have the ability to change the burden resistor on the CTs? I use CTs on the openenergymonitor.org project and their device has the ability to swap out the CT termination resistor so you can use larger CTs if necessary. My house is wired for 400 A service but has 200 A service so I had to go with larger CTs on this project since the SCT-013-000 would not fit on my wiring. I am using these:

https://spwindustrial.com/wattcore-wc4-split-core-current-transformer-200-0-1a-opening-1-3x1-7/

I assume the CTs will use something like a 1/8 inch plug to attach to the HAT so that they can be easily adaptable to other CTs if needed.

skipfire, 2021-11-05:
I have a default resistor plus 4 open holes to support 2 more resistors, and a 3-pin header for jumper to pick between default and custom resistors.  For the plug, the specs I have seen say that it is a 3.5.

jgyates, 2021-11-05:
I am sure you are right, I was spouting off from memory and 1/8 inch came out. Sounds good on the CT front. I have an interface in genmon that will accept the external CT. I can help with the add on for the management of the temp sensors and CTs. Let me know how I can help. The 1 wire temp sensor should work with the gentemp.py plugin.

skipfire, 2021-11-05:
No concerns about the pins it's going to use?  It does overlap with some of the GENGPIO entries.

jgyates, 2021-11-05:
35,37, 38 and 40 do conflict with the defaults in gengpio.py but you have the ability to change the GPIO uses in the conf file so if someone is trying to use both plugins then they can always change the default.

Here are the two conf files that use GPIO:

https://github.com/jgyates/genmon/blob/master/conf/gengpio.conf
https://github.com/jgyates/genmon/blob/master/conf/gengpioin.conf

I can also make gengpio.py ignore some gpio values if the entry in the conf file is invalid or zero.

lmamakos, 2021-11-06:
Regarding the CT connection; I've seen some CT versions that have an integral burden resistor, so you'd want to easily be able to disable the one on board.  Perhaps the way you have the 3-pin header configured would allow that option.  

The power monitoring device that I use has spring terminals to accept bare/tinned wires rather than jacks.  It's got 32 channels, so physical space could be an issue for that product.  (The Brulltech [GreenEye Monitor](https://www.brultech.com/greeneye/) if you're curious.)  FWIW.

skipfire, 2021-11-06:
The 3-pin header would allow it. I'll check out using 2-pair of screw/spring terminals in addition to the 3.5mm plugs.

closed 2021-12-09 by jgyates

jgyates, 2022-01-11:
Support for this add on board is included in V1.18.04. You can get the HAT here:

https://pintsize.me/store/ols/products/generator-sensor-hat