jgyates/genmon #596: Does Genmon bypass the generac ev2.0 activation code on a new generator?

mikewesner, 2021-11-05:
First, Huge thanks to the creators/maintainers of this project.  

This is not a fault/error/issue, more of a question about activation.

I had a new 22kw generac installed yesterday.  I had bought the custom hat and built the box ahead of time.  We installed the genmon box even before first startup.   It was a new unit, had to install the battery and all the usual stuff.

The installer was surprised that he didn't need to input the activation code and everything was just up and running.  Entered the time and that was it.   The genmon stuff worked right out of the box on the evolution 2.0 controller.  MQTT to home assistant also.  Fantastic!

Again, huge thanks to the project and the docs.  I loved that it was easy enough to find the hardware to build this little project.  

### Expected Behavior

Input of generac activation code required?

### Actual Behavior

No activation code needed, Genmon worked perfectly!

### Steps to Reproduce (including precondition)

Brand new generac 22kw, ev2.0 controller.

### Your Environment

- Generator Model: 7042 generac
- Genmon Logs: Submit via the About page
- Genmon Version: 1.17.04

mikewesner, 2021-11-05:
i should also tack on a comment here that I think the reported power output seemed pretty close on first test run.   I need to do a longer test, but I have an Iotawatt meter on my panel so I should be able to do some testing on power output accuracy on my 22kw.

skipfire, 2021-11-05:
I have no clue on the activation, but the 22kws have had CTs since early 2019 (around when I got mine) and should have fairly good usage numbers.  I have a Sense and the numbers are usually very close, enough I could say the readings are just seconds apart.

jgyates, 2021-11-05:
I honestly do not know as that has never been asked before. I think the wifi equipped units do activation over the internet once connected so it may have a period that it works without activations, but that is just a guess. The software does nothing specific to bypass activation. It assumes it has already been activated. Let me know in this thread if activation is required after a period of time. It may work but just not activate the transfer switch so once it is connected to the switch a full power loss test may be in order.

grsthegreat, 2021-11-06:
the generator should still have needed an activation code...with or without genmon. very odd.

jgyates, 2021-11-06:
I am going to close this issue for now. Feel free to post here for additional questions or and update on this topic.

closed 2021-11-06 by jgyates

mikewesner, 2021-11-11:
I just noticed on the web interface, maintenance page.   Generator Serial Number : None - Controller has been replaced
I am pretty sure this was a brand new unit.  Based on how it arrived and such.   Just a curiosity.   We have done a load test and I have messed with it a bunch and its all working.   I have the activation code handy in case I ever need it I guess.

grsthegreat, 2021-11-11:
that is weird. a replaced controller does not need reactivation. have you verified with generac that that unit it registered to you?

jgyates, 2021-11-11:
I can confirm that a replaced controller will have a blank serial number and will not need activation.