jgyates/genmon #597: gentankdiy application

jmcjclark, 2021-11-07:
Awesome program! Been using for several years and it just keeps getting better!!!
I am going to implement the gentankdiy but the tank is so far away from my gen. I was wondering if the gentankdiy could be implemented on a separate raspi and send the needed data to the genpi? Any input and suggestions greatly appreciated!

### Your Environment

- Generator Model: G0070631
- Genmon Version: V1.17.04

jgyates, 2021-11-07:
This should be possible with a few extra steps.

Run gentankdiy.py with this command line:

       sudo python gentankdiy.py -a 192.168.1.20 -c /etc/genmon

where -a is the ip address of the system running genmon and -c is the path of the config files that gentankutl.py will use.

The config path referenced  should contain gentankdiy.conf and genmon.conf. genmon.conf should have, at minimum the following entries:

# the log file location typically /var/log/
loglocation = /var/log/
# The TCP port used for communicating with the other apps like
# ClientInterface.py and the web interface (required)
server_port = 9082
# Set to True for Multiple Instances of Genmon on a single system, for use
# with multiple Generators with one server. If true the genmon and the add on
# programs will not check if and instance is already loaded.
multi_instance = False

You will also need gentankdiy.py to be in a directory with the /genmonlib directory and all of its contents one level below.

In short have two copies of the entire project on two separate system and run gentankdiy.py with the -a and -c options manually.

jgyates, 2021-11-08:
I am going to close this issue but feel free to post here if you have any questions or updates.

closed 2021-11-08 by jgyates