jgyates/genmon #598: Genmon 1.18.0 - Python 3.x is the default going forward

jgyates, 2021-11-09:
genmon started as a python 2.x project and eventually supported both python 2.x and 3.x, with python 2.x being the default. Starting with genmon v1.18.0, which will be on 11/9/2021, the default will be python 3.x. Python 2.x will not be tested going forward.

This change should only affect most users in one way: when you upgrade from a previous version to v1.18.0 the upgrade may take longer, so if you web interface does not immediately come back after the upgrade please wait up to 15 min before taking any action. This longer wait is due to the software installing the python 3.x versions of the needed libraries.

Python 2.x has been deprecated for a while and beginning with Raspbian 11 (bullseye) it is no longer included with the latest version raspbian OS.

nicktdot, 2021-11-10:
Weather module seems to have broken after update with weather_at_place error.

2021-11-10 08:51:11,930 : Error in GetObservation: (**My City**) : 'OWM' object has no attribute 'weather_at_place' : myweather.py:
1082021-11-10 08:51:11,930 : Error in GetObservation: (**My City**) : 'OWM' object has no attribute 'weather_at_place' : myweather.py:108

jgyates, 2021-11-10:
I looked into this and the issue is that the Open Weather API has changed. My genmon system is using pyowm library version 2.10 but my test system is using V3.2.0, which shows the same error you are seeing. Both systems are using python3.x. I will need to modify the code to address this. I should have something to correct this later today. Thanks for finding this.

nicktdot, 2021-11-10:
Thanks! I went by the genmonmaint.sh script and installed the "non Python3" option of OWM 2.9.0 and it obviously fixed it.

So might be quicker to change the conditional check and just go with 2.9.0 so you dont have to rush a fix out

if [ "$usepython3" = true ] ; then
    sudo $pipcommand install pyowm -U
  else
    sudo $pipcommand install pyowm==2.9.0 -U
  fi

jgyates, 2021-11-10:
Good idea!

jgyates, 2021-11-10:
I checked in an update. You will need to uninstall the pyowm library then just re-load the new genmon. Via ssh do this:

     cd genmon
     ./startgenmon.sh stop
     sudo pip3 uninstall pyowm
     ./genmonmaint.sh -u
     ./startgenmon.sh restart

This worked for me. Let me know how it goes.

nicktdot, 2021-11-10:
Worked, now on 2.10.0

zekyl314, 2021-11-10:
FYI - I ran into an issue with upgrading to 1.8.0 because I am on jessie on my raspberry pi, and the sources.list was pointing to a wrong repo.  It could not locate anything and I wasn't even able to install pip3.  So I updated the sources.list and I am now re-running the process to upgrade.  If I run into any other issues, I'll post here.  But I wanted to put this out there in case anyone else runs into this issue.

zekyl314, 2021-11-10:
I also didn't see any output in the Genmon logs about having issues trying to install the modules.  Should I have seen anything?

nicktdot, 2021-11-10:
You should see something in genloader.log

zekyl314, 2021-11-10:
> You should see something in genloader.log

Ahh!  The one log I failed to check.  Thanks.  Looks like I'm having some jessie python3 issues.  I'll fix those next.

wgstarks, 2021-11-11:
I tried updating but got an out of space error. Tried to revert to backup but now getting errors that python 3 iis not installed.
Complete errors attached. I only have access via ssh currently.
[genmon.txt](https://github.com/jgyates/genmon/files/7517087/genmon.txt)
d.

jgyates, 2021-11-11:
Where are you getting the out of space error?

I just checked in an update to genloader.py that may give us more details than then genmon.txt file has now.

To update type:

         cd genmon
         ./genmonmaint.sh -u

If you are out of space you can type this:

       df

and it should show each filesystem and the percentage full. You may can free up some space on log files, look in /var/log/. To remove old genmon log files type:

       sudo rm /var/log/gen*.log.?
       sudo rm /var/log/my*.log.?

To remove old archived log files of all kinds type:

       sudo rm /var/log/*.gz

then type 'df' again and  see if you have more free space, if you have free space type this to restart genmon:

       ./startgenmon.sh restart

wgstarks, 2021-11-11:
Ran the update and installed without error.

I had run apt update & apt upgrade after my last post and noticed a bunch of python 3 packages installing. Maybe that has something to do with the success. IDK

Running great now though.

jgyates, 2021-11-13:
FYI:

If you are running raspbian Jessie you will have issues upgrading since Jessie software updates are not longer support by the Raspberry Pi Foundation. If you are running Raspbian Jessie you can either not upgrade to this version, upgrade but force python 2 to be used, or reinstall the operating system with the latest version (Raspbian Bullseye)

You can find the version of the OS your are using by typing this or looking on the About page:

     cat /etc/os-release

Jessie will look like this:

      PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
      NAME="Raspbian GNU/Linux"
      VERSION_ID="8"
      VERSION="8 (jessie)"
      ID=raspbian
      ID_LIKE=debian
      HOME_URL="http://www.raspbian.org/"
      SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
      BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

The buster is version 10 and bullseye is version 11. If you have jessie version then I would suggest one of two things:

Edit your crontab to force using python2

      sudo crontab -e

If you are prompted to pick and editor select nano, if not then you will be prompted with a file that has the following line

      @reboot sleep 50 && /bin/bash /home/pi/genmon/startgenmon.sh start

Change this to

     @reboot sleep 50 && /bin/bash /home/pi/genmon/startgenmon.sh -p 2 start

Then save the file by typing Ctrl+X then Y then Enter

Then reboot, this should start genmon with python 2 instead of python 3. Future versions of genmon may not work as python 2 will not longer be tested or supported.

Get a new SD card and install bullseye and import your genmon conf files to /etc/genmon

Of course you can run python 2 with the above instructions to get you going then upgrade your OS at a later time.

wgstarks, 2021-11-13:
> If you are running raspbian Jessie you will have issues upgrading since Jessie software updates are not longer support by the Raspberry Pi Foundation

Thanks. Im running buster though.

After running apt update and apt upgrade genmon is working great though. It looks like that’s all that was needed.

The packages install too fast to really keep track of but I saw that python was a big part of the update.

jgyates, 2021-11-13:
That comment was to for you specifically. I created this thread as a companion to the release notes for this release and this comment is intended to help anyone trying to upgrade.

wgstarks, 2021-11-13:
😳
😁

nekitti, 2021-11-14:
Had a glitch with mine when I upgraded to 1.18.0.  It stopped working.  I did the sudo apt update in the raspberry pi terminal and then it worked fine.  Probably just needed to update some of the Python 3.0 stuff.  Thanks for keeping this going jgyates!

tcflint, 2021-11-21:
> Had a glitch with mine when I upgraded to 1.18.0. It stopped working. I did the sudo apt update and sudo apt full-upgrade in the raspberry pi terminal and then it worked fine. Probably just needed to update some of the Python 3.0 stuff. Thanks for keeping this going jgyates!

+1 for jgyates always maintaining and improving this software.
It is fantastic and everything I was hoping to see/do when I got my Gen.
Thanks again

closed 2022-03-15 by jgyates