jgyates/genmon #599: Fresh install, on boot showing library not installed.

j0mama, 2021-11-09:
Fresh install on pi3, used scripts, upon installation:

E: Package 'python-pip' has no installation candidate, 
(not sure if thats standard)

Everything else looks fine when starting using /bin/bash /home/pi/genmon/startgenmon.sh start

/bin/bash /home/pi/genmon/startgenmon.sh start
Starting genmon python scripts
Warning: required library flask not installed. Attempting to install....
Warning: required library pyserial not installed. Attempting to install....
Warning: required library crcmod not installed. Attempting to install....
Warning: required library pyowm not installed. Attempting to install....
Warning: required library pytz not installed. Attempting to install....
Warning: required library pysnmp not installed. Attempting to install....
Warning: required library ldap3 not installed. Attempting to install....
Warning: required library smbus not installed. Attempting to install....
Warning: required library pyotp not installed. Attempting to install....
Warning: required library psutil not installed. Attempting to install....
Starting....
Starting /home/pi/genmon/genmon.py 
Starting /home/pi/genmon/genserv.py 

But checking:

flask --version
Python 3.9.2
Flask 1.1.2
Werkzeug 1.0.1

The webpage does not come up, so I dont know how to troubleshoot further.

jgyates, 2021-11-09:
Hi @j0mama 

I assume you are using raspbian 11 (bullseye). There is a new version that is going to be posted tonight or tomorrow that would fix this issue.

This thread has more details:

https://github.com/jgyates/genmon/issues/598

Short story is that the release of raspbian 11 two days ago removed python 2.7.

In the intirrum you can install forcing python 3 to be used with this command:

     ./genmonmaint.sh -p 3 -i

Let me know if that does not fix the issue.

jgyates, 2021-11-09:
also, when starting genmon use

     ./startgenmon.sh -p 3 restart

The version update I am testing now will make these options the default.

j0mama, 2021-11-09:
Thanks! that worked!

closed 2021-11-09 by j0mama

jgyates, 2021-11-10:
FYI, Version 1.18.0 has been posted

j0mama, 2021-11-10:
Cheers! Thanks for your hard work! donated!