jgyates/genmon #600: Generator Monitoring not loading after update to 1.18.0

jjmacneil, 2021-11-10:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Generator details loading after login

### Actual Behavior

Browser displays "Loading Generator Monitor" even after 45 minutes.

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

GenMon 1.18.0

- Generator Model: Generac 22kw
- Genmon Logs: Submit via the About page
- Genmon Version: 1.18.0

jgyates, 2021-11-10:
Hi @jjmacneil ,

Have you tried restarting the pi?

If that does not correct the issue I will need to see the logs to troubleshoot. If the genmon web interface does not come up after a reboot then you can log into the pi via ssh then type this:

      cd genmon
      ./startgenmon.sh restart

copy and paste the output into this thread. If genmon starts or restarts with the above command then type

      python3 ClientInterface.py

This will give you a prompt if genmon is loaded, it will let you know if genmon is not loaded.

If you get the prompt type:

         generator: sendlogfiles

This will send me your log files if you have outbound email enabled and working. They type this to exit the program:

       exit

Let me know if you have any unusual things about your install: for example, what version were you upgrading from. This is a raspberry pi? If yes what version of the operating system are you using?

jgyates, 2021-11-10:
Another thing you can try once you have logged in via ssh:

     cd genmon
     bash ./genmonmaint.sh -i

This will reinstall the libraries with the python3 version. It will prompt you to copy conf files, select no (you have already done this). It will prompt you to start genmon on boot, select no (you have already done this)

jjmacneil, 2021-11-10:
Hi,

I restarted and now the GenMon is not loaded so I can't even get to the login page. It appears it's having trouble installing python3 libraries for some reason? How can I get the logs to you if GenMon is not loaded?

jjmacneil, 2021-11-10:
Hi,
I see a message stating pip3: command not found

jgyates, 2021-11-10:
if you can log in to the pi via ssh then type

     cd genmon
     bash ./genmonmaint.sh -i

This should fix the issue if pip3 is not installed. From the command line on the pi type

      cat /var/log/genloader.log

And copy and paste the output here.

jgyates, 2021-11-10:
when do you see the message "pip3: command not found"

jjmacneil, 2021-11-10:
Linux genpi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l

  _______ _______ ______      _______ ___ 
 |   _   |   _   |   _  \    |   _   |   |
 |.  |___|.  1___|.  |   |   |.  1   |.  |
 |.  |   |.  __)_|.  |   |   |.  ____|.  |
 |:  1   |:  1   |:  |   |   |:  |   |:  |
 |::.. . |::.. . |::.|   |   |::.|   |::.|
 `-------`-------`--- ---'   `---'   `---'
                                          

Last login: Tue Nov  9 22:53:49 2021 from 192.168.200.17
pi@genpi [~] cd genmon
pi@genpi [~/genmon] ./startgenmon.sh restart
Restarting genmon python scripts
Warning: required library flask not installed. Attempting to install....
Error in CheckBaseSoftware: [Errno 2] No such file or directory: 'pip3': 'pip3'
Installing python3-pip
Error in InstallBaseSoftware: name 'libraryname' is not defined
Error installing module: flask: [Errno 2] No such file or directory: 'pip3': 'pip3'
Error: unable to install library flask
Warning: required library pyserial not installed. Attempting to install....
Error in CheckBaseSoftware: [Errno 2] No such file or directory: 'pip3': 'pip3'
Installing python3-pip
Error in InstallBaseSoftware: name 'libraryname' is not defined
Error installing module: pyserial: [Errno 2] No such file or directory: 'pip3': 'pip3'
Error: unable to install library pyserial
Warning: required library crcmod not installed. Attempting to install....
Error in CheckBaseSoftware: [Errno 2] No such file or directory: 'pip3': 'pip3'
Installing python3-pip
Error in InstallBaseSoftware: name 'libraryname' is not defined
Error installing module: crcmod: [Errno 2] No such file or directory: 'pip3': 'pip3'
Error: unable to install library crcmod
Warning: required library pyowm not installed. Attempting to install....
Error in CheckBaseSoftware: [Errno 2] No such file or directory: 'pip3': 'pip3'
Installing python3-pip
Error in InstallBaseSoftware: name 'libraryname' is not defined
Error installing module: pyowm: [Errno 2] No such file or directory: 'pip3': 'pip3'
Error: unable to install library pyowm
Warning: required library pytz not installed. Attempting to install....
Error in CheckBaseSoftware: [Errno 2] No such file or directory: 'pip3': 'pip3'
Installing python3-pip
Error in InstallBaseSoftware: name 'libraryname' is not defined
Error installing module: pytz: [Errno 2] No such file or directory: 'pip3': 'pip3'
Error: unable to install library pytz
Warning: required library pysnmp not installed. Attempting to install....
Error in CheckBaseSoftware: [Errno 2] No such file or directory: 'pip3': 'pip3'
Installing python3-pip
Error in InstallBaseSoftware: name 'libraryname' is not defined
Error installing module: pysnmp: [Errno 2] No such file or directory: 'pip3': 'pip3'
Error: unable to install library pysnmp
Warning: required library ldap3 not installed. Attempting to install....
Error in CheckBaseSoftware: [Errno 2] No such file or directory: 'pip3': 'pip3'
Installing python3-pip
Error in InstallBaseSoftware: name 'libraryname' is not defined
Error installing module: ldap3: [Errno 2] No such file or directory: 'pip3': 'pip3'
Error: unable to install library ldap3
Error in CheckBaseSoftware: [Errno 2] No such file or directory: 'pip3': 'pip3'
Installing python3-pip
Error in InstallBaseSoftware: name 'libraryname' is not defined
Error installing module: pyasn1: [Errno 2] No such file or directory: 'pip3': 'pip3'
Warning: required library smbus not installed. Attempting to install....
Error in CheckBaseSoftware: [Errno 2] No such file or directory: 'pip3': 'pip3'
Installing python3-pip
Error in InstallBaseSoftware: name 'libraryname' is not defined
Error installing module: smbus: [Errno 2] No such file or directory: 'pip3': 'pip3'
Error: unable to install library smbus
Warning: required library pyotp not installed. Attempting to install....
Error in CheckBaseSoftware: [Errno 2] No such file or directory: 'pip3': 'pip3'
Installing python3-pip
[... 303 lines cut ...]
2021-11-09 23:02:13,096 : Installing python3-pip
2021-11-09 23:02:22,639 : Error in InstallBaseSoftware: name 'libraryname' is not defined : genloader.py:2142021-11-09 23:02:22,644 : Error installing module: psutil: [Errno 2] No such file or directory: 'pip3': 'pip3' : genloader.py:317
2021-11-09 23:02:22,646 : Error: unable to install library psutil
2021-11-09 23:02:22,647 : Warning: required library chump not installed. Attempting to install....
2021-11-09 23:02:22,652 : Error in CheckBaseSoftware: [Errno 2] No such file or directory: 'pip3': 'pip3' : genloader.py:184
2021-11-09 23:02:22,653 : Installing python3-pip
2021-11-09 23:02:28,837 : Error in InstallBaseSoftware: name 'libraryname' is not defined : genloader.py:2142021-11-09 23:02:28,842 : Error installing module: chump: [Errno 2] No such file or directory: 'pip3': 'pip3' : genloader.py:317
2021-11-09 23:02:28,843 : Error: unable to install library chump
2021-11-09 23:02:28,845 : Warning: required library twilio not installed. Attempting to install....
2021-11-09 23:02:28,849 : Error in CheckBaseSoftware: [Errno 2] No such file or directory: 'pip3': 'pip3' : genloader.py:184
2021-11-09 23:02:28,850 : Installing python3-pip
2021-11-09 23:02:34,754 : Error in InstallBaseSoftware: name 'libraryname' is not defined : genloader.py:2142021-11-09 23:02:34,759 : Error installing module: twilio: [Errno 2] No such file or directory: 'pip3': 'pip3' : genloader.py:317
2021-11-09 23:02:34,760 : Error: unable to install library twilio
[... cut ...]
2021-11-09 23:02:40,883 : Error in CheckBaseSoftware: [Errno 2] No such file or directory: 'pip3': 'pip3' : genloader.py:184
2021-11-09 23:02:40,884 : Installing python3-pip
2021-11-09 23:02:46,716 : Error in InstallBaseSoftware: name 'libraryname' is not defined : genloader.py:2142021-11-09 23:02:46,722 : Error installing module: pyopenssl: [Errno 2] No such file or directory: 'pip3': 'pip3' : genloader.py:317
2021-11-09 23:02:46,722 : Error: unable to install library pyopenssl
2021-11-09 23:02:46,723 : Error check system readiness. Exiting
pi@genpi [~/genmon]

jgyates, 2021-11-10:
This is good info.

Try typing this:

        sudo apt-get update
        sudo apt-get install python3-pip

 These look like the commands that are failing.

jjmacneil, 2021-11-10:
Thank you, but no luck. See log below

//archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Hit:3 https://deb.nodesource.com/node_10.x stretch InRelease
Reading package lists... Done
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
E: Repository 'http://archive.raspberrypi.org/debian buster InRelease' changed its 'Suite' value from 'testing' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
pi@genpi [~] sudo apt-get install python3-pip
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  dh-python libpython3-dev libpython3.7-dev
  python3-asn1crypto python3-cffi-backend
  python3-crypto python3-cryptography python3-dbus
  python3-dev python3-distutils python3-entrypoints
  python3-gi python3-keyring python3-keyrings.alt
  python3-lib2to3 python3-secretstorage
  python3-setuptools python3-wheel python3-xdg
  python3.7-dev
Suggested packages:
  python-crypto-doc python-cryptography-doc
  python3-cryptography-vectors python-dbus-doc
  python3-dbus-dbg gnome-keyring libkf5wallet-bin
  gir1.2-gnomekeyring-1.0 python-secretstorage-doc
  python-setuptools-doc
The following NEW packages will be installed:
  dh-python libpython3-dev libpython3.7-dev
  python3-asn1crypto python3-cffi-backend
  python3-crypto python3-cryptography python3-dbus
  python3-dev python3-distutils python3-entrypoints
  python3-gi python3-keyring python3-keyrings.alt
  python3-lib2to3 python3-pip python3-secretstorage
  python3-setuptools python3-wheel python3-xdg
  python3.7-dev
0 upgraded, 21 newly installed, 0 to remove and 1 not upgraded.
Need to get 47.7 MB/49.5 MB of archives.
After this operation, 80.5 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Err:1 http://raspbian.raspberrypi.org/raspbian buster/main armhf libpython3.7-dev armhf 3.7.3-2+deb10u2
  404  Not Found [IP: 93.93.128.193 80]
Err:2 http://raspbian.raspberrypi.org/raspbian buster/main armhf python3.7-dev armhf 3.7.3-2+deb10u2
  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/python3.7/libpython3.7-dev_3.7.3-2+deb10u2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/python3.7/python3.7-dev_3.7.3-2+deb10u2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
pi@genpi [~]

jjmacneil, 2021-11-10:
I re-ran the following command

sudo apt update

and answered y to the questions as required which allowed the updates to install.

I was then able to install python3 just waiting for the install to complete now.

jjmacneil, 2021-11-10:
Is there anything I have to do after python install finishes to get GenMon running again?

jgyates, 2021-11-10:
That is good news. So far this looks like the issue is  your raspberry pi's ability to reach the raspbian software repository. 

The IP address that it is failing to reach is  93.93.128.193 80 which is raspbian.raspberrypi.org

I read that comcast xfinity has some outages that affect parts of the internet so this may be what is causing the problem.

Let me know how it goes.

If pip3 is installed you can run this

      ./startgenmon.sh restart

This should stop any running instances and restart genmon. This will also attempt to install any missing libraries using pip3. This could take a few minutes but as long as you have a good connection to the internet you should be OK.

once it finishes running then genmon should be loaded.

jjmacneil, 2021-11-10:
Thank you! I appreciate your quick responses! All looks good now. Thanks again. I guess I need to update the Pi OS to ensure it's running the most stable OS.

jgyates, 2021-11-10:
you can just update your OS software with this command:

     sudo apt-get update
     sudo apt-get upgrade

this will update all the system software. You probably will need to reboot after that to make it take effect.  I use several pis and I run this every so often to keep them updated. bullseye was just released but looks like you are running buster, which still has at least 2 years, possibly 3 of updates.

jgyates, 2021-11-10:
I am going to close this issue but feel free to post here if you have any other question or updates.

closed 2021-11-10 by jgyates

jjmacneil, 2021-11-10:
Thank you! I've been hesitant to update as the last time I did that I lost the custom network configuration which required me to manually connect to the Pi directly to re-configure. Is there a way to ensure that the custom WiFi settings are not overwritten with defaults? I have custom info in the conf file to disable onboard WiFi and previously had special commands to limit WiFi frequencies that could be used.

jgyates, 2021-11-10:
Normally if a config file is going to be overwritten with apt-get it will prompt you with a choice to keep using your old one or upgrade to the new one. To be on the safe side I would make a backup any files that may be customized. At a minimum just make a copy in the same directory.

jjmacneil, 2021-11-10:
Thank you! I believe the last upgrade removed the drivers that I installed for the USB network adapter so I had no choice but to remove the Pi from the Generator for direct network connectivity to re-install the drivers. I wonder if the upgrade will do the same thing again or if it will actually leave the drivers in-place.

griffths, 2021-11-10:
@jgyates 

This is the console output of the apt-get update command.

I have not gone any further with "apt-get upgrade" at this point because I am suspicious that the results are not good but am not an expert.

At the moment, to say my WI-Fi is sketchy is a gross understatement (-82-85 dBm) so I am hesitant to proceed with the update but would you please comment on the following console output and give me your opinion on proceeding with the OS update.

pi@raspberrypi:~ $ sudo apt-get update
Err:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
  Temporary failure resolving 'raspbian.raspberrypi.org'
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:3 http://archive.raspberrypi.org/debian buster/main armhf Packages [393 kB]
Fetched 426 kB in 18s (23.8 kB/s)
Reading package lists... Done
N: Repository 'http://archive.raspberrypi.org/debian buster InRelease' changed its 'Suite' value from 'testing' to 'oldstable'
W: Failed to fetch http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease  Temporary failure resolving 'raspbian.raspberrypi.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.

Thanks

Jay 

> you can just update your OS software with this command:
>
> ```
>  sudo apt-get update
>  sudo apt-get upgrade
> ```
>
> this will update all the system software. You probably will need to reboot after that to [...]

jgyates, 2021-11-10:
That output is normal. The values changed from testing to oldstable because of the new release (buster is old stable now, bullseye is stable)

jgyates, 2021-11-10:
the last line my be due to your poor connection.