jgyates/genmon #601: Not able to Login to Web Interface - Post Python 3 upgrade

sambeetm, 2021-11-10:
Getting following error in the /var/log/genserv.log.

I do have a "%" character in the password. Looks like its causing issue now after the python 3 upgrade. 

Error in MyConfig:ReadValue: GenMon: http_pass: '%' must be followed by '%' or '(', found: '%' : myconfig.py:85
2021-11-10 13:02:37,306 : Starting /home/pi/genmon/genserv.py, Port:443, Secure HTTP: True, SelfSignedCert: True

jgyates, 2021-11-10:
I found a problem I introduced in V1.17.5 in gengpio.py. I checked in an update so click update from the About page or type

      ./genmonmaint.sh -u
      ./startgenmon.sh restart

Let me know how it goes.

sambeetm, 2021-11-10:
Thanks for the quick fix. Working as before !

closed 2021-11-10 by jgyates