jgyates/genmon #603: Broke genmon with 1.18.0 upgrade attempt

tmorse305, 2021-11-11:
_Use_ the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

I broke genmon when I tried to upgrade to 1.18.0.  I attempted the upgrade from the console and suspect I didn't wait long enough for it to complete based on what I read (after the fact) on this page.  

### Actual Behavior

The upgrade proceeded to the end it seemed but never restarted, just a spinning circle.  I rebooted the RPi which came back on line.  It came back online using the internal WiFi radio instead of a dongle that I had been using so I suspect there have been changes to some basic configuration.

### Steps to Reproduce (including precondition)

I tried a re-install with the following results, see screen shots.  I ran git clone https://github.com/jgyates/genmon.git first then  bash ./genmonmaint.sh -i

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: attempted 1.18.0 from 1.17.x
- Linux raspberrypi 5.4.83-v7+ #1379 SMP Mon Dec 14 13:08:57 GMT 2020 armv7l
- PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
[image]
[image]

jgyates, 2021-11-11:
This is an almost identical issue to this thread:

https://github.com/jgyates/genmon/issues/600

Notice the "404 Not Found [IP: 93.93.128.193 80]"

Try typing this:

    sudo apt-get update
    sudo apt-get install python3-pip

If this fails, wait for a while and try again. What is happening is that your pi is failing to connect to the server that allows software to be installed. It is an internet / routing problem.

If the above passes then type:

    ./startgenmon.sh restart

It may take a few minutes to install any needed libraries.

Let me know how it goes.

RoyRomeo, 2021-11-12:
GenMon will not load after last update.  Tried first line "sudo apt-get update" which works but second line "sudo apt-get install python3-pip" fails.  Been trying for two days.  Any suggestions?

jgyates, 2021-11-12:
What is the failure message?
Are you getting any messages like "404 Not Found [IP: 93.93.128.193 80]"?

jgyates, 2021-11-12:
Just a thought, if you have a marginal wifi connection and you are finding that this upgrade is failing you can try to disconnect the pi from the generator, bring the pi closer to your wifi access point by powering it via the micro USB connector, then try to perform the upgrade. If you have a good wifi signal and it is still failing let me know.

zekyl314, 2021-11-12:
> GenMon will not load after last update. Tried first line "sudo apt-get update" which works but second line "sudo apt-get install python3-pip" fails. Been trying for two days. Any suggestions?

What OS are you running?  I was on jessie and was having a similar issue due to the apt sources.list not having resolvable URLs.

RoyRomeo, 2021-11-12:
How do you copy and paste?

RoyRomeo, 2021-11-12:
The following NEW packages will be installed:
  dh-python libpython3-dev libpython3.7-dev python3-asn1crypto
  python3-cffi-backend python3-crypto python3-cryptography python3-dbus
  python3-dev python3-distutils python3-entrypoints python3-gi python3-keyring
  python3-keyrings.alt python3-lib2to3 python3-pip python3-secretstorage
  python3-setuptools python3-wheel python3-xdg python3.7-dev
0 upgraded, 21 newly installed, 0 to remove and 1 not upgraded.
Need to get 47.7 MB/49.5 MB of archives.
After this operation, 80.5 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Err:1 http://raspbian.raspberrypi.org/raspbian buster/main armhf libpython3.7-dev armhf 3.7.3-2+deb10u2
  404  Not Found [IP: 93.93.128.193 80]
Err:2 http://raspbian.raspberrypi.org/raspbian buster/main armhf python3.7-dev armhf 3.7.3-2+deb10u2
  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/python3.7/libpython3.7-dev_3.7.3-2+deb10u2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/python3.7/python3.7-dev_3.7.3-2+deb10u2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

RoyRomeo, 2021-11-12:
I also have a good wifi signal.

zekyl314, 2021-11-12:
That is interesting, I tried hitting those URLs from my systems and I get the same error message.  Can you post the contents of /etc/apt/sources.list?

RoyRomeo, 2021-11-12:
I get "Permission denied"

zekyl314, 2021-11-12:
> I get "Permission denied"

You may have to change to root, or run "sudo cat /etc/apt/sources.list"

RoyRomeo, 2021-11-12:
pi@genpi [~] sudo cat /etc/apt/sources.list
deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi

zekyl314, 2021-11-12:
> pi@genpi [~] sudo cat /etc/apt/sources.list deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi
> 
> # Uncomment line below then 'apt-get update' to enable 'apt-get source'
> #deb-src http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi

Thanks, this is the same exact as mine.  So that's good at least.  

Have you done a sudo apt-get update and then sudo apt-get upgrade to see if that changes anything?

RoyRomeo, 2021-11-12:
I just tried that and it appears to have run successfully.

zekyl314, 2021-11-12:
> I just tried that and it appears to have run successfully.

But installing these few modules still fails?  I'm at a loss why it's trying to grab a module that doesn't exist on the sources location.

RoyRomeo, 2021-11-12:
Yes, still fails.

zekyl314, 2021-11-12:
I'm at a loss, maybe someone else following will have another idea.

RoyRomeo, 2021-11-12:
Okay, thanks.

tmorse305, 2021-11-12:
@jgyates , I tried the suggestion above but I still get the file not found error.  When I looked at the repository it looks like the files it's trying to retrieve are not there.  See the screen shots.  I also tried to do an OS update thinking my system was too out of date but I get a similar error because it can't find the file.  It doesn't seem to be a internet problem.
[image]
m
[image]
[image]

[image]
.

jgyates, 2021-11-12:
I think this may be a problem with Jessie. Can you check to see if you are running jessie?

You can find the version of the OS your are using by typing this:

       cat /etc/os-release

Jessie will look like this:

     PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
     NAME="Raspbian GNU/Linux"
     VERSION_ID="8"
     VERSION="8 (jessie)"
     ID=raspbian
     ID_LIKE=debian
     HOME_URL="http://www.raspbian.org/"
     SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
     BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

The buster is version 10 and bullseye is version 11. If you have jessie version then I would suggest one of two things:

  1) Edit your crontab to force using python2

         sudo crontab -e

If you are prompted to pick and editor select nano, if not then you will be prompted with a file that has the following line

     @reboot sleep 50 && /bin/bash /home/pi/genmon/startgenmon.sh start

Change this to 

     @reboot sleep 50 && /bin/bash /home/pi/genmon/startgenmon.sh -p 2 start

Then save the file by typing Ctrl+X then Y then Enter

Then reboot, this should start genmon with python 2 instead of python 3. Future versions of genmon may not work as python 2 will not longer be tested or supported.

2) Get a new SD card and install bullseye and import your genmon conf files to /etc/genmon

Of course you can run python 2 with the above instructions to get you going then upgrade your OS at a later time. Unfortunately it appears the older versions of the OS (Jessie) will need to be upgraded.

jgyates, 2021-11-12:
you can also test this out with this command:

       cd genmon
       ./startgenmon.sh -p 2 restart

RoyRomeo, 2021-11-12:
I have the problem and mine is Linux 10 (buster).

RoyRomeo, 2021-11-12:
cd genmon
./startgenmon.sh -p 2 restart.

Did the trick.  Thanks very much.

tmorse305, 2021-11-12:
@jgyates , I’m running Buster

jgyates, 2021-11-12:
@tmorse305 

Yes, python 3 will be supported going forward, not python 2. Yes, this is an issue outside of genmon.

If you are on buster, try this:

      sudo apt-get update

Type 'y' for yes when prompted, then when that is finished type:

      sudo apt-get upgrade

If you see any errors let me know. If this completed without errors, then restart your pi with this command

     sudo reboot now

Wait a min or two for the pi to restart then log in via ssh and type:

    sudo apt-get update
    sudo apt-get install python3-pip

tmorse305, 2021-11-13:
@jgyates, I actually tried that already and it fails.  See my last post that included screen shots.  It's is looking for files in the repository that doesn't exist.  It seems the file that is there is newer perhaps, ...u3 vs ...u1  that it's looking for.

jjmacneil, 2021-11-13:
I would try the command below if you have not already and acknowledge the Y prompts to continue. If all updates install then move on to installing python 3 following directions that were mentioned previously in this case.

**sudo apt update**

jgyates, 2021-11-13:
@tmorse305 

Can you provide details about how the pi is networked? Static or dynamic IP? Is this pi in the same subnet as the rest of your network? Can you ping www.google.com? or any other relevant details. These are pretty basic commands to run on a raspberry pi running Buster. The only reason they do not run on Jessie is that the raspberry pi foundation ended software updates for jessie. What is failing with the update/ upgrade commands is just a basic software update of the system software on the pi, which is not related to genmon. Are you running Raspbian lite or the full graphic version?

I am short of ideas on what to suggest so any other details that you can provide may help me think of other things to try. If that does not work then you can try using python 2 as suggested for Jessie and try reinstalling the OS with BullsEye.

tmorse305, 2021-11-13:
@jgyates,  I am able to ping Google from my RPi so I don't think it's a network issue.  The files requested don't exist in the repository so I'm thinking something got corrupted in the OS or some basic configuration file.  I think it's time to wipe it and install Bullseye.  Thank you for all the help and advice with this.

jjmacneil, 2021-11-13:
@tmorse305 did you try the command I mentioned? I had similar trouble and because the Pi OS is older it required manual intervention when running the update process.

tmorse305, 2021-11-13:
> @tmorse305 did you try the command I mentioned? I had similar trouble and because the Pi OS is older it required manual intervention when running the update process.
> […](#)

I did try sudo apt update, but I got the same file not found error.

I just updated to Bullseye and I still can't install Python 3.
sudo apt update runs successfully now
sudo apt-get upgrade runs successfully
sudo apt-get install python3-pip fails with this message:

The following packages have unmet dependencies:
 libc6-dev : Breaks: libgcc-8-dev (< 8.4.0-2~) but 8.3.0-6+rpi1 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.

bash ./genmonmaint.sh -i fails, "pip3: command not found"

jgyates, 2021-11-13:
describe how you update to bullseye?

jjmacneil, 2021-11-13:
@tmorse305 try this

sudo apt install gcc-8-base

> On Nov 13, 2021, at 10:05 AM, tmorse305 ***@***.***> wrote:
> 
> The following packages have unmet dependencies:
> libc6-dev : Breaks: libgcc-8-dev (< 8.4.0-2~) but 8.3.0-6+rpi1 is to be installed
> E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
> 
> bash ./genmonmaint.sh -i fails, "pip3: command not found"
>

jgyates, 2021-11-13:
unless you did a full download of the bullseye image and reflashed the SD card I fully expect you to have the same issue. If you upgraded to bullseye from the command line you will likely have the same issue with apt-get. apt-get is suppose to resolve dependencies and install missing libraries but it is not for you in this instance. You may solve this with @jjmacneil's suggestion so it is worth a try, however I believe you still may have underlying issue with your pi system software.

tmorse305, 2021-11-13:
@jjmacneil, thank you, sudo apt install gcc-8-base ran successfully and then so did sudo apt-get install python3-pip.  The Genmon install then proceeded successfully as well.
@jgyates , I did do a command line upgrade, I figured it was worth a shot before nuking the SD and starting over.

Thanks again to both of you, Genmon is up and running again.

jgyates, 2021-11-13:
Great. I am glad it is working for you.

closed 2021-11-13 by jgyates