jgyates/genmon #604: NO issues just a lot of praise .  I knew nothing about pi when i stum

lampe208, 2021-11-11:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

lampe208, 2021-11-11:
I knew nothing about pi when I found this project.  I wanted to monitor my generator more than the wifi mod.  After reading thru the hardware , to me it was a little more than I thought, so I almost gave up.   Then I found PINTSIZE me board.  This is truly plug and play and very simple.  The only thing I needed was an extension power extender.    The PINTSize me even comes with power cables that are attached to the cable to go directly to the gen battery.  SO easy..  I hope if your struggling with how to hook this up use the PINTSize me , you will be pleased.   

(https://www.amazon.com/gp/product/B000M802RG/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1)

jgyates, 2021-11-11:
@lampe208 

thanks for the the feedback. I agree the PintSize.me board makes the setup a *LOT* easier. Thanks to @skipfire for creating this.

skipfire, 2021-11-11:
@lampe208 that feedback is why I made the boards, thanks for sharing it.

lampe208, 2021-11-11:
You bet , it takes the guesswork out of it.

lampe208, 2021-11-11:
I wonder if you could make a gauge for the raspberry pi info?

skipfire, 2021-11-11:
@lampe208 what Raspberry Pi info are you thinking of gauges for?  Since these are normally outside I could see CPU temp being a good one.

lampe208, 2021-11-11:
Raspberry pi temp for sure with red and green based off what is a problem when it starts getting that high  CPU load there’s another one I’ll keep thinking about it

lampe208, 2021-11-11:
Wi-Fi signal as well

jgyates, 2021-11-12:
I added this to the feature list

https://github.com/jgyates/genmon/blob/master/feature_requests.md

lampe208, 2021-11-12:
Im not seeing the gauges for raspberry… I just updated it in the about section?

https://github.com/jgyates/genmon/blob/master/feature_requests.md

jgyates, 2021-11-12:
All I have done is add this new request for functionality to the feature todo list. The software does not support this feature at this time.

lampe208, 2021-11-12:
ok gotcha

closed 2022-01-17 by jgyates

lampe208, 2022-06-03:
Sir I’m giving the error generator not receiving data I went outside to see
if a loose connection check the ground everything‘s fine seems like when I
updated the versions just stopped it was intermittent but now it’s steady
it won’t start the generator either from the app and when I started the
generator manually I get no data I’ll send you the logs if you could check
into what’s wrong I appreciate it Tom play

jgyates, 2022-06-04:
I have not received your logs. Your outbound email must be working for your logs to be submitted via the about page.

If you use gmail for outbound email then read this: https://github.com/jgyates/genmon/issues/682

You could have a bad serial connection. This page has serial trouble shooting info that can help you narrow down the source of the problem: https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

lampe208, 2022-06-04:
i just sent them

jgyates, 2022-06-04:
I have not recieved anything today. If you use gmail with genmon you will need to read the link above to make sure your gmail still works. Google changed the way things work as of 5/30/2022. See this thread: https://github.com/jgyates/genmon/issues/682

lampe208, 2022-06-04:
Didu get my email with attachment?

jgyates, 2022-06-04:
I have not.

jgyates, 2022-06-04:
you should be able to test your email on the settings page. You can specify a destination email address and send a test email,

lampe208, 2022-06-05:
The test email worked. Advise if you get the logs now

jgyates, 2022-06-05:
No logs. Try this:

Enter this URL, substituting your IP address of the pi:

      http://192.168.1.10:8000/cmd/support_data_json

copy and paste the output to a text file and attached the file to this thread.

then issue this command:

      http://192.168.1.10:8000/cmd/get_logs

This will download a file named genmon_logs.tar.gz. Attached this archive file to this thread.

Note that you must attached these files to the web interface at github. Do not send them as attachments in a reply email as they will stripped by github.

lampe208, 2022-06-05:
In  a browser?  Or in pi?  No such file when I run the command in pi. Browser gets no response. Refused to connect.    From: jgyates ***@***.***>Date: Saturday, June 4, 2022 at 23:08To: jgyates/genmon ***@***.***>Cc: lampe208 ***@***.***>, Mention ***@***.***>Subject: Re: [jgyates/genmon] NO issues just a lot of praise . I knew nothing about pi but this was great and fun to do. (Issue #604)No logs. Try this:Enter this URL, substituting your IP address of the pi:  http://192.168.1.10:8000/cmd/support_data_jsoncopy and paste the output to a text file and attached the file to this thread.then issue this command:  http://192.168.1.10:8000/cmd/get_logsThis will download a file named genmon_logs.tar.gz. Attached this archive file to this thread.Note that you must attached these files to the web interface at github. Do not send them as attachments in a reply email as they will stripped by github.

jgyates, 2022-06-05:
I don't know what you pasted above. This is not related to genmon. 

The following URLs should go in the browser address field, it will show support info:

      http://IPADDRESS:8000/cmd/support_data_json

The following command in the browser will download the logs to your local computer.

     http://IPADDRESS0:8000/cmd/get_logs

The text IPADDRESS above should be replaced with the IP address of your pi.

lampe208, 2022-06-05:
Sorry I was trying to paste the text from the logs I downloaded.   From: jgyates ***@***.***>Date: Sunday, June 5, 2022 at 17:45To: jgyates/genmon ***@***.***>Cc: lampe208 ***@***.***>, Mention ***@***.***>Subject: Re: [jgyates/genmon] NO issues just a lot of praise . I knew nothing about pi but this was great and fun to do. (Issue #604)I don't know what you pasted above. This is not related to genmon.The following URLs should go in the browser address field, it will show support info:  http://IPADDRESS:8000/cmd/support_data_jsonThe following command in the browser will download the logs to your local computer. http://IPADDRESS0:8000/cmd/get_logsThe text IPADDRESS above should be replaced with the IP address of your pi.

jgyates, 2022-06-05:
I got your logs via the submission on the about page. Looks like you may of had a password issue based on the info in your logs.

I am not sure how you are responding to the messages in this thread, but there is a lot of extra data that is included. I would suggest that you reply to github messages in a browser on a PC in at the github site.

Based on the data in your logs it looks like you have intermittent connection. Here are your comm stats:

Comm Stats : 
Packet Count : M: 112, S: 59
CRC Errors : 0 
CRC Percent Errors : 0.00%
Timeout Errors : 49
Timeout Percent Errors : 43.75%
Modbus Exceptions : 3
Validation Errors : 0
Sync Errors : 0
Invalid Data : 0
Discarded Bytes : 3
Comm Restarts : 2
Packets Per Second : 1.09
Average Transaction Time : 0.0329 sec

This shows that out of 112 attempted modbus reads, 59 of them were successful. The rest failed. Most of them (49) were timeouts. This is very likely a bad connection in your cables or connector. A bad ground connection could also cause this. I don't know of any other thing that could cause the communications to work sometimes, but fail at others.

lampe208, 2022-06-05:
genmon_logs/                                                                                        0000755 0000000 0000000 00000000000 14247227420 012067  5                                                                                                    ustar   root                            root                                                                                                                                                                                                                   genmon_logs/genexercise.log                                                                         0000644 0000000 0000000 00000005375 14247227420 015105  0                                                                                                    ustar   root                            root                                                                                                                                                                                                                   2021-11-20 18:08:55,516 : Execise: Quiet, Weekly at 12:0 on Friday for 12.0 min. Warmup: 0.0
2021-11-25 11:16:46,446 : Execise: Quiet, Weekly at 12:0 on Friday for 12.0 min. Warmup: 0.0
2021-11-26 16:43:54,321 : Execise: Quiet, Weekly at 12:0 on Friday for 12.0 min. Warmup: 0.0
2021-12-06 20:12:50,728 : Execise: Quiet, Weekly at 12:0 on Friday for 12.0 min. Warmup: 0.0
2021-12-17 12:09:05,183 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 0.0
2021-12-17 12:11:38,963 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-01-12 17:13:53,161 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-02-02 10:50:04,016 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-02-18 17:30:47,634 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-03-09 19:16:45,310 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-03-09 19:26:59,299 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-03-22 18:58:30,124 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-04-05 20:07:04,308 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-05-28 07:25:38,890 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-05-28 07:27:05,564 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-05-28 07:28:26,303 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-06-02 21:32:11,175 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-06-02 21:37:55,222 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-06-03 15:46:25,195 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-06-03 15:51:57,558 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-06-03 15:54:52,595 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-06-04 09:53:28,861 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-06-04 10:00:01,319 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-06-04 10:04:47,785 : Error: Connect : [Errno 111] Connection refused : myclient.py:47
2022-06-04 10:06:44,687 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-06-04 10:13:39,142 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-06-04 10:17:08,888 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-06-04 10:20:22,595 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-06-04 10:22:29,930 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
2022-06-04 11:02:33,714 : Execise: Normal, Weekly at 12:0 on Friday for 12.0 min. Warmup: 2.0
                                                                                                                                                                                                                                                                   genmon_logs/gensms_modem.log                                                                        0000644 0000000 0000000 00000026231 14247227420 015253  0                                                                                                    ustar   root                            root                                                                                                                                                                                                                   2022-05-28 07:28:26,253 : ->AT

2022-05-28 07:28:27,506 : ->AT

2022-05-28 07:28:29,759 : ->AT

2022-05-28 07:28:31,011 : ->AT

2022-05-28 07:28:33,265 : ->AT

2022-05-28 07:28:34,517 : ->AT

2022-05-28 07:28:36,771 : ->AT

2022-05-28 07:28:38,023 : ->AT

2022-05-28 07:28:40,277 : ->AT

2022-05-28 07:28:41,529 : ->AT

2022-05-28 07:28:43,783 : ->AT

2022-05-28 07:28:45,035 : ->AT

2022-05-28 07:28:47,289 : ->AT

2022-05-28 07:28:48,541 : ->AT

2022-05-28 07:28:50,795 : ->AT

2022-05-28 07:28:52,047 : ->AT

2022-05-28 07:28:53,310 : Unable to power up Modem.
2022-05-28 07:28:53,311 : Error starting device.
2022-05-28 07:28:53,339 : Modem Init FAILED!
2022-06-02 21:32:10,345 : ->AT

2022-06-02 21:32:11,598 : ->AT

2022-06-02 21:32:13,850 : ->AT
[... 2046 lines cut ...]
2022-06-05 17:02:24,902 : Error: timeout receiving slave packet for register 0011 Buffer: 0, sequence 24953
[... cut ...]
2022-06-05 17:03:52,654 : Error: timeout receiving slave packet for register 05f1 Buffer: 0, sequence 24973
2022-06-05 17:03:58,698 : Error: timeout receiving slave packet for register 001f Buffer: 0, sequence 24974
2022-06-05 17:04:01,726 : Error: timeout receiving slave packet for register 0020 Buffer: 0, sequence 24975
2022-06-05 17:04:04,755 : Error: timeout receiving slave packet for register 0021 Buffer: 0, sequence 24976
2022-06-05 17:04:07,780 : Error: timeout receiving slave packet for register 0022 Buffer: 0,

lampe208, 2022-06-05:
I have the ground directly to the battery ground.. never had any issues since i hooked up until lately.  I hope it is just the bad cable or pihat.  I will let you know.

skipfire, 2022-06-05:
@lampe208 the HAT can just be swapped out.

lampe208, 2022-06-05:
ok good

jgyates, 2022-06-05:
Also, in looking at your logs you appear to have a lot of add ons loaded (almost all of them). You even have the add on that uses a hardware LTE modem to send SMS messages. If you do not have this extra hardware needed to send LTE SMS, then I suggest you disable this add on and any others that you do not need. Enabling unneeded add ons will only reduce the life of your SD card. You have almost every Add on enabled and most of them do not appear to be configured properly. This generates errors for each add on that is not configured properly and these errors are written to the log files. Excessive log file writing will reduce the life of your SD card.

Do not load any add on that you do not need. If you are just using genmon and are notified via email then you don't need *any* add ons.

You have these add ons loaded and all of them are failing according to the log:

gengpio - external GPIO out - not needed unless you have external circuitry connected to your PI
gengpioin - external GPIO in  - not needed unless you have external circuitry connected to your PI
gensms - not needed unless you have Twilio account
gensms_modem - Not needed unless you have an additional LTE hard hardware with a SIM card loaded on the HAT
genslack - not needed unless you have a Slack account
genexercise - not needed unless you want to run a second exercise cycle (in additional to the one specified on the maint page)
genemail2sms - not needed unless you want to use your mobile carrier email to SMS converter 
gentankutil - not needed unless you have the hardware from tankutility.com and a valid tankutility.com account.
gentankdiy - not needed unless you have built your own gauge circuit per the wiki 
gentemp - not needed unless you have purchased and installed additional temp sensors
gencthat - not needed unless you have the CT HAT installed with two current transformers.

one of the first things I would try is to disable all of these on the the add on page and restart your pi. NOTE: disabling each one of these will require a restart of the software so you will need disable one, then save, disable another, then save, etc. until each add on is disabled.

lampe208, 2022-06-06:
Thank you , i do use twillo.  i do have tank utility account . will it give me a tank reading if i disable that?

jgyates, 2022-06-06:
Leave the ones that you are using enabled but disable the others. No, you will not get a tank utility reading if you disable the tank utility add on.

lampe208, 2022-06-06:
ok great

lampe208, 2022-06-06:
OK everybody I don't know what's going on but I disable the LTE SMS rebooted and now it's working fine I don't get it

jgyates, 2022-06-06:
That actually makes sense if the following thing happened (and your log files sort of point to this)

1) you enabled the LTE SMS add on (this is only useful if you have the additional hardware)
2) The LTE SMS add on communicates with the LTE Hat via the serial port. This should *NOT* be the same serial port that genmon uses. This is mentioned in the [wiki](https://github.com/jgyates/genmon/wiki/1----Software-Overview#gensms_modempy-optional). The default port for the LTE SMS add on is /dev/ttyAMA0 which is a USB based serial port. If you changed this to /dev/serial0 or the same port that is used for genmon then this would explain your issue. 

In summary, you told the LTE SMS add on to use the same serial port that genmon is using. You made two errors in your setup. 1) you enabled the LTE SMS add on without having the LTE HAT hardware installed and 2) you told the LTE SMS add on to use the same serial port that genmon is using.

Your log file for the LTE SMS add on shows it is trying to send commands to the LTE modem over the serial port so this means that you either have a USB serial port installed (the default values for the LTE SMS add on) or you changed the default values to match the built in serial port on the pi (the serial port genmon is using)

lampe208, 2022-06-06:
Great info. I disable alot of the add-ons you said to .  I'm so happy it's working  and you are a great asset. Thank you for your time! I guess I will have a spare pihat . But that's ok