jgyates/genmon #605: exercise status doesn’t match generator setting

jjmacneil, 2021-11-11:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

My Generac 20kw generator is set to bi-weekly exercise, but GenMon shows the status of weekly.

### Actual Behavior

GenMon should display the correct status based on current generator exercise setting.

### Steps to Reproduce (including precondition)

I confirmed bi-weekly exercise option was set via the generator controller, but GenMon status is displaying weekly.

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Generac 20kw
- Genmon Logs: Submit via the About page
- Genmon Version: 1.18.0 and 1.17.05 too

jgyates, 2021-11-11:
Hi @jjmacneil 

Look on the settings page under "Generator Model Specific Settings" for the setting "Enhanced Exercise Time" this allows the software to interpret bi-weekly, etc. Once enabled, the save the setting and it will restart with the correct exercise time. This is not enabled by default since this functionality was added with firmware updates and not all firmware version support this. Let me know if this does not fix the issue.

closed 2021-11-11 by jgyates

jjmacneil, 2021-11-11:
Thank you! Sorry, I should have checked that before opening a new issue. It’s working now.

jgyates, 2021-11-11:
no worries. I am glad it is working.