jgyates/genmon #608: Message level for SMS notifications

smeisner, 2021-11-16:
This is more of a feature request. I have successfully set up Twilio for SMS notifications, but could not find a way to indicate a severity level (such as warnings or fatal notifications) to be sent, like with the email notifications. I looked thru the Wiki but found no reference to this.

Thanks ...and Genmon is a great piece of software!!!! Thanks!!

### Expected Behavior

Ability to set a severity level for notifications sent via SMS.

### Actual Behavior

It's either all or nothing.

### Steps to Reproduce (including precondition)

Set up Twilio SMS notification via Genmon Web UI

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Generac 14Kw
- Genmon Logs: Not submitted since this is a UI request. If you really want/need them, let me know.
- Genmon Version: 1.18.00

jgyates, 2021-11-24:
This feature has been added to V1.18.01 see the release notes for more details. Let me know if you have any issues or questions,

https://github.com/jgyates/genmon/blob/master/changelog.md

smeisner, 2021-11-30:
Thanks! This looks great.

One question: the "Utility Service is Up" message is not considered Info (since I disabled Info level and this message is still being sent). Where are these severities set?

I like the RPi temp and wifi indicators too!

Steve

jgyates, 2021-11-30:
Hi @smeisner 

This message is included in the Outage category. This message occurs at the end of an outage. The Utility line state is also sent when genmon is started.

smeisner, 2021-12-01:
Yeah, I was seeing it when Genmon was being started.

 

OK, this is still a huge improvement from my point of view.

 

Thanks again!!