jgyates/genmon #609: Error in MyConfig: Additional Modbus Timeout and Warning: Reading Config File

yguy75, 2021-11-17:
Model : G0071760     16KW
Controller Detected :   Evolution 2.0, Air Cooled
Hardware Version :      V1.00
Firmware Version :       V1.12
Genmon Version:        V1.18.0

**Raspberry Pi-3B installed in house basement powered by UPS connected with about 40 feet of Thermostat 18 Gauge wire.**

**Issue  no. 1**

**Genmon.log     
Started Oct 25. – Before V1.18.0**
2021-10-25 15:13:10,925 : Error in MyConfig:ReadValue: GenMon: additional_modbus_timeout: could not convert string to float:  : myconfig.py:89
2021-10-25 15:13:11,358 : GenMon Loaded for site: Genmon-Pi3B- JJ using python 2.7
2021-10-27 21:11:44,492 : Generator Monitor Shutdown
2021-10-27 21:11:47,023 : Error in MyConfig:ReadValue: GenMon: additional_modbus_timeout: could not convert string to float:  : myconfig.py:89
2021-10-27 21:11:47,305 : GenMon Loaded for site: Genmon-Pi3B- JJ using python 2.7
2021-10-29 13:36:18,279 : Generator Monitor Shutdown
2021-10-29 13:36:20,835 : Error in MyConfig:ReadValue: GenMon: additional_modbus_timeout: could not convert string to float:  : myconfig.py:89

**Continued after  V1.18.0**

2021-11-09 22:17:25,935 : New version detected: Old = V1.17.04, New = V1.18.00
2021-11-09 22:17:25,953 : Error in MyConfig:ReadValue: GenMon: additional_modbus_timeout: could not convert string to float:  : myconfig.py:89
2021-11-09 22:17:25,967 : GenMon Loaded for site: Genmon-Pi3B- JJ using python 3.7
2021-11-11 18:27:30,933 : Generator Monitor Shutdown
2021-11-11 18:28:22,983 : Error in MyConfig:ReadValue: GenMon: additional_modbus_timeout: could not convert string to float:  : myconfig.py:89
2021-11-11 18:28:23,000 : GenMon Loaded for site: Genmon-Pi3B- JJ using python 3.7
2021-11-15 12:14:01,363 : Generator Monitor Shutdown
2021-11-15 12:14:05,400 : Error in MyConfig:ReadValue: GenMon: additional_modbus_timeout: could not convert string to float:  : myconfig.py:89

**Issue  No. 2
Genserv.log
Happened Before and after V1.18.0
Seems to happen soon after a login to the Genmon Web Interface**
2021-11-09 22:22:38,839 : Admin Login
2021-11-10 14:35:28,138 : Warning: Reading Config File (value not in list): controllertype : 
2021-11-10 14:37:03,484 : Warning: Reading Config File (value not in list): controllertype : 
2021-11-10 16:35:15,957 : Admin Login
2021-11-11 11:05:08,944 : Warning: Reading Config File (value not in list): controllertype : 
2021-11-11 11:57:36,162 : Admin Login
2021-11-11 11:59:04,859 : Warning: Reading Config File (value not in list): controllertype : 
2021-11-11 18:27:23,886 : Warning: Reading Config File (value not in list): controllertype : 
2021-11-11 18:28:28,129 : Starting /home/pi/genmon/genserv.py, Port:443, Secure HTTP: True, SelfSignedCert: True

2021-11-15 12:14:48,741 : Admin Login
2021-11-15 12:16:26,193 : Warning: Reading Config File (value not in list): controllertype : 
2021-11-15 12:19:56,782 : Starting /home/pi/genmon/genserv.py, Port:443, Secure HTTP: True, SelfSignedCert: True
2021-11-15 12:20:20,668 : Admin Login

**No warnings for past 2 days**

**Issue  #3
Mymodbus.log
Modbus timeout usually when Generator starts exercising. (Sometimes random)**
2021-11-15 10:57:34,902 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-11-15 10:57:34,903 : Error Receiving slave packet for register 0007
2021-11-15 11:00:35,096 : Error: timeout receiving slave packet for register 05f5 Buffer: 0
2021-11-15 11:10:57,933 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-11-15 11:10:57,935 : Error Receiving slave packet for register 0001
Back to 8 minutes
2021-11-15 11:19:11,566 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2021-11-15 11:19:11,568 : Error Receiving slave packet for register 0052

**For Information Only
Genloader.log
Warnings while Installing V1.18.0**
2021-11-09 22:14:51,394 : Warning: required library crcmod not installed. Attempting to install....
2021-11-09 22:15:12,126 : Warning: required library pyowm not installed. Attempting to install....
2021-11-09 22:15:28,364 : Warning: required library pytz not installed. Attempting to install....
2021-11-09 22:15:40,881 : Warning: required library pysnmp not installed. Attempting to install....
2021-11-09 22:16:04,668 : Warning: required library ldap3 not installed. Attempting to install....
2021-11-09 22:16:25,971 : Warning: required library pyotp not installed. Attempting to install....
2021-11-09 22:16:36,944 : Warning: required library chump not installed. Attempting to install....
2021-11-09 22:16:47,232 : Warning: required library twilio not installed. Attempting to install....
2021-11-09 22:17:08,358 : Warning: required library paho-mqtt not installed. Attempting to install....

**These warning went away after restarting the Pi.**

Submitted Logs and Registers

jgyates, 2021-11-17:
Hi @yguy75 

None of these are reasons to be concerned and these are in the log files for support reasons. Here is the background on each of these:

The error:

      MyConfig:ReadValue: GenMon:

means that an entry was read but the value is either empty or an invalid type. In this case ' additional_modbus_timeout' is an optional value that is set on the advanced page. The default value of this is 

     additional_modbus_timeout = 0.0

This warning means that the value read is not a floating point value. If the value is 0.0. or any other value it is ignored by genmon so other than the entry in your log file this does not change the way genmon works.

The same is true for the error:

     Warning: Reading Config File (value not in list): controllertype :

'controllertype' is blank by default and then evolution / nexus is assumed for the controller type. 

For this error:

     Modbus Exception: Acknowledge: 05 , Modbus Command: 83

The line immediately following this is the same condition "Error Receiving slave packet for register 0007". We have covered this before. This is an anomaly of the Evolution 2.0 firmware. See this thread:

     https://github.com/jgyates/genmon/issues/539

For genloader.log:

       Warning: required library XXXXX not installed. Attempting to install....

Where XXXXX is a python library required by genmon.  Genmon V1.18.00 stopped using python 2.x and started using 3.x by default. When you initially installed genmon part of the setup was to install needed libraries. These libraries were python 2.x compatible. For python 3.x you need to have the python 3.x versions of these libraries installed. Instead of making everyone reinstall genmon the loader is installing the missing python 3 libraries so these warnings are expected if you upgraded genon to V1.18.00. These warnings are put in the logs for support reasons.

I got you logs. Everything looks great.

I hope this helps. I am going to close this thread but feel free to post here if you have additional questions on this topic.

closed 2021-11-17 by jgyates

yguy75, 2021-11-17:
Thank you very much for clarification.

Now we can sleep better at night.