jgyates/genmon #611: cant log in local pi

lampe208, 2021-11-25:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

lampe208, 2021-11-25:
i did last update it restarted . looked at new update logging in to web browser like i always have.  Got a warning on the bottom of the program of being offline and errors   cant log in ever since .  times out  . 
My network shows up online with same IP as before but it won’t load

lampe208, 2021-11-25:
pi is online on my network but i cant reach it in my browser as before https://192.168.1.181

jgyates, 2021-11-26:
You issues is likely the same one mentioned here:

https://github.com/jgyates/genmon/issues/598

and here:

https://github.com/jgyates/genmon/issues/603

Please review these threads and let me know if you still have questions. 

You need to log in to your pi via ssh.

Once you log in the solution will depend on what version of the Raspbian operating system you are using on you pi. This command will tell you if you are running the "Jessie" version.

     cat /etc/os-release

If you are the the above thread 603 will tell you how to resolve the issue.

If you still need help after reading this let me know.

lampe208, 2021-11-26:
i will try this but after it updated and restarted i saw the new version and was using it  .  then after a few minutes it died and now j can’t reach the server

jgyates, 2021-11-26:
You can look at the logs for genmon and genloader by doing this on the command line:

      cat /var/log/genmon.log

and 

     cat /var/log/genloader.log

lampe208, 2021-11-26:
Here is the log .. what is your recommendation ?

2021-11-25 11:06:19,191 : Error in GetObservation: (50125) : Exception in calling OWM Weather API.
Reason: HTTPConnectionPool(host='api.openweathermap.org', port=80): Max retries exceeded with url: /data/2.5/weather?APPID=6a7e5972d6ec8c94fe938549de546406&zip=50125%2Cus&lang=en (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x6b0798d0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))
Caused by: None : myweather.py:106
2021-11-25 11:16:37,388 : Generator Monitor Shutdown
2021-11-25 11:16:40,651 : New version detected: Old = V1.18.00, New = V1.18.01
2021-11-25 11:16:42,051 : GenMon Loaded for site: Lampe Generac using python 3.7
2021-11-25 11:27:30,883 : Error in GetWiFiSignalStrength: invalid literal for int() with base 10: '' : myplatform.py:308
2021-11-25 11:27:33,438 : Error in GetWiFiSignalStrength: invalid literal for int() with base 10: '' : myplatform.py:308
2021-11-25 11:27:36,512 : Error in GetWiFiSignalStrength: invalid literal for int() with base 10: '' : myplatform.py:308
2021-11-25 11:27:42,475 : Error in GetWiFiSignalStrength: invalid literal for int() with base 10: '' : myplatform.py:308
2021-11-25 11:27:43,191 : Error in GetWiFiSignalStrength: invalid literal for int() with base 10: '' : myplatform.py:308
2021-11-25 11:27:46,678 : Error in GetWiFiSignalStrength: invalid literal for int() with base 10: '' : myplatform.py:308
2021-11-25 11:27:48,497 : Error in GetWiFiSignalStrength: invalid literal for int() with base 10: '' : myplatform.py:308
2021-11-25 11:36:48,282 : Error in GetObservation: (50125) : Exception in calling OWM Weather API.
Reason: HTTPConnectionPool(host='api.openweathermap.org', port=80): Max retries exceeded with url: /data/2.5/weather?APPID=6a7e5972d6ec8c94fe938549de546406&zip=50125%2Cus&lang=en (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x6f073dd0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))
Caused by: None : myweather.py:106
2021-11-26 06:29:59,870 : Error in GetObservation: (50125) : Exception in calling OWM Weather API.
Reason: API call timeouted
Caused by: None : myweather.py:106
2021-11-26 08:21:18,429 : Error in GetObservation: (50125) : Exception in calling OWM Weather API.
Reason: HTTPConnectionPool(host='api.openweathermap.org', port=80): Max retries exceeded with url: /data/2.5/weather?APPID=6a7e5972d6ec8c94fe938549de546406&zip=50125%2Cus&lang=en (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x6b826e70>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))
Caused by: None : myweather.py:106
2021-11-26 11:22:48,777 : Error in GetObservation: (50125) : Exception in calling OWM Weather API.
Reason: API call timeouted
Caused by: None : myweather.py:106
pi@raspberrypi:~ $

lampe208, 2021-11-26:
Never mind i just unplugged and rebooted... now its accessbile .. thanks for your attention

closed 2021-11-26 by lampe208

lampe208, 2021-11-26:
Any reason why simply rebooting would solve this/

jgyates, 2021-11-26:
it could one of many reasons. just as you have to reboot personal computers and phones periodically due to software issues this can occur with linux. If you want to send me your logs from the About page I can take a look.

lampe208, 2021-11-26:
how do i do that remotely>  What command

jgyates, 2021-11-26:
pull up the web interface, go to the about page and there are buttons on that page to send log files.

from the command line you can type:

     cd genmon
     python3 ClientInterface.py
     

at the prompt type:

     generator: sendlogfiles

To exit the ClientInterface.py program type:

     exit

Sending log files requires outbound email to be functional in the genmon setup.

lampe208, 2021-11-27:
Ok I sent it.

jgyates, 2021-11-27:
I got you logs. The only thing that I see that is a problem is some error messages from the weather API failing due to a timeout. From the softwares perspective it can not access the internet. Everything else looked fine.

lampe208, 2021-11-27:
Super, I noticed that as well.  Thats good news.  Its working perfectly now
after reboot.  Thanks so much for adding that temp sensor and wifi monitor
gauges.