jgyates/genmon #612: Service Log not updating.

RoyRomeo, 2021-12-02:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Service Journal updated.

### Actual Behavior
Service Log not updating.
{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2021-12-02:
@RoyRomeo 

I will need more info. This is working for me so at a minimum I will need:

- you log files
- What version of genmon are you using
- what browser and operating system you are using to host the browser.
- more details on what exactly is not working, can you not save the log or does it appear to save but when you revist the service journal it is does not show the last entry. Just saying it does not work is not descriptive enough to give me any clue as to the problem you are seeing.

Thanks
Jason

RoyRomeo, 2021-12-02:
Hi Jason,
I submitted the logs.Genmon version V1.18.01Browser: Google Chrome/OS Windows 10
When I make an entry in the "Service Journal" and save it, the entry remains.  However when I go to the "Logs" tab  the "Service Log" entries are not updating.

Thanks,Joe

jgyates, 2021-12-02:
I got you logs. I think the problem is one of expectations. The "Logs" tab contains the logs generated by the generator controller. These logs are not editable by the user. The "Service Journal" tab only contains info entered by the user. The data on the "Service Journal" page will not show up on the "Logs" page.

The "Logs" -> Service Log contains things like service due alarms with date/time and when the controller service  timers were reset from the controller front panel (with time and date).

The "Service Journal" is extra info that a user may wish to log, like did you change the air filter and what type of oil did you use.

Also, a quick look at your logs shows one issue with your open weather API key. You keep getting this error:

2021-12-01 21:56:22,165 : Error in GetObservation : Your API subscription level does not allow to perform this operation. Reason: Invalid API Key provided 

You may want to double check the API key you provide on the setting page.

Let me know if you have any other questions.

closed 2021-12-02 by jgyates

RoyRomeo, 2021-12-02:
Hi Jason Thanks for the clarification. I didn't realize how the Logs worked.You worked with me in the past on trying to set up the Weather API, and if my memory serves me correctly, you determined that I was not able to make it work using Google Chrome/Windows 10.Thanks again,JoeSent from my T-Mobile 5G Device

Closed #612.

jgyates, 2021-12-02:
I dont know of any open weather API issues on this project.  The errors in your log are occurring on the pi, not windows 10.

RoyRomeo, 2021-12-02:
Hi Jason,
Thanks for your help.  It was back on issue #465 where you tried to get my weather API up and running.
Thanks again.

jgyates, 2021-12-02:
These are two separate issues. 

The issue in #465 is related to your browser settings not letting the field on the settings page auto fill. The issue in your logs is not related to this. The log error is stating that your openweathermap API key is invalid.

I would either correct this or disable the weather API on the settings page. These repeated errors logged to the SD card will eventually cause your SD card to fail at a faster rate that it normally would.

RoyRomeo, 2021-12-02:
Thanks Jason, I will take care of that.Sent from my T-Mobile 5G Device