jgyates/genmon #613: Firmware Release 1.20 NO COMMS to Genmon

fdcaptjd, 2021-12-03:
Hello,

My 2016  22KW unit has just been had a stator replaced under the extended warranty by a dealer. 

I have been told this also required a new controller.

This new controller is running 1.20 firmware.

Genmon is not communicating at this point. I have attached a screen shot of the stats.

Thanks,

Jeff
[image]

jgyates, 2021-12-03:
I have many reports of firmware 1.20 working. I am guessing that this is either a connection problem or a defective controller.   The timeout errors indicate that either the packets sent by the pi are not being received or the response packets sent by the controller are not being received by the pi.

Make sure that your cable is attached to the new controller properly. If you can do that then try the loopback test to validate you pi is still functioning as expected. 

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

I once had a controller replaced and the company doing the service initially tried to replace it with a controller with defective comms (the box was marked "no comms" with a sharpie). After I pointed this out they replaced it with another one. Of course your mileage may vary with another service tech.

fdcaptjd, 2021-12-03:
Thank you for the quick response, and all the work you do on this project. 

Jeff

grsthegreat, 2021-12-04:
that unit uses the evolution 1.0 or sync 2.0 controller (if labeled honeywell). the v1.20 software will not effect genmon operation.

take a look and see which of the 2 ports its plugged into, there are 2 accessory ports. labeled port 1 and port 2

fdcaptjd, 2021-12-04:
Thanks for the info.

I did the loopback test yesterday and believe the problem is a failed TTL/235 convertor. Got good tests on the Pi, nothing after the convertor. 

JD

jgyates, 2021-12-06:
I am going to close this issue for now but feel free to post here with any status updates or questions.

closed 2021-12-06 by jgyates