jgyates/genmon #614: Evolution 2.p 24kw new install everything seems to work but set exercise time.

tr6scott, 2021-12-06:
### Expected Behavior

Set Exercise time. UI is working by setting, but when you refresh web page, old data is unchanged. 

### Actual Behavior

UI works, but the data dose not stick on the generator. 

### Steps to Reproduce (including precondition)

Try to set a new excercise time from maintenance page, I can set a new time, click set exercise time button, opens prompt to set the new time, click ok, the time changes. refresh the page, the old data is still in generator. 

### Screenshot or Pictures relating to the problem (if possible)
[image]

### Your Environment
Maintenance :
    Model : Generic Air Cooled
    Generator Serial Number : 3009736750
    Controller Detected : Evolution 2.0, Air Cooled
    Nominal RPM : 3600
    Rated kW : 24 kW
    Nominal Frequency : 60 Hz
    Fuel Type : Natural Gas
    Engine Displacement : 999 cc
    Ambient Temperature Sensor : 45 F
    kW Hours in last 30 days : 0.00 kWh
    Fuel Consumption in last 30 days : 0.56 cubic feet
    Total Power Log Fuel Consumption : 0.56 cubic feet
    Run Hours in last 30 days : 0.01 h

    Controller Settings :
        Calibrate Current 1 : 1492
        Calibrate Current 2 : 1549
        Calibrate Volts : 1049
        Nominal Line Voltage : 240 V
        Rated Max Power : 24 kW

    Exercise :
        Exercise Time : Weekly Tuesday 10:00 Quiet Mode Off

    Service :
        Service A Due : 199 hrs or 11/26/2023
        Service B Due : 399 hrs or 11/25/2025
        Battery Check Due : 11/26/2022
        Total Run Hours : 0.00 h

        Hardware Version : V1.01
        Firmware Version : V1.12

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: 18.01

jgyates, 2021-12-06:
@tr6scott 

thanks for the report. I got your logs. This is a bug. I checked in V1.18.02 that corrects this issue. This is related to the python 3 update that occured with V.18.0. Let me know if you have any issues once you upgrade to V1.18.03 via the About page.

closed 2021-12-06 by jgyates

tr6scott, 2021-12-06:
Updated to 1.18.02 and was able to set exercise time correctly.

Also when I was reviewing the issues and troubleshooting pages before I submitted, There was a note that firefox browser had issues on windows. I have only used firefox on windows, and have not experienced any issues. Just a datapoint. 

Let me know if you need any feedback in the future with my setup/debug. I am not big in linux, or python, but a controls engineer by trade, and able to figure my way around following webpages and such. 

Thanks for the effort, great code, and followup.

MiddleSiggy, 2021-12-06:
tr6scott, can you share how you connected to the Evolution 2 connector inside the generator? I am unsure what connector is the correct one. On the back of mine is a Port 1 and Port 2, Port 2 had a sticker covering it, and no connector soldered onto the board, port one has a male molex connector plugged into the female connector attached to the board, with 4 of the 8 connectors pinned. Did you just remove the connector? and connect into that port? or did you tap into pin 7 & 8 for tx and rx?

I am thinking of using this cable:
https://www.amazon.com/gp/product/B000M802RG/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1

and then splicing the tx and rx to pins 7 & 8 and leaving the wifi connector connected. If this is the connector.

thank you

jgyates, 2021-12-07:
This thread has some info on this topic

https://github.com/jgyates/genmon/issues/616