jgyates/genmon #615: Update Issue

RoyRomeo, 2021-12-06:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Genmon update and restart.

### Actual Behavior

Update takes a long time and monitor does not restart after update.  Problem began several updates ago.  After update I have to Putty in and type "cd genmon" then "./startgenmon.sh -p 2 restart".  The monitor will then start and work normally until the next update.  Any idea as to what could be causing this issue?

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: G0062440
- Genmon Logs: Submitted
- Genmon Version: V1.18.02

jgyates, 2021-12-06:
I got your logs. I would try rebooting your pi. Your logs indicate that your pi has been running for 318 days. While this is not a problem, this issue could be that some piece of software running needs to be reset.

RoyRomeo, 2021-12-06:
Okay.  What is the best way to reboot the pi.

RoyRomeo, 2021-12-06:
Is it the "Reboot System" Reboot button?

jgyates, 2021-12-06:
yes

RoyRomeo, 2021-12-06:
I tried that but now the monitor will not start and I cannot Putty in via the IP.

RoyRomeo, 2021-12-06:
Okay, got it running again.  Is it necessary to hit the save button on the bottom of the advanced settings page?

jgyates, 2021-12-06:
If you are changing settings, yes. If you are rebooting or restarting the software with the buttons, no.

jgyates, 2021-12-06:
I am going to close this issue but let me know if you have questions on this topic via this thread.

closed 2021-12-06 by jgyates

RoyRomeo, 2021-12-06:
Okay, thanks Jason.