jgyates/genmon #616: Molex Cable connector for Evolution 2.0 unit with wifi

MiddleSiggy, 2021-12-06:
I was looking for some help to expand on your opinion around the proper way to connect the genmon to my new installation of a generac 24kw unit model G0072101. 

It has the Evolution 2.0 wifi module in it from the manufacturer already, and I believe I found the connector on the back side of the user interface.

So I have 2 questions, because I do not have the diagnostic manual for this model, I assume the back side of the screen, which has a 8 pin molex connector with only 4 of the pins connected is the right port to connect to. 

Reading your directions on the wiki here:
https://github.com/jgyates/genmon/wiki/3.1--Making-a-Cable

It states that I have 2 options, modify the existing connector and add in pins 7 & 8 which are not used, or remove the cable and connect only pins 7 & 8 to genmon and the slot labeled Port 1. 

I was considering using this cable which has a male and female 8 pin molex connector:
https://www.amazon.com/gp/product/B000M802RG/ref=ppx_yo_dt_b_asin_title_o05_s01?ie=UTF8&psc=1

Then tap into the wire for pin 7 & 8 as in the directions on the previous "Making a Cable" page, and then plug the existing connector into the female side of the wire, this way I would not have to modify any cabled but instead could choose to connect or disconnect the cable should I wish.

So first, do you know if this is the correct port? and how would you go about connecting to the generator? would you leave the wifi module unplugged?

Thank yo
[image]
u
[image]
.

skipfire, 2021-12-06:
Where the wifi plugs in is the correct spot.  You have a few options, you can make your own cable to connect and if you want you can splice in the wifi when doing so.  Or you can leave the wifi disconnected, which is how I have mine since the generac app being so unhelpful is why I installed Genmon in the first place.  Or you can get one of the HATs from PintSize.Me (disclosure - that's me) and the HATs have a 4-pin passthrough connector that the generac wifi can be plugged into or left vacant.

gaylordw, 2021-12-06:
I have the same setup you are describing and it works fine.  I cut the wires 7 and 8 on the extension cable and connected them to the genmon.  The cable that was connected to the generac wifi module was then connected to the female plug on the new cable.  The genmon and the generac wifi module both work fine when both are connected. The generac app still connects to the original wifi module.

johnsdiyplayground, 2021-12-07:
I would disable the wifi on the generator itself.  Because as we've seen
last year, Generac will push out firmware updates without warning that add
security to their module and make Genmon a paperweight (for a little while,
at least).  I will never turn my generator's wifi on ever again after
learning that lesson.

grsthegreat, 2021-12-07:
your port 2 doesnt even have the molex connector installed. i believe your just looking at the circuit board. unplug port1 which is the wifi for the generator, and use that port.

jgyates, 2021-12-09:
I am going to close this issue but free fee to post here with follow up questions / status

closed 2021-12-09 by jgyates

dano-code, 2022-03-04:
Can multiple devices share the same port? My old 2011 generator had its oil pump fail, so it was either new engine or new generator. I decided to get a new generator, and will go from a 20kW to 22kW (I assume) Evolution controller. It comes with wifi and maybe I don't want theirs connected, but if I do, can they both share that port 1 if port 2 does not have a connector on it?

danorum, 2022-03-04:
Can multiple devices share the same port? My old 2011 generator had its oil pump fail, so it was either new engine or new generator. I decided to get a new generator, and will go from a 20kW to 22kW (I assume) Evolution controller. It comes with wifi and maybe I don't want theirs connected, but if I do, can they both share that port 1 if port 2 does not have a connector on it?

skipfire, 2022-03-04:
@danorum physically no, electrically yes.  The Generac Mobilink WiFi unit uses 2 data pins and GenMon uses 2 different data pins, however they are all in the same physical connector.  The PintSize.Me HATs (disclosure, I make those) do have a passthrough for the WiFi pins, and I am very soon going to have a splitter available on my site that can be used without my HATs.

danorum, 2022-03-04:
OK.. Electrically was my question based on splitting out for genmon above. That makes sense. I wasn't thinking the wifi module used different pins than genmon. Not sure if it would be worth shipping vs me picking up a few molex connectors and making a Y, but I will take a look. New generator isn't until next week, and I probably don't want their wifi anyway.

skipfire, 2022-03-04:
Effectively it is a left half/right half situation with one side being RS232 with 12v for GenMon, and the other side being RS485 with 5v for the WiFi module.  If you have the stuff and are comfortable doing so, by all means make your own cables, it would certainly be cheaper.  Especially since you already have whatever hardware you need for connecting to the generator already in place.

That said, I think the Generac software sucks (just doesn't provide any data I want).  Then they added encryption to the firmware awhile back so I keep my wifi module disconnected because I don't want to risk a new encryption that permanently breaks genmon.

danorum, 2022-03-04:
If I was not already farther along on my journey, I would probably buy your stuff as it looks really nice. The hard part right now would be finding Raspberry Pi's. I built 3 of these myself for family and friends with Pi Zero's with ethernet adapters attached, but with your HATs, I could have been done so much faster.

griffths, 2022-03-04:
Dan,

Just throwing this out there. 

I am planning to purchase one of Jons sensor HATs so maybe we can “group buy “ and split shipping cost if that is your hesitation then we will have an excuse to meet for lunch somewhere since we haven’t done that for a long time.  

Jay