jgyates/genmon #617: Can't open genmon page on the PI

jcherney, 2021-12-07:
### Expected Behavior
Login and see a genmon web page. 

### Actual Behavior
No connection according to web browser from 3 different browsers. Fixed IP address can be pinged. I can SSH in with no issues. I'm not that familiar with SSH to go further than reboot or shutdown. Both Xfinity and external programs show the PI online. 

### Screenshot or Pictures relating to the problem (if possible)

### Your Environment

- Generator Model: QT04842ANAX
- Genmon Logs: Submit via the About page  CAN'T LOG IN TO SUBMIT
- Generator Model: 
- Genmon Version: LATEST this morning. At least I think so.
- Genmon Version: 

It started acting like this, this morning after the install of 1.18.02. Made a note to check it when I got home.  Today we lost power (of course!) and I have no access to genmon. I went outside and pulled the power plug first. It rebooted (I think). I did an SSH reboot also. Nothing changed
Weird fact: Wife got the email outage notification at around 4 pm when the power went out- I never did. 

Any suggestions would be appreciated, thanks.

jgyates, 2021-12-07:
if you power cycled the pi and you can't access the web interface then I would suggest validating your IP address you are using to access the pi via the browser and make sure it matches the ssh IP address. 

If you can log in via ssh then you should be able to issue this command:

       cd genmon
       ./startgenmon.sh restart

Note what is displayed on the scene after the above command is entered. Copy and paste that here.

This should restart genmon. You should get an email that genmin is starting. you  an also do this from the genmon folder:

       python ClientInterface.py

Then type:

       generator: sendlogfiles
        exit

This will send you log files.

jcherney, 2021-12-07:
pi@genmon [~] cd genmon
pi@genmon [~/genmon] ./startgenmon.sh restart
Restarting genmon python scripts
Stopping....
Stopping genserv.py
kill: (795): No such process
Stopping genmon.py
kill: (766): No such process
Starting....
Starting /home/pi/genmon/genmon.py
Starting /home/pi/genmon/genserv.py

No email ever came.

jcherney, 2021-12-07:
After the python ClientInterface.py I got a genmon not loaded response.

jgyates, 2021-12-07:
Try this 

      cd genmon
      ./genmonmaint.sh -u

Type 'y' for yes to update

       ./startgenmon.sh restart

Then give it a bout 15-30  seconds and see if it is working.

If that does not work type:

       cat /var/log/genmon.log

And paste the last 20 lines here.

jcherney, 2021-12-07:
still not working. Here are the last bunch of lines:

olutionController now is False
2021-11-25 08:02:39,754 : Generator Monitor Shutdown
2021-11-25 08:02:44,192 : New version detected: Old = V1.18.00, New = V1.18.01
2021-11-25 08:02:44,803 : GenMon Loaded for site: Harlan using python 3.5
2021-11-25 08:02:44,861 : DetectController auto-detect override (controller). Ev                  olutionController now is False
2021-12-06 08:38:03,572 : Generator Monitor Shutdown
2021-12-06 08:38:08,220 : New version detected: Old = V1.18.01, New = V1.18.02
2021-12-06 08:38:08,239 : Error opening modbus device: Error opening modbus devi                  ce. : generac_evolution.py:248
2021-12-06 08:38:08,239 : Possible cause is invalid serial port specified.
2021-12-06 18:17:40,393 : Error opening modbus device: Error opening modbus devi                  ce. : generac_evolution.py:248
2021-12-06 18:17:40,395 : Possible cause is invalid serial port specified.
2021-12-06 19:00:04,781 : Error opening modbus device: Error opening modbus devi                  ce. : generac_evolution.py:248
2021-12-06 19:00:04,783 : Possible cause is invalid serial port specified.
2021-12-06 19:35:58,357 : Error opening modbus device: Error opening modbus devi                  ce. : generac_evolution.py:248
2021-12-06 19:35:58,358 : Possible cause is invalid serial port specified.
2021-12-06 19:41:01,637 : Error opening modbus device: Error opening modbus devi                  ce. : generac_evolution.py:248
2021-12-06 19:41:01,638 : Possible cause is invalid serial port specified.
2021-12-06 21:12:40,320 : Error opening modbus device: Error opening modbus devi                  ce. : generac_evolution.py:248
2021-12-06 21:12:40,321 : Possible cause is invalid serial port specified.

jgyates, 2021-12-07:
Now type:

      cat /var/log/myserial.log

and send the last 20 lines.

jgyates, 2021-12-07:
also, what is the output of this command:

     pip3 show pyserial

jcherney, 2021-12-07:
2018-10-29 18:34:16,096 : Error in SerialDevice:Flush : /dev/serial0:Attempting to use a port that is not open : myserial.py:146
2018-10-29 18:34:39,680 : Error in SerialDevice:Flush : /dev/serial0:Attempting to use a port that is not open : myserial.py:146
2018-10-29 18:39:44,060 : Error in SerialDevice:Flush : /dev/serial0:Attempting to use a port that is not open : myserial.py:146
2018-10-29 18:40:07,431 : Error in SerialDevice:Flush : /dev/serial0:Attempting to use a port that is not open : myserial.py:146
2018-10-29 18:40:15,247 : Error in SerialDevice:Flush : /dev/serial0:(9, 'Bad file descriptor') : myserial.py:146
2018-10-29 18:58:26,923 : Error in SerialDevice:Flush : /dev/serial0:Attempting to use a port that is not open : myserial.py:146
2018-10-29 18:58:42,602 : Error in SerialDevice:Flush : /dev/serial0:(9, 'Bad file descriptor') : myserial.py:146
2018-10-29 18:59:06,011 : Error in SerialDevice:Flush : /dev/serial0:Attempting to use a port that is not open : myserial.py:146
2018-10-29 18:59:13,835 : Error in SerialDevice:Flush : /dev/serial0:Attempting to use a port that is not open : myserial.py:146
2020-05-21 09:40:35,303 : Error in SerialDevice:Flush : /dev/serial0:Attempting to use a port that is not open : myserial.py:165

Name: pyserial
Version: 3.2.1
Summary: Python Serial Port Extension
Home-page: https://github.com/pyserial/pyserial
Author: Chris Liechti
Author-email: cliechti@gmx.net
License: BSD
Location: /usr/lib/python3/dist-packages
Requires:
pi@genmon [~/genmon]

jgyates, 2021-12-07:
Lets try this:

      ./startgenmon.sh stop
      sudo python3 genmon.py

This command will start genmon.py and only genmon.py. Post the output of the screen here.

If genmon does load you will need to type Ctrl+C to stop it as this will load genmon in the current console context so you can run anything else until you kill genmon with Ctrl+C.

The main thing that changed in V1.18.02 is that genmon now requires exclusive access to the serial port. If some other program is using the port it will fail.

jcherney, 2021-12-07:
pi@genmon [~/genmon] sudo python3 genmon.py
Error opening modbus device.

jgyates, 2021-12-07:
Now let's see the 20 lines of this:

     cat /var/log/mymodbus.log

jcherney, 2021-12-07:
2021-04-10 14:19:37,721 : Error: timeout receiving slave packet for register 020a Buffer: 0
2021-04-13 18:47:29,628 : Error: timeout receiving slave packet for register 0052 Buffer: 0
2021-04-17 14:26:04,153 : Error: timeout receiving slave packet for register 002d Buffer: 0
2021-05-02 16:30:25,684 : Error: timeout receiving slave packet for register 0246 Buffer: 0
2021-05-06 09:42:54,733 : Error: timeout receiving slave packet for register 0053 Buffer: 0
2021-12-06 08:38:08,238 : Error opening modbus device: unexpected keyword arguments: {'exclusive': True} : mymodbus.py:69
2021-12-06 08:38:08,238 : Error opening modbus device.
2021-12-06 18:17:40,388 : Error opening modbus device: unexpected keyword arguments: {'exclusive': True} : mymodbus.py:69
2021-12-06 18:17:40,392 : Error opening modbus device.
2021-12-06 19:00:04,779 : Error opening modbus device: unexpected keyword arguments: {'exclusive': True} : mymodbus.py:69
2021-12-06 19:00:04,780 : Error opening modbus device.
2021-12-06 19:35:58,356 : Error opening modbus device: unexpected keyword arguments: {'exclusive': True} : mymodbus.py:69
2021-12-06 19:35:58,356 : Error opening modbus device.
2021-12-06 19:41:01,636 : Error opening modbus device: unexpected keyword arguments: {'exclusive': True} : mymodbus.py:69
2021-12-06 19:41:01,637 : Error opening modbus device.
2021-12-06 21:12:40,319 : Error opening modbus device: unexpected keyword arguments: {'exclusive': True} : mymodbus.py:69
2021-12-06 21:12:40,320 : Error opening modbus device.
2021-12-06 21:32:03,980 : Error opening modbus device: unexpected keyword arguments: {'exclusive': True} : mymodbus.py:69
2021-12-06 21:32:03,980 : Error opening modbus device.
2021-12-06 21:33:18,334 : Error opening modbus device: unexpected keyword arguments: {'exclusive': True} : mymodbus.py:69
2021-12-06 21:33:18,335 : Error opening modbus device.

jgyates, 2021-12-07:
There is the problem.

It looks like The new code I put in is not compatible with older versions of pyserial. I will code a fix for this but in the meantime try this:

       ./genmonmaint.sh -r
   

wait for it to compete, then type:

        ./startgenmon.sh restart

jcherney, 2021-12-07:
I'm back! Thank you so much for your help w this!

jgyates, 2021-12-07:
I got your logs, thanks!

I checked in version 1.18.03 that should fix this if anyone else has issues.

closed 2021-12-07 by jgyates