jgyates/genmon #618: Confirmation that Genmon works on Pi Zero 2

bdwilson, 2021-12-07:
I wanted to confirm that genmon works great on Pi Zero 2. I upgraded from Pi 2 b+ and it is much faster/responsive.  I'm also powering the unit from the Generator not direct from battery and haven't experienced any issues.  I did power tests using a USB power meter and the power usage for Pi Zero 2 is almost the same as Pi 2b+.   I'm also running an extra USB->SD reader so I can run rpi-clone once a week to ensure I have a backup if my primary SD card dies. 

The only thing I believe I had to change to make it work was add this to 
/boot/config.txt as the options have changed:

dtoverlay=disable-bt
dtoverlay=disable-wifi  (I'm using external Wifi adapter).

[image]

jgyates, 2021-12-07:
thanks for the info! Is that a USB hub and additional SD card in the pic?

Also, can you share the USB Wifi adapter you are using? I have not seen that one.

bdwilson, 2021-12-07:
Yes.  The 5ghz adapter was the cheapest 5ghz adapter at the time. Funny that I only connect at 2.4 even though I bought this one. It is *not* supported out of the box (see below for drivers).

* USB Hub: https://www.amazon.com/MakerSpot-Accessories-Charging-Extension-Raspberry/dp/B01JL837X8
* 2.4/5ghz adapter: https://www.amazon.com/gp/product/B018TX8IDA
* SMA Cable: https://www.amazon.com/gp/product/B0085X3PRE
* Serial to Terminal Module: https://www.amazon.com/gp/product/B016PUM9IO
* TTL Converter: https://www.amazon.com/gp/product/B00EJ9NAKA (what I purchased.. what looks to be compatible since this is OOS. https://www.amazon.com/HiLetgo-MAX3232-Converter-Module-Serial/dp/B00LPK0Z9A
* Cables: https://www.amazon.com/Fpz-Adapter-Female-Extension-Converter/dp/B085S5CKZN

Driver for Wireless card: https://github.com/brektrou/rtl8821CU
You'll need to edit Makefile for this and add support for Raspi:

```
###################### Platform Related #######################
CONFIG_PLATFORM_I386_PC = n
CONFIG_PLATFORM_ARM_RPI = y
CONFIG_PLATFORM_ARM64_RPI = n
```
Install kernel headers: **apt-get install raspberrypi-kernel-headers**

Then:  ** pi@genmon:~/git/rtl8821CU $ sudo ./dkms-install.sh ** 

If you run into errors, you'll need to do a **./dkms-remove.sh** before attempting an install before.

closed 2021-12-09 by jgyates