jgyates/genmon #619: Scheduled restart/reboot

dlaporte, 2021-12-08:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

I am a long-time genmon user and have experienced two minor issues over the past few years.  The first is that the GUI tends to go unresponsive randomly, getting stuck on the "Loading Generator Monitor" screen.  The other is that the MQTT integration seems to occasionally stop sending data to my MQTT broker.

In both cases a restart of the processes or a reboot of the pi resolves the issue.  However, it also sends a flurry of notifications.  Is it possible to have genmon restart itself or reboot the pi and suppress notifications until everything comes back up?

I've submitted the logs to see if we can get to root cause on both issues, but they occur randomly over the course of weeks and are hard to replicate on demand.

Thank you for all your efforts, this is a wonderful piece of software!

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: G0058702, 8KW/410 GUARD+10C T/SW
- Genmon Logs: Submit via the About page
- Genmon Version: Version V1.18.03

jgyates, 2021-12-08:
I have not received any logs today. I did receive some yesterday but the model does not match your model. You must have outbound email working for the logs to be submitted.

If you want to restart your pi, or genmon on a regular periodic basis, I would suggest using cron as this is best way to do. This is how genmon starts on boot. 

You can add an entry to cron if you ssh into your pi and type:

       sudo crontab -e

This will bring up an editor (or possibly ask you to pick and editor, if it does select 'nano' as the editor)

Once the crontab file is in the editor add this line:

     0 0 * * 0 /bin/bash /home/pi/genmon/startgenmon.sh -p 3 restart

This will restart genmon weekly. The symbols (0 0 * * 0) at the beginning of the line are the frequency of the command.

https://crontab.guru/examples.html

This will not suppress messages.

The larger question is: what is causing this and how can it be fixed? 

This is likely caused by the Flask libraries. Flask is used to allow python to be used in a web app however the internal web server that is provided is not ideal for a production environment. The ideal fix would be to replace the internal flask server with the apache2 web server, but that is something that I have not decided to tackle yet. Setting it up is one thing but getting it working with the install and update process is a bit more complicated. The short story is that there are limitations with the current implementation that I am aware of, however I have not had an opportunity to fully address this limit.

jgyates, 2021-12-08:
This thread has some related info:

https://github.com/jgyates/genmon/issues/324

Also, if you are exposing the web interface to the internet, I would suggest not doing that and us a VPN instead.

dlaporte, 2021-12-09:
Understood on the Flask issue.  I think your restart suggestion is probably the most straight-forward approach and I’ll see if I can suppress the notifications further upstream.

I configured outbound email, so I think you may now have logs.  No worries on the access – this is behind a firewall and all inbound access is restricted to internal networks and VPN.

Best,
Dave

https://crontab.guru/examples.html

jgyates, 2021-12-09:
One other suggestion is to update the system software if you have not done so recently. From the console type:

     sudo apt-get update
     sudo apt-get upgrade

Type 'y' to accept when prompted. This will update the system software. Once this is complete then reboot with this command.

     sudo reboot

This could make things a bit more reliable. I noticed in your logs you have a few instances of memory allocation and socket errors. This is not a smoking gun as there are many things that could cause this. 

Is there anything else running on the pi other than genmon?

dlaporte, 2021-12-09:
I have a temperature sensor wired up to the GPIO pins and a daemon running on 8080/tcp for my Hubitat hub to pull the stats.  That’s all that’s running other than genmon and load is ~1.25-1.5.

If you think it’s CPU/resource issue, I’ve been thinking about upgrading to a Pi Zero 2 and that would be all the excuse I need 😊

Dave

jgyates, 2021-12-09:
That is a high utilization. your CPU is overloaded %120 of the time. That is not the end of the world but since you are collecting data (temp and generator stats) it is not ideal and means that the software is alway waiting longer than it was intended. Do you have "Optimize for slower CPUs" enabled on the settings page? This will lower it a little. That option was designed to help slower CPUs line the Pi Zero. If you have the room using a pi 3 or 4 will give a lot more processing and memory. The pi zero should work fine, I just don't know how much of that load is the temp sensor. I rarely see memory allocation errors in log files and you had some in genmon.log from October and November 2021. I could not see back farther than that due to log rotation.

dlaporte, 2021-12-09:
Correct, "Optimize for slower CPUs" is enabled.  I tweaked memory usage for the monitoring process, we’ll see if that fixes things.

Thanks!
Dave

jgyates, 2021-12-19:
I am going to close this issue but feel free to post to this thread with any updates or questions on this topic.

closed 2021-12-19 by jgyates