jgyates/genmon #620: Web Page Not Loading On New Installation

stellagits, 2021-12-08:
I am in the process of moving my existing Pi based GENMON over to a Debian Linux server. 
I have installed Genmon on the Debian server per the instructions. No issues were encountered.
I manualy started the server using:

 ` ./genmon/startgenmon.sh start -p 3`

I get the following feed back in the CLI:

_Using Python 3.x...
Starting genmon python scripts
Starting....
Starting /home/user/genmon/genmon.py
Starting /home/user/genmon/genserv.py_

However, I am unable to access the UI from my local or remote clients.

I did get the following response in the log files:

_user@debian:/var/log$ cat genserv.log
2021-12-08 18:33:01,831 : Starting /home/user/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2021-12-08 18:33:10,838 : Error: Connect : [Errno 111] Connection refused : myclient.py:47
2021-12-08 18:34:59,429 : Starting /home/user/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2021-12-08 18:35:08,441 : Error: Connect : [Errno 111] Connection refused : myclient.py:47
2021-12-08 18:39:04,737 : Starting /home/user/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2021-12-08 18:39:13,748 : Error: Connect : [Errno 111] Connection refused : myclient.py:47_

I have added rules to the firewall to access the appropriate ports (23, 8000, 9082) but I have since disable d he firewall entirely for troubleshooting.
At this point I have not started the configuration of my serial interface and therefore the server is NOT connected to my generator.

Any ideas on where I can start with this?

jgyates, 2021-12-08:
This error (myclient.py:47) is due to genserv.py (the web server) failing to connect to the TCP socket in genmon.py. This socket is the way these two apps communicate. It is possible that genmon.py is not loaded due to an error. Does /var/log/genmon.log have any errors?

one thing to try is to use the ClientInterface.py app. This will connect to genmon.py the same way but it is a command line app. From the genmon folder on the command line type:

     python3 ClientInterface.py

If this work you can type:

       generator: status
       exit

if it does not work what is the error displayed to the screen?
It appears that two python apps can not establish a TCP socket on the same machine (localhost). Do both apps have access to /etc/genmon/genmon.conf ?

stellagits, 2021-12-08:
> Does /var/log/genmon.log have any errors?

It shows the following:

_2021-12-08 19:19:14,015 : Selected Controller: generac_evo_nexus
2021-12-08 19:19:14,017 : Error opening modbus device: Error opening modbus device. : generac_evolution.py:247
2021-12-08 19:19:14,017 : Possible cause is invalid serial port specified._

Would it be better to send you all my log files? If so which ones and to what email?

> one thing to try is to use the ClientInterface.py app.

I tried this and received the folowing message in the CLI:

_user@debian:~/genmon$ python3 ClientInterface.py
Genmon not loaded._

jgyates, 2021-12-08:
First make sure you are running the very latest (V1.18.03). You can update on the command line like this in the genmon directory:

      ./genmonmaint.sh -u

The try loading again.

      ./startgenmon.sh restart

you can send log files to generatormonitor.software@gmail.com

/etc/genmon/genmon.conf is the config file. This file hold configuration info. If the serial port specified in this file does not match the serial port on your system then this would also cause the error.

The problem is likely the default names of the serial port in raspbian is /dev/serial0 and in debian it is something like /dev/tty0 so the default entry in /etc/genmon/genmon.conf would be incorrect for debian. 

Are you trying to use the serial port on the debian server? Is this running serial or serial of IP? Is this a raspberry pi?

stellagits, 2021-12-08:
> Are you trying to use the serial port on the debian server? Is this running serial or serial of IP? Is this a raspberry pi?

This is a standalone PC running Debian11. I am using it to replace an existing Pi based GENMON. The existing Pi installation is configured to use serial over TCP via a serial gateway. I plan on continuing with this with the PC based Debian server I am creating now.

I have made some progress. I made changes to the genmon.conf file to reflect the correct port naming in Debian and Serial TCP address and port.

It looks like the server is starting but I still cant connect via my web clients.

I will email the current log files and genmon.conf file shortly

jgyates, 2021-12-08:
ok. the problem is related to your /etc/genmon/genmon.conf file is expecting /dev/serial0 and your system does not ahve this device name. Genmon is not loading because the default settings are not present on your system. You will need to edit /etc/genmon/genmon.conf to enable serial over TCP before it will load properly on your debian system. Here is the default values in /etc/genmon/genmon.conf

https://github.com/jgyates/genmon/wiki/1----Software-Overview#configuring-genmonpy

If you change this from

       use_serial_tcp = False

to this:

     use_serial_tcp = True

and then edit these two lines to match your serial to IP server address and port

     # if use_serial_tcp is enabled, this parameter sets the IP address used.
     serial_tcp_address =

     # if use_serial_tcp is enabled, this parameter sets the IP port used.
     serial_tcp_port = 8899

      
 Then save the file and restart genmon.

Of course you can just copy the contents of your pi /etc/genmon to your debian system /etc/genmon and this will transfer all of your settings.

stellagits, 2021-12-08:
OK more progress. I performed the above as you suggested and that was what resulted in the progress I had previously mentioned. those changes should have been in the config file I just emailed. uh oh if they weren't. Anyway, I also used the wrong serial port name. On my system, I should have specified ttyS0 as the port name. Once I fixed that, things started happening.

> Of course you can just copy the contents of your pi /etc/genmon to your debian system /etc/genmon and this will transfer all of your settings.

Why do things the easy way? :)

I restarted the server and I am now able to access the web page. It looks like once I can connect to my serial gateway then I should be good to go.

Thanks for the help !!!!

jgyates, 2021-12-08:
No worries. I am going to close this thread but feel free to post here with any questions on this topic.

jgyates, 2021-12-09:
@xjrguy 

Can you give more specific details and send your logs? The OPs problem does not appear to be the same as yours since he is using TCP over serial and using a different operating system.

When you say genmon will not start, are you referring to genmon.py or are you describing the web interface?

When you use serialtest, are you explicitly stopping genmon and using a loopback test?

jgyates, 2021-12-09:
copy and paste the last 20 lines of /var/log/genmon.log to this thread.

copy and paste the output of the console when trying to start genmon using the suggestions above as a guide. 

Let me know if you have any questions.

xjrguy, 2021-12-09:
Hi and thank you.

That is correct, OP is using TCP over serial and mine is local serial port and Raspberry latest OS. His errors were the same after updating.

Genmon.py appears to start but it doesn’t run. The web interface is also not working due to that reason.

Yes I am explicitly confirming that genmon is not running and using the loopback test

See attached screenshots.

Again, the only thing I did was update to the latest build and genmon stopped working,

I really appreciate your help and this project.

Tim

jgyates, 2021-12-09:
Attachments to email on this forum get stripped. To post a screenshot you must do it on the web

https://github.com/jgyates/genmon/issues/620

Have you tired doing the things I suggested he do? His problem is resolved. At this point I still do not know what specific error you are seeing in your log files. 

copy and paste the last 20 lines of /var/log/genmon.log to this thread.

jgyates, 2021-12-10:
@xjrguy 

Now show the output of /var/log/myserial.log (last 20 lines)

And show the output of this command:

     cd genmon
     cd OtherApps
     sudo python3 serialconfig.py -c

Do you have any other software installed on this pi?

xjrguy, 2021-12-10:
That log file exists but it’s empty

***@***.***:~/genmon $ cd OtherApps
***@***.***:~/genmon/OtherApps $ sudo python3 serialconfig.py -c

Checking : Is enable UART in boot config: OK
Checking : Is BT overlay disabled: OK
Checking : Serial console command line removed: OK
Checking : Serial console service disabled: OK
Checking : BT service disabled: OK

Serial port settings are OK. A reboot is needed if changes were made.

***@***.***:~/genmon/OtherApps $ 

No other software on this device at all.

The action item that made the change was upgrading from .02 to .03 (last known version to the latest version via the genmon web portal on this Pi)

Thank you again

Tim

jgyates, 2021-12-10:
Can you show the last 20 lines of /var/log/mymodbus.log

Also, let's see the output of this command:

     pip3 show pyserial

and this command:

     cat /proc/device-tree/model

xjrguy, 2021-12-10:
Last 20 lines of mymodbus.log

2021-12-08 13:20:33,052 : Error opening modbus device: invalid literal for int() with base 10: '5b0' : mymodbus.py:69
2021-12-08 13:20:33,052 : Error opening modbus device.
2021-12-08 13:31:40,904 : Error opening modbus device: invalid literal for int() with base 10: '5b0' : mymodbus.py:69
2021-12-08 13:31:40,904 : Error opening modbus device.
2021-12-08 13:36:37,689 : Error opening modbus device: invalid literal for int() with base 10: '5b0' : mymodbus.py:69
2021-12-08 13:36:37,690 : Error opening modbus device.
2021-12-08 13:45:26,955 : Error opening modbus device: invalid literal for int() with base 10: '5b0' : mymodbus.py:69
2021-12-08 13:45:26,955 : Error opening modbus device.
2021-12-08 14:08:48,154 : Error opening modbus device: invalid literal for int() with base 10: '5b0' : mymodbus.py:69
2021-12-08 14:08:48,155 : Error opening modbus device.
2021-12-08 14:13:34,453 : Error opening modbus device: invalid literal for int() with base 10: '5b0' : mymodbus.py:69
2021-12-08 14:13:34,453 : Error opening modbus device.
2021-12-08 14:20:13,137 : Error opening modbus device: invalid literal for int() with base 10: '5b0' : mymodbus.py:69
2021-12-08 14:20:13,137 : Error opening modbus device.
2021-12-09 15:08:29,223 : Error opening modbus device: invalid literal for int() with base 10: '5b0' : mymodbus.py:69
2021-12-09 15:08:29,223 : Error opening modbus device.
2021-12-09 15:08:48,392 : Error opening modbus device: invalid literal for int() with base 10: '5b0' : mymodbus.py:69
2021-12-09 15:08:48,392 : Error opening modbus device.
2021-12-09 17:33:21,140 : Error opening modbus device: invalid literal for int() with base 10: '5b0' : mymodbus.py:69
2021-12-09 17:33:21,140 : Error opening modbus device.

***@***.***:~ $ pip3 show pyserial
Name: pyserial
Version: 3.5b0
Summary: Python Serial Port Extension
Home-page: https://github.com/pyserial/pyserial
Author: Chris Liechti
Author-email: ***@***.***
License: BSD
Location: /usr/lib/python3/dist-packages
Requires: 
Required-by: mote

***@***.***:~ $ cat /proc/device-tree/model
Raspberry Pi 4 Model B Rev 1.4

Hope that helps.

Thank you

Tim

jgyates, 2021-12-10:
for some reason you have a beta version of pyserial. The "pip3 show pyserial" shows version 3.5b0. "b0 is beta 0".
I have 3.5 installed (no b0). Try this:

     sudo pip3 uninstall pyserial
     sudo pip3 install pyserial

Then this:

     sudo pip3 show pyserial

It is odd that you don't have a /var/log/myserial.log. After you reinstall pyserial, try loading genmon again.

      ./startgenmon.sh restart

then look and see if you have a /var/log/myserial.log. It could be that log rotation kicked in. You should a file named /var/log/myserial.log.1, /var/log/myserial.log.2, etc if you have other archived log files. The log files get rotated once they reach a certain size.

xjrguy, 2021-12-10:
That was it!!

It was the darn beta pyserial.

I have no idea how that got installed. I do regular sudo apt update and sudo apt full-upgrade.

Your command didn’t work I had to use this (maybe this will help you in the future)

sudo pip3 install --ignore-installed -U pyserial

Thank you for the help and for the awesome genmon!!

jgyates, 2021-12-10:
Excellent! I am glad it is working. 

I am going to close this thread but if anyone has other questions or status on this topic feel free to post to this thread.

closed 2021-12-10 by jgyates