jgyates/genmon #621: Question on the PHAT cable

GitBeagle, 2021-12-08:
I purchased a PHAT from pintsize.me and the standard cable before fully understanding the differences between Evolution 1&2. I bought the one that fits my Pi zero. So I need to modify the cable to provide the 12V missing on pin 3. I have confirmed that there is no 12V present on pin 3. I haven't been able to locate the exact information on what connections the PHAT requires. I'm trying to make a round cable that I can seal with a normal cable gland. From the Wiki "making a cable" section it appears I only need power, ground Tx and Rx (pins 3 or battery+, 2, 7 & 8). Is this still true for the PHAT?

I just want to confirm the connections to avoid letting out the magic smoke on my P1 Zero! I'm posting here as I could not see a way to ask questions on the pintsize.me site.

### Your Environment

- Generator Model: Response 22Kw installed 6/2018 
- I believe it is an Evolution 1.0 controller with Firmware V1.20, hardware V1.06
- It does not have WiFi

(The Response model is a standard air cooled Generac that our power utility can supposedly utilize in case they are unable to supply enough power to satisfy the demand. They will reimburse the customer for the fuel used during this period. It extended our warranty from 5 to 7 years.)

skipfire, 2021-12-08:
The PintSize.Me site does have a contact form on it, but I watch here as well.  The pHAT uses the 4 pins on the right half of the connector, data is the right 2 pins on the top row and the two pins below them are the 12v and GND.  If you have the cable plugged in so that the black wire is in the bottom right, that will be GND, 12v will be red.  The cables sold with battery terminals just remove the red and black wires from the generator end and then have loops crimped on (and some shrink for appearance).  The cables are sold by themselves as well, though shipping makes them a bit expensive to buy alone.

closed 2021-12-09 by jgyates

GitBeagle, 2021-12-09:
Thanks. Just what I needed.