jgyates/genmon #622: Linux Autostart Options - Requesting advice, not a problem

stellagits, 2021-12-09:
I've been working with Linux off and on for a few years but have never really understood the "correct" way to auto run something, especially on start-up.

I've noticed that in the case of GENMON (also with other projects), the method used to auto start a program is with an entry in the cron. Alternaticely, others have chosen creating and managing a service through systemd.

Not sure why one method is preferred over another. Could GENMON be run as a service via systemd?

jgyates, 2021-12-09:
I opted for cron because it was easy to implement and mostly ubiquitous across Linux platforms. You could use systemd if you wanted to. I am sure there are minor differences between the two. Also, I liked the fact that cron just runs a given command at a given time and everything else is up to the program running. I could delay the command if needed and I don't think that flexibility is available with systemd.

stellagits, 2021-12-09:
>  I opted for cron because it was easy to implement and mostly ubiquitous across Linux platforms.

Didnt even consider that systemd doesnt exist on all platforms. Good to know. I'll stick with the cron because it works. I was just curious why some chose one method over the other.

Thank you for the information.

closed 2021-12-09 by stellagits