jgyates/genmon #623: Looks to be errant system in Alarm condition reported. Possible More data on #576

tr6scott, 2021-12-12:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Not to Alarm when no Alarm present? 

### Actual Behavior

During a power outage, got message generator state change, system not in alarm, with Switch State: System in Alarm. Followed shortly after that system is not in alarm. 

Time: 07:51:48 AM
Date: Sun 12-Dec-2021

NOTE: This message is a notice that the state of the generator has changed. The system is not in alarm.

Status : 

    Engine : 
        Switch State : System in Alarm
        Engine State : Running
        Battery Voltage : 14.10 V
        RPM : 3610 
        Frequency : 60.00 Hz
        Output Voltage : 240 V
        Output Current : 7.30 A
        Output Power (Single Phase) : 1.75 kW
        Active Rotor Poles (Calculated) : 2 

    Line : 
        Utility Voltage : 0 V
        Utility Max Voltage : 0 V
        Utility Min Voltage : 0 V
        Utility Threshold Voltage : 156 V

    Last Log Entries : 

        Logs : 
            Alarm Log : 12/04/21 11:45:40 ESTOP Pressed : Alarm Code: 2800
            Service Log : 
            Run Log : 12/11/21 13:24:32 Running - Utility Loss 

    Time : 
        Monitor Time : Sunday December 12, 2021 07:51:47
        Generator Time : Sunday December 12, 2021 07:51

No Alarms: 0001:00032020

Followed by:

Time: 07:52:01 AM
Date: Sun 12-Dec-2021

NOTE: This message is a notice that the state of the generator has changed. The system is not in alarm.

Status : 

    Engine : 
        Switch State : Auto
        Engine State : Running
        Battery Voltage : 14.20 V
        RPM : 3595 
        Frequency : 59.00 Hz
        Output Voltage : 238 V
        Output Current : 7.30 A
        Output Power (Single Phase) : 1.74 kW
        Active Rotor Poles (Calculated) : 2 

    Line : 
        Utility Voltage : 0 V
        Utility Max Voltage : 0 V
        Utility Min Voltage : 0 V
        Utility Threshold Voltage : 156 V

    Last Log Entries : 

        Logs : 
            Alarm Log : 12/04/21 11:45:40 ESTOP Pressed : Alarm Code: 2800
            Service Log : 
            Run Log : 12/11/21 13:24:32 Running - Utility Loss 

    Time : 
        Monitor Time : Sunday December 12, 2021 07:51:58
        Generator Time : Sunday December 12, 2021 07:51

No Alarms: 0001:00030000

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: G0072099
        Hardware Version : V1.01
        Firmware Version : V1.12
- Genmon Logs: Done. 
- Genmon Version: 18.03

tr6scott, 2021-12-12:
Sent a 2nd set of logs, as it just happened again.

jgyates, 2021-12-12:
@tr6scott 

This is a known issue with Evolution 2.0 controllers. See number 11 on this page:

https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues

You can mitigate this my going to the advanced settings page (double click the gear icon on the main web page) and enable "Ignore Unknown Values" then click save at the bottom of the page.

I

closed 2021-12-12 by jgyates