jgyates/genmon #624: Multiple Devices Running Genmon - Question

stellagits, 2021-12-12:
I am currently running GENMON on a Debian Linux Server. I would like to build a stand-alone remote monitoring panel that provides for remote monitoring via pushbuttons and LED indicators. I would like to run this on a RasPi to make use of it's GPIO. Is there any problem with running a second instance of GENMON on my network?

jgyates, 2021-12-12:
If you want to run multiple instances of genmon on separate pi systems on the same network then that is not an issue at all. If you want to run multiple instances of genmon on the same systems then look at this article for guidance:

https://github.com/jgyates/genmon/wiki/Appendix-J-Multiple-Instances-of-Genmon

stellagits, 2021-12-13:
Excellent, thanks for theinformation

closed 2021-12-13 by stellagits