jgyates/genmon #625: Not receiving data error intermittent.

farberm, 2021-12-19:
I am getting intermittent -“Not receiving data” emails from the system. It has been running great for two years no issues. 

Can you look to see if it is a bad serial board? Or some other issue?

jgyates, 2021-12-19:
I received your logs. You do have some CRC errors, which indicates that you are receiving some data but based on the logs the data is not what is expected. I would guess that you either have a bad RS-232 converter or some intermittent cabling.

farberm, 2021-12-20:
Thanks. I will recheck the cabling and if it still persists I will replace the 232 converter

jgyates, 2021-12-20:
I am closing this issue for now but feel free to post here is you have an update or more questions.

closed 2021-12-20 by jgyates

jgyates, 2022-06-02:
I did not receive your registers. Your outbound email must be working for register / log submission to work.

Running these tests listed here should be able to narrow it down to the pi, the RS-232 converter, your cabling or the controller:

       https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

DJT1423, 2022-06-02:
Thanks. I am sure something in my cabling has failed. I found a backup SD card from more than a year ago and I inserted that in my Pi. Same error occurred, so I know it is not a software issue. I will run the tests you specified.

David

David Thompson
***@***.***

   https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

DJT1423, 2022-06-02:
The loopback test #2, serial 9-pin connector fails - actually, it does not fail, rather it sends a signal but then simply hangs waiting for a response. I have double checked the wiring per the schematics, and all appears correct (and of course it had worked for almost 2 years). Is it likely, or even possible that either the RS232 to TTL Converter Module or the DB9 adapter have simply failed?
David

   https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

jgyates, 2022-06-02:
That is certainly possible. Electronic components can fail at seaming random moments as well as obvious incidents like lightening strikes.

While you are testing, make sure that you try to check for marginal connections in the wiring. Anything connected to the generator is subject to vibrations when the generator is running. This  can cause marginal electrical connections over time.

DJT1423, 2022-06-15:
Hi, I'm hoping for some suggestions again on this issue. I have replaced both the components of the RS-32 converter, and still am failing the serial test. (See attached image)  Curiously, when the original DCE is in place the failure reads:
"FAILED: Sent data does not match receive. Received 14 bytes"
But when the replacement DCE unit is in place (and I tried three as the package I received contained 5 units), the error reads:
"FAILED: Sent data does not match receive. Received 0 bytes"
I also replaced the 4 cables from the Pi to the DCE unit.
Oh by the way, I again confirmed that the serial test passes at the "Loopback at the Pi Expansion Connector".

Any wisdom you might be able to share?

Thanks, David

***@***.***

jgyates, 2022-06-15:
Images that are attached to email replies are stripped off so I did not get the image. To attach and image you must use the gitbub.com web interface. The email should have a link to the thread.

Pictures of your power and cabling may help.

davisgoodman, 2022-06-15:
@DJT1423 Just one thought here.. Are you using a PiHat to get your TTL to pins 14-15 or you are connected directly. Could also be the UART having issues. Cold joint or something like that. I believe you are not even getting the loopback test through. 

My two cents...

DJT1423, 2022-06-15:
Photos attached.

[image]
[image]
[image]
[image]

DJT1423, 2022-06-15:
> @DJT1423 Just one thought here.. Are you using a PiHat to get your TTL to pins 14-15 or you are connected directly. Could also be the UART having issues. Cold joint or something like that. I believe you are not even getting the loopback test through.
> 
> My two cents...

Always happy to hear suggestions. No, I am not using a HAT. Direct connection to the GPIO pins to the RS-232 converter.
David

davisgoodman, 2022-06-15:
Ok so that is going to be an easy one I believe... Your black cable is on the wrong pin... tx/rx are on pin 14-15
[image]

DJT1423, 2022-06-15:
@davisgoodman  - I think that is what I have, is it not? The first pin is empty, and I am using the next four. If I reverse the white and black and run the serial test, the result is:
"Serial port opened
write data: sent test string
waiting to receive data"
and then it hangs

DJT1423, 2022-06-15:
I'll play again in the AM and let you know how it goes. It appears the TX and RX pins are reversed on the new RS 232 converters I received from the original that worked for the last year +

David

David Thompson
***@***.***

[image]<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F37666918%2F173717266-f22cddb7-06ae-4010-a5d7-1bf0b85204ba.png&data=05%7C01%7C%7Cd8c91b697b1c4364c7a808da4e6ee3a0%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637908535411225153%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=deocIb76rK8xNSsO6KvrPVjWfMtshfsgcW7dXF7cVIE%3D&reserved=0>

davisgoodman, 2022-06-15:
Ok maybe the angle of the picture threw me off. Couldn't see the first empty pin. My apologies.. :-)

Have you tried a loopback directly between pin14 and 15 without the 232/TTL converter. 

@jgyates that should be a valid test I suppose?

skipfire, 2022-06-15:
The wiring looks right (unless you have tx & rx swapped).  Wire colors isn't great, it is standard to use black for gnd rather than data.

- Connect Pin pins 8 & 10 directly and retry the loopback, this will test the Pi's serial configuration and functionality without the RS232.  If it fails, you have Pi level problems to solve first
- Connect the pins back to the RS-232 and retry the loopback.  If it passes, great.
- Swap the tx and rx pins on the RS-232 and retry the loopback.  If it still fails, the RS232 component is probably bad.
- If you have a multi-meter, you might check for continuity between the two DB9 screws you are using to make sure that your loopback wire is well connected.

DJT1423, 2022-06-15:
Yes, that works fine.
D

David Thompson
***@***.***

DJT1423, 2022-06-15:
Thx. I'll work on it tomorrow. I don't have a multimeter but will try to work something out.
David

David Thompson
***@***.***

jgyates, 2022-06-15:
It looks like you have tx and rx swapped. Your 232 convert has TX connected to white wire, but you have the white wire connected to Rx on the pi. Also you black wire is connected to RX on the converter but to TX on the pi. 

Powering the converter with 5V is known to work, according to the data sheet 3.3V is recommended so instead of using the 5V on the pi header, use 3.3V.

This page has notes on 5V vs 3.3V for the 232 converter.

https://github.com/jgyates/genmon/wiki/3.1--Making-a-Cable

DJT1423, 2022-06-15:
I re-did all my wires last night and passed the loopback test, finally. The RX and DX pins on the new RS-232 converters are reversed from the previous device! This morning I moved power to the 3.3 volt pin, and passed the test again.  Next up, back to the garage and reconnect to the generator.
(PS: I re-wired using coloured wires that match your coloured wiring diagram :-). 

David

DJT1423, 2022-06-15:
WooHoo, SUCCESS - back online. Thanks for your patience and suggestions and comments. Much appreciated.
David
[image]