jgyates/genmon #627: Any way to add second SMTP Configuration?

TopBanana123, 2021-12-22:
I currently have SMTP setup to go through my Office365 account, and the emails work great.

My goal is to setup some kind of local notifications, and one of the ways I found is to use local SMTP into this HomeBridge plugin, which would give me notifications through Apple HomeKit

https://github.com/SphtKr/homebridge-smtpsensor

But, that would wipe out my existing SMTP setup. Is there any way to add a second SMTP configuration?

jgyates, 2021-12-22:
There is already a request for HomeBridge integration here

https://github.com/jgyates/genmon/issues/552

I will think on this feature vs the original homekit plugin method

TopBanana123, 2021-12-23:
HomeBridge support would be AWESOME. Oh man, the cool things you could do

Plus, the all local notifications that would then be available

jgyates, 2021-12-25:
I am going to close this issue as the hombridge support is already on the proposed feature list.

closed 2021-12-25 by jgyates