jgyates/genmon #628: No Data

farberm, 2021-12-22:
I have a system that has been working great for two years. The other day the circuit breaker to the outlet to my rasp pi 4 tripped twice in a row. Coincidently I am now not receiving data from the generator.

My step is as follows:

RPI4 connected to rs232 controller. The pi os powered by the wall outlet. A three wire cable goes to the generator. One cable attached to the ground of the batter and the other two cables to the tx/rx pin 8 and 9. The ground wire connects to pin 5 and the tx and rx to pins 2 and 3. 

I checked the cable and they all seem connected. I replaced the RS232 converter. Still no data receiving. (jgyates, I send another log to you just now) One red LED on the RS232 is on all the time and the other blinks. Previously the second LED was LED and pulsed bright and dim.

I also reset the controller with by removing the fuse and the battery connector and restarting

1. Is it possible that the raspberry pi sd cards needs to be reprogrammed
2. Is it possible I need to reinstall the Genmon software
3. Is it possible the rip is damaged? (seems to start fine?

Tomorrow I will also run a serial check out to the generator. The serial cable check inside is working fine.

jgyates, 2021-12-22:
I assume this is the same issue mentioned here: https://github.com/jgyates/genmon/issues/625

For the sake of others who may running into a similar issue and searching through past posts, I am adding the link above to reference the previous post. For future reference please try to keep all posts for a single issue in a single thread.

Your logs show the same result as before. I agree the next step is to look at your cabling to the generator and validate this with a loopback test.

jgyates, 2021-12-31:
any update on this issue?

farberm, 2021-12-31:
As a matter of fact just fixed it about an hour ago. I replaced all the wiring and reset the controller by unplugging the battery and fuse. It is not communicating. Not sure it it was the cabling or controller reset that did it.

jgyates, 2021-12-31:
Outstanding, I am glad it is working for you now.

closed 2021-12-31 by jgyates