jgyates/genmon #629: Support for 8kw/OG7901

jmertic, 2021-12-23:
Hi!

First off - an amazing project and work being done here. As someone big into open source, seeing efforts like this really embody strong community efforts.

So I inherited a Generac model 55180 - which is an 8kw unit. It uses the OG7901 pre-nexus controller; my original went bad and I had tried to replace it with 0H06430SRV which is what is in the pictures in the repo, but the generator wouldn't run for more than 10 seconds or so, so I got an OG7901SRV to replace it and it works great.

My question is - does this project support the OG7901? I see on the back of the controller the pinout for the molex plug, but unlike the 0H06430SRV there is no molex header for it.

Thanks again!

jgyates, 2021-12-23:
Hi @jmertic 

I don't know a lot about the various pre-Nexus controller variants. If there is a serial port on the controller then there is the possibility that it can be supported. If your controller does not have a molex header then I would guess it does not have a serial port. If anyone else wants to weigh in on this feel free.

Thanks
Jason

jmertic, 2021-12-24:
Here are some pictures if that helps....

liltux, 2021-12-24:
@jmertic sorry the only controllers that were designed for MobileLink were the digital display ones.  Since this project piggy backs off of the connection for mobile link, which in this case you do not have the connection, I don’t think this will work.  Possibly you could solder in a connection to the pads or same place on the board as the digital display version may make it compatible.  But there is a risk of damaging the controller and it may not be supported by the firmware either.

jgyates, 2021-12-24:
@jmertic 

I am going to close this issue for now but feel free to post to this thread if you have other questions or want to try some other things to make this happen.

closed 2021-12-24 by jgyates