jgyates/genmon #630: Total failure with Bullseye on a Pi Zero W 2

aa777888-2, 2021-12-24:
I did a fresh install of the latest Raspi Bullseye on a brand new Zero W 2 using the Pi Imager.

Got through the installation procedure and manually starting genmon results in no error messages but also no python processes running.

Should I not be loading Bullseye? If not, what should I load?

Really looking forward to using this outstanding tool, but very stuck right now!

Thanks!

jgyates, 2021-12-24:
@aa777888-2 ,

Bullseye has worked as expected in my testing. What is the output of the following files?

     /var/log/genmon.log
     /var/log/genloader.log
     /var/log/myserial.log

You can display this by typing :

      cat /var/log/genmon.log

and 

      cat /var/log/genloader.log

Also, any additional info you can provide would be helpful, for example how did you install, how did you manually load, have you made any changes to the config files, are you using the provided genmonmaint.sh and startgenmon.sh script files to install and load?

aa777888-2, 2021-12-24:
I'm currently rebuilding the system from scratch in order to meticulously go through the process again. If it does not work this second time I will post the requested information here.

aa777888-2, 2021-12-24:
OK, here we go...

I followed all instructions here: https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software

I do not have the pi attached to a generator at this time. I just wanted to see the web interface come up. It would appear that maybe this has something to do with it?

`
pi@genmon:/var/log $ more genmon.log
2021-12-24 14:10:32,108 : New version detected: Old = V1.18.03, New = V1.18.03
2021-12-24 14:10:32,131 : Selected Controller: generac_evo_nexus
2021-12-24 14:10:32,141 : Error opening modbus device: Error opening modbus device. : generac_evolution.py:247
2021-12-24 14:10:32,142 : Possible cause is invalid serial port specified.
pi@genmon:/var/log $ more genloader.log
2021-12-24 14:10:30,381 : Updated entry gengpio.conf
pi@genmon:/var/log $ more genserv.log
2021-12-24 14:10:36,110 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2021-12-24 14:10:45,130 : Error: Connect : [Errno 111] Connection refused : myclient.py:47
pi@genmon:/var/log $ more myserial.log
pi@genmon:/var/log $
`

jgyates, 2021-12-24:
Looks like you have an issue with the serial ports. Have you enabled the onboard pi serial port?

Show the output of this commad:

     cd OtherApps
     python3 serialconfig.py -c 

This should show us if the serial port is setup properly

Have you edited /etc/genmon/genmon.conf at all?

What does this command show:

       ls -la /dev/ser*

It should show something like this:

 $ ls -la /dev/ser*
lrwxrwxrwx 1 root root 7 Dec 24 14:17 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root 5 Dec 24 14:17 /dev/serial1 -> ttyS0

aa777888-2, 2021-12-24:
pi@genmon:~/genmon/OtherApps $ python3 serialconfig.py -c
You need to have root privileges to run this script.
Please try again, this time using 'sudo'. Exiting.
pi@genmon:~/genmon/OtherApps $ sudo python serialconfig.py -c

Checking : Is enable UART in boot config: OK
Checking : Is BT overlay disabled: OK
Checking : Serial console command line removed: OK
Checking : Serial console service disabled: OK
Checking : BT service disabled: OK

Serial port settings are OK. A reboot is needed if changes were made.

I have not edited genmon.conf at all. All I did was follow the directions at https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software and answer "Yes" to all setup questions.

pi@genmon:~/genmon/OtherApps $ ls -la /dev/ser*
lrwxrwxrwx 1 root root 7 Dec 24 14:09 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root 5 Dec 24 14:09 /dev/serial1 -> ttyS0

Thanks for helping me work through this.

jgyates, 2021-12-24:
how about /var/log/mymodbus.log

is any other software loaded on this pi?

jgyates, 2021-12-24:
also, lets try this?

     cd genmon
      sudo python3 genmon.py

What does the output show.

aa777888-2, 2021-12-24:
pi@genmon:/var/log $ more mymodbus.log
2021-12-24 14:10:32,140 : Error opening modbus device: invalid literal for int() with base 10: '5b0' : mymodbus.py:69
2021-12-24 14:10:32,141 : Error opening modbus device.

No other software loaded. Box stock build from the pi imager tool. I did run apt-get update and upgrade, of course. Turned on SSH and VNC. Changed the password, name and setup WiFi. That's it.

pi@genmon:~/genmon $ sudo python3 genmon.py
Error opening modbus device.

jgyates, 2021-12-24:
what is the output of this command:

      pip3 show pyserial

jgyates, 2021-12-24:
I just checked in a mod that may give a little more info in the log file. To update type this:

    cd genmon
    ./genmonmaint.sh -u

Then try starting genmon and then show /var/log/myserial.log. It looks like there is an error occuring on myserial.py that is not getting logged as it is not in an exception block so I don't know specifically what error is occuring. The error is being passed up to mymodbus.py but the specific error condition is not logged. This update should help shed some light in what is happening.
 
Showing "pip3 show pyserial" is still useful. I have seen once instance where someone, somehow, had a beta version of pyserial installed. Fixing that solved the problem. Their issue looks similar to yours at this point but I don't know that for sure.

aa777888-2, 2021-12-24:
pi@genmon:~/genmon $ pip3 show pyserial
Name: pyserial
Version: 3.5b0
Summary: Python Serial Port Extension
Home-page: https://github.com/pyserial/pyserial
Author: Chris Liechti
Author-email: cliechti@gmx.net
License: BSD
Location: /usr/lib/python3/dist-packages
Requires:
Required-by: mote

Update installed. Results...

pi@genmon:~/genmon $ more /var/log/myserial.log
2021-12-24 16:53:55,087 : Error in init: invalid literal for int() with base 10: '5b0' : myserial.py:60
2021-12-24 16:53:55,088 : Error on serial port init!

jgyates, 2021-12-24:
That is it, you have the beta version of pyserial.

see this thread:

https://github.com/jgyates/genmon/issues/620

This command should fix it.

      sudo pip3 install --ignore-installed -U pyserial

jgyates, 2021-12-24:
I checked in an update to mymail.py that *should* be more forgiving of a beta version. I am not sure why this version is installed for some people. This has happened twice that I know of.

This should give the new update.

     cd genmon
    ./genmonmaint.sh -u

aa777888-2, 2021-12-24:
Brilliant! It is up! Thank you! Fantastic support, most appreciated!

Perhaps the installer should check for this and throw an error?

But this is ridiculous, of course. Why would an out of the box, totally stock pi build have the beta of pyserial? Quite annoying!

closed 2021-12-25 by jgyates