jgyates/genmon #631: Outbound notification email not working

hokie21, 2021-12-24:
### Expected Behavior
I was getting email notifications each week for the weekly test. 

### Actual Behavior
I just noticed that I have not received any notifications since 20 November.  mymail.log shows no activity after 20 November. "Test Email Settings" from the Settings menu works reliably each time and these emails are received at the same email address configured in the "notifications" menu.  The notifications menu is set to show outage, error, warn and info.

### Steps to Reproduce (including precondition)
Turn generator from "auto" to "off" or vice versa, no email is received. Wait for weekly test. No email is received.

### Your Environment

- Generator Model: 5875, Air Cooled, 20 KW.
- Genmon Logs: Submit via the About page. Submitted.
- Genmon Version: Version V1.17.01

jgyates, 2021-12-24:
The last set of logs I received was two days ago. I would recommend restarting genmon and your pi an see if that corrects the issue.  Nothing in the email support has changed in quite a while so this may be a networking issue with your pi. If the problem persists after a restart / reboot I would update your system software and genmon.

hokie21, 2021-12-24:
I've seen this before and a reset did temporarily resolve the problem. A few months later it happens again. Is there anything you want to do to root cause this or collect any additional data on this before I reset? Do you expect test emails sent via the GUI to show up in the mymail.log log? Would you expect networking issues to be indicated in mymail.log?

jgyates, 2021-12-24:
No, the test emails are separate and should not be logged to myserial.log. Once it comes back up send your logs. Before you reboot type this command an post the output here:

     cat /proc/meminfo

hokie21, 2021-12-24:
I rebooted. I did receive the "Generator Monitor Starting" and "Generator Notice" emails as expected.

I just did "submit logs."

Tnx!

cat /proc/meminfo
MemTotal:         442820 kB
MemFree:           73348 kB
MemAvailable:     295192 kB
Buffers:           48420 kB
Cached:           232152 kB
SwapCached:           68 kB
Active:           193028 kB
Inactive:         125836 kB
Active(anon):      33188 kB
Inactive(anon):    30280 kB
Active(file):     159840 kB
Inactive(file):    95556 kB
Unevictable:          16 kB
Mlocked:              16 kB
SwapTotal:        102396 kB
SwapFree:         102140 kB
Dirty:                 0 kB
Writeback:             0 kB
AnonPages:         38268 kB
Mapped:            25360 kB
Shmem:             25180 kB
Slab:              38948 kB
SReclaimable:      22800 kB
SUnreclaim:        16148 kB
KernelStack:         720 kB
PageTables:         1132 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:      323804 kB
Committed_AS:     318548 kB
VmallocTotal:     573440 kB
VmallocUsed:           0 kB
VmallocChunk:          0 kB
Percpu:              608 kB
CmaTotal:           8192 kB
CmaFree:            6032 kB

jgyates, 2021-12-24:
Both your  /var/log/genmon.log and /var/log/mymail.log have errors related to DNS failures. I would make sure you have a backup DNS server listed in /etc/resolve.conf

    nameserver 8.8.8.8
    nameserver 8.8.4.4

Aded the above lines will give you two DNS servers from google

hokie21, 2021-12-24:
Updated now.

closed 2021-12-25 by jgyates