jgyates/genmon #632: error communicating...: invalid literal for int() with base 10: '5b0' serialtest.py:82

danthorndyke, 2021-12-25:
I'm having an issue running the serial test script.

I received the following when trying to test the serial port. 
error communicating...: invalid literal for int() with base 10: '5b0' serialtest.py:82

Here is the enviroment.
pi@genmon:~/genmon/OtherApps $ python --version
Python 3.9.2

pi@genmon:~/genmon/OtherApps $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

I used the latest image of raspberry and their Windows image loader. It's a clean install with nothing else installed.

Thank you for all your hard work!

- Generator Model: Generac 22 kw
- Genmon Version: 1.18.03

jgyates, 2021-12-25:
I checked in a fix for this so you can update to the latest version from be web interface or by this command:

       cd genmon
       ./genmonmaint.sh -u

I added a check for the version number of the python serial library a couple of version back. For some reason, some folks have the beta version of pyserial installed on their system. The beta version caused the version check I added to choke because there was a letter (b for beta) in the version string. 

This command should replace the beta version with the release 3.5 of pyserial 

     sudo pip3 install --ignore-installed -U pyserial

Let me know if that does not fix the problem.

danthorndyke, 2021-12-25:
Oh my! A fix today was very unexpected but a great Christmas present! 😃Worked like a charm!

Should I go ahead and replace the beta version?

Thank you and Merry Christmas!

Loopback testing for serial port /dev/serial0...

Using python serial library V3.5b0
Serial port opened
write data: sent test string
waiting to received data....
PASSED! Loopback successful

jgyates, 2021-12-26:
You can probably leave it with no issues.

danthorndyke, 2021-12-26:
I replaced it with version 3.5 just in case. 

It's working great! 

It doesn't seem to know the Model or Rated kW. Is there a definition for the 22 kW or maybe one that needs to be updated?

Thanks again!

[image]

jgyates, 2021-12-26:
The software will attempt to look up the model number and, depending on the model the rated kw based on the serial number by performing an internet lookup. The look is not successful the software will attempt to perform the look up again, but only when the version of the software changes. If you initially loaded genmon without the serial connected to the controller this can occur. If you want to force the lookup to be performed again you can edit the file /etc/genmon/gemon.conf and delete the following line:

         version = V1.18.03

Then restart genmon.

If you ssh into the pi then type:

        sudo nano /etc/genmon/genmon.conf

scroll to the bottom of the file and remove the line:

         version = V1.18.03

Then save the file by typing Ctrl+X, then press Y to save the file, then enter. This will save the file and exit the nano editor

To restart type:

         cd genmon
         ./startgenmon.sh restart

After you try this, if you still don't have a recognized model and you have a valid serial number in your controller, if you outbound email is configured you can submit your error logs from the About page and I can see if i can tell what is happening. 

If your controller has ever been replaced, replacement controllers do not have a serial number but you can override the serial number value from the advanced page (double click the gear icon in the top right of the web interface to access the Advanced settings page).

danthorndyke, 2021-12-26:
Excellent! Great instructions and explanation! The unknown information has now populated.

I just happened to notice the Fuel Type is listed as Propane. The generator is new and has only run maybe 30 minutes with exercise time. I did not confirm it was set to Natural Gas from the factory as I read but I will tomorrow. It is connected to Natural Gas.

I sent a donation your way today. Keep up the great work!

Thanks again!

jgyates, 2021-12-26:
For air cooled models, the fuel type should be set by the user on the settings page. For liquid cooled models this value is read from the controller.

thanks for the donation!

I am going to close this issue for now but if you have any other questions on this topic feel free to post them to this thread

closed 2021-12-26 by jgyates