jgyates/genmon #633: Weather no longer showing up.

j0mama, 2021-12-27:
### Expected Behavior

Weather.

### Actual Behavior

Nothing.

### Steps to Reproduce (including precondition)

Double checked city code as well as API key

### Your Environment

- Genmon Version: 1.18.03

jgyates, 2021-12-27:
No logs have been submitted. Weather is working for me and nothing has changed on the weather specific code for quite some time so this is likely something specific to your system or openweather account. Your logs would show any errors that have occurred related to the weather api.

j0mama, 2021-12-27:
2021-12-27 01:11:03,584 : Error in GetObservation: (5991370) : 'OWM' object has no attribute 'weather_at_id' : myweather.py:108

I just looked in the logs

griffths, 2021-12-27:
I enabled weather a few days ago and have been unsuccessful getting it to
work/display in the app.

Based on the instructions I should not have to manually install any of the
Openweather support files (fresh install of V 1.18) so all I did was
enable the option in settings as well as create an account with
Openweather.

I did read an old comment that it may take a few hours for the OW API key
to fully activate but that has long passed without successfully displaying
anything in Genmon.

I just submitted my logs to you in case they may uncover something further

Regards,

Jay

jgyates, 2021-12-27:
@j0mama  and @griffths 

What is the output of this command:

     pip3 show pyowm

it should show something like this:

     Name: pyowm
     Version: 2.10.0
     Summary: A Python wrapper around OpenWeatherMap web APIs
     Home-page: http://github.com/csparpa/pyowm
     Author: Claudio Sparpaglione (@csparpa)
     Author-email: csparpa@gmail.com
     License: MIT
     Location: /usr/local/lib/python3.9/dist-packages
     Requires: geojson, requests
     Required-by: 

The important part is  Version: 2.10.0. If this is not the same version type this:

       sudo pip uninstall pyowm

Type y to confirm yes. Then type this:

      sudo install pyowm==2.10.0 

Let me know if this resolves the issue for you.

j0mama, 2021-12-27:
[image]

jgyates, 2021-12-27:
sorry, typo on my part, it should be

     sudo pip3 unistall pyowm

Then

     sudo pip3 install pyowm==2.10.0 

pip is for python2 pip3 is for python3. It shows above that you have the python 3 version of pyowm V3.2. This is the problem.  I you type the correct commands above it should work.

The problem is that the pyowm library changed the way you get the weather using the city ID. Once I realized this I made genmon stay with version 2.10 for python3 and 2.9 for python2. It is possible that some installs were made between the time that openweathermap made this change to their api and the time that I limited the version for genmon.

j0mama, 2021-12-27:
[image]
[image]

Same error in logs

j0mama, 2021-12-27:
Rebooted, and it came up

j0mama, 2021-12-27:
[image]

Thank you!

closed 2021-12-27 by j0mama

jgyates, 2021-12-27:
I forgot to mention that you have to restart genmon since it is actively using the library you are replacing. I am glad it is working.

@griffths, let me know if this solves your issue as well

griffths, 2021-12-27:
Just got home...

Here is the requested output....

***@***.***:~ $ pip3 show pyowm
Name: pyowm
Version: 3.2.0
Summary: A Python wrapper around OpenWeatherMap web APIs
Home-page: https://github.com/csparpa/pyowm
Author: Claudio Sparpaglione
Author-email: ***@***.***
License: MIT
Location: /usr/local/lib/python3.7/dist-packages
Requires: PySocks, requests, requests, geojson
Required-by:

So based on this I will run your suggested commands and let you know the
results....

Thanks for the ongoing support of your awesome software.

Jay

griffths, 2021-12-27:
@jgyates,

That did it, weather came up immediately.

Thanks again for your awesome ongoing effort and support!

Jay