jgyates/genmon #634: Pi Zero 2 W build for Nexus Controller (brag/example post)

aa777888-2, 2021-12-30:
Hello fellow genmon enthusiasts! 

I had noted a few other "this is my implementation/execution" type issue posts here so I thought it might help someone to see how I did it. Plus it's fun to brag. 🙂

First off, many, many thanks to JGY for his incredible accomplishment, and for the super support he brings to the community, which is clearly evident when perusing the issues section of this project.

I would have jumped on this bandwagon two years ago when it all started, but I only just became aware of the project, quite by accident, on my Google news feed just a few weeks ago.

[image]

It seemed clear that this was a project perfect for the Pi Zero platform. I would have gone full hardwire PoE but there is zero room left in the wire conduits to my generator, a 2011 vintage Generac 17KW unit running on propane. Needing WiFi, I quickly found that Pi Zero W's were currently unobtanium. Even Zero 2 W's were hard to find, but I did get one for the standard price of $15. And I was happy to see that the Nexus provides 5V, so no worries about fussing with power supplies. All things considered, this is actually a simpler and much less expensive approach than pulling another cable from my switch and integrating a PoE hat.

I wanted the smallest footprint easily possible. By easy that meant minimal mechanical fabrication and obtainable from Amazon (Prime, free shipping, free returns if I hated it, all that sort of thing). I found this very nice, highly rated RS232-TTL level translator board which would easily nest within the outline of the Pi Zero: [https://www.amazon.com/dp/B07BJJ3TZR](url)

Using that board I could probably have fit the darn thing in a large pill bottle. I seriously considered just putting the two boards back to back and dunking it in Plasti Dip. But ultimately I decided I wanted something a bit more repairable, just in case. I found this very small, IP68 rated, junction box on Amazon that seemed perfect: [https://www.amazon.com/dp/B07TFSLLBY](url) Having built many systems in my career that meet MIL-STD-810, the nice gasketing and molded in cable glands were a big draw. It was plastic, so no worries about the on-board WiFi antenna on the Zero. No shielding was a slight worry, but it did not turn out to be a problem. The extra cable gland would be perfect to captivate an external WiFi antenna if required. As already noted it was not necessary. My nearest WAP is quite near, and signal levels are right around -60dBm in both directions per the Pi and the WAP. The extra gland was sealed up by the simple expedient of putting a large brass bolt into it and tightening the gland down.

[image]

Of course I also used the cable assembly recommended on the genmon wiki, as it was cheaper and easier to get it from Amazon than ordering the parts from Digikey or Mouser: [https://www.amazon.com/dp/B000M802RG](url) Their shipping always kills you, and adding pigtails of the right wire gauge was not any harder than crimpling contacts, if somewhat less elegant. I did think about getting some jacketed cable that would work better with the cable gland, but sometimes perfection is the enemy of good enough.

[image]

I had originally thought I would attach the level translator board back to back with the Pi using some high-bond, double-sided tape, then screw down the Pi into the case. But that would disturb the watertight integrity of the case, and it turned out that the two boards fit perfectly onto the case lid. By using two layers of tape, the boards stand proud of the lip on the lid thereby providing easy access, without any further disassembly, to the SD card and other ports should future maintenance be necessary.

[image]

If the boards look a little weird, that's because they are mostly covered in a urethane conformal coating: [https://www.amazon.com/gp/product/B008OA7AWE](url) This will protect the covered areas from any condensing humidity, and there will be plenty of that in an outdoor application like this. IP68 does not mean hermetically sealed. I did not add a drain (weep) hole to the box, and we'll see how that goes. The level translator board is slathered front and back, as is the back of the Pi, but it's not possible to do that on the front of the Pi without damaging the connectors and ports.. The front of the Pi is carefully coated to avoid those areas. Urethane doesn't look as pretty as acrylic conformal coat, and doesn't flow out as well, but it's easier to apply with no masking or back-potting of connectors. I did mask the WiFi antenna footprint with a bit of Kapton tape, just in case I want to add that in the future, but so far it appears unnecessary.

Waxing a bit more poetic about conformal coating, I've had two throttle stepper motor failures in this generator. Both times it was corrosion of the poorly made, completely uncoated, solder joints on the junction PCB inside the stepper motor. When the spare failed I cracked them both open, found the problem, re-soldered both, and urethane coated the boards. I haven't had a lick of trouble since then.

[image]

Installation isn't fancy. I plugged it in and rested the box on an existing wire harness. This seems more than secure enough. Simple, and out of the weather. The little junction box is very light, and by doing it this way it avoids any vibration the box would be subject to if it were rigidly mounted to the generator structure. Again, my nearest WAP is only about 20 feet away and signal strength has not been a problem. EMI has also not proven to be a problem, perhaps because of the very short cable length. However, the Pi certainly is as snug as a bug, showing a CPU temperature rise of approx. 50°F degrees above ambient on a cloudy day (e.g. running 90 on a 40 deg. day). It will be interesting to see how that pans out on a 95°F summer day with the generator running. Some cooling modifications may prove necessary. **If anyone has any tips/hints on how to maybe slow the processor down in order to lower its temperature they would be most welcome.**

[image]

Total expenditure was only $39: $15 for the Pi, $8 for the level translator, $7 for the cable assembly, and $9 for the junction box. Got to love free Amazon shipping! Everything else I already had: wire, tie wraps, shrink tubing, solder, flux, coating, etc.

I'll wrap this up with a schematic. I was quite fussy about paying attention to serial signal paths. I never trust "TXD" and "RXD" labels.

Cheers and Happy New Year to all!

[image]

jgyates, 2021-12-30:
Wow, great setup. I really like this enclosure.

aa777888-2, 2021-12-30:
Thank you! 🙂

I'm unhappy with the cooling, though. I just ordered some heat sinks off of Amazon. I'll post an update after I get those installed. Silly me, I put urethane on top of the AU (CPU). Now I've got some scraping to do! I really didn't think it would be a 50°F rise! Even at 100 ambient that's within limits but it's asking for trouble. At least I've got the rest of winter to get that sorted out. If I can get the rise down to 30 that would be perfect. so that on a below zero day self-heating is good, and on a hot summer day it'll be more reasonable.

skipfire, 2021-12-30:
@aa777888-2 I have a Pi 3B+ running mine and it is located inside my generator enclosure.  After Ida (I live north of New Orleans and got a solid hit from the storm) I was on generator for 6 days with mid-90s for highs and mid-80s for lows, and the Pi stayed at a happy and healthy temperature the entire duration - until the generator shut off.  Once the generator shut off the temperature spiked and the CPU was between 85C and 90C until I opened the enclosure up.  On mine, the air intake for the generator is pulled through the cavity under the controller so the generator running actually decreased the CPU temperature by 5C to 10C due to increased airflow.  Once the generator stopped, the radiant heat went into the cavity and heated it quickly until I opened the enclosure for an hour (within 5 minutes of it shutting down).  There are no notable heat increases when I do a 1-hour transfer test, nor did it heat up when I shut down the generator to check the oil; probably because the lid was open for the shutdown and oil checks so it could radiate out and only closed the lid once the generator had restarted.

If you keep the Pi inside a small watertight non-metallic enclosure I doubt a heatsink will help since there is no airflow to allow heat exchange, but I also don't think it could hurt.  YMMV

aa777888-2, 2021-12-30:
skip, you make an excellent point about airflow through the generator when it is running! So I'll no longer worry about that!

One step at a time here in terms of mechanical complexity. We'll see what just more efficiently transferring heat into air mass of the box does. If that's not good enough then I think I just found a purpose for the unused cable gland, which is to pass a heat pipe through it like this one: https://www.amazon.com/uxcell-Copper-Cooling-Laptop-Heatsink/dp/B0987M4LTC

aa777888-2, 2022-01-03:
The heatsink kit arrived. Removing the urethane from the top of the processor with a razor blade wasn't hard at all. It took 10 degrees out of the problem, down to a 40°F rise from ambient from 50. So not as good as I would have liked. I'd prefer to be at 30°F rise. But at this point, again, perfect might be the enemy of good enough, because 100 + 40 is still only 140°F, and that is well within the allowable CPU temperature range. So I will now wait for the serious summer weather to hit and update this issue in six months or so as I see what the higher temperatures bring to my design. If there is a problem I have some good ideas on how to fix it with an inexpensive heat pipe. In the meantime, Happy New Year to all!

[image]

jgyates, 2022-01-05:
Let us know how it goes. I am going to close this thread for now but feel free to post here with an update later.

w-u-2-o, 2022-07-03:
Update on thermal performance: now that it is the middle of summer, I can report that even seeing a genmon reported CPU temperature between a 35 to 40°F rise from ambient, everything seems to be running just fine. On a 90°F day, with full sunshine on the generator enclosure, I'm seeing a CPU temp. of about 130°F and performance remains normal.

jgyates, 2022-07-03:
Thanks for the update!

aa777888-2, 2022-12-27:
**Update: modification to provide for an external Wi-Fi antenna.**

I recently rearranged the wireless access points in the house and this caused a significant drop in signal strength to my genmon. Not unsurprisingly, since I was relying completely on the built-in Pi Zero 2 W on-board antenna and the genmon unit is located entirely within the generator housing. Before the WAP moved I saw numbers in the middle of the yellow, say around -67dBm give or take. Not fantastic but very acceptable. After the WAP move these dropped by approx. 10dB, well into the red range on the genmon signal strength meter.

The first thing I had to do was to install a U.FL connector at the location provided on the Zero 2 W board. I had masked this off with some Kapton tape when I originally conformally coated the board. I peeled the tape, soldered on the connector, and cut the trace to the printed board antenna.

Before:

[image]

After:

[image]

[image]

Then I prepared a pre-built U.FL to SMA pigtail assembly by installing the SMA in some nylon spacer material. This allowed me to use the extra cable gland on the housing to hold the SMA connector.

[image]

[image]

There are a million versions of these products on Amazon, eBay and elsewhere. It's not worth it to provide links, just buy whichever stuff floats your boat.

Finally, I provisioned an inexpensive, outdoor style Wi-Fi antenna on the outside of the generator housing. 

[image]

[image]

This was the antenna I purchased: [https://www.amazon.com/dp/B08PCDDG8J?th=1](https://www.amazon.com/dp/B08PCDDG8J?th=1)

The wire loom is important in order to prevent critters from chewing on things and also to protect the cable from vibration and abrasion as it passed through the vent screen.

The results saw my signal levels back up into the yellow zone where they were before.

jgyates, 2022-12-27:
Very nice! Great info