jgyates/genmon #635: Upgrade broke GenMon. Attempting to access sits on "Loading Generator Monitor" page

Starz26, 2022-01-01:
### Expected Behavior

Clicking the "Upgrade" button successfully upgrades the software and GenMon can be accessed.

### Actual Behavior

Clicked the upgrade button and now the page that is shown has a spinner and "Loading Generator Monitor"

### Steps to Reproduce (including precondition)

See above

### Screenshot or Pictures relating to the problem (if possible)

[image]

### Your Environment

[image]

- Generator Model: [image]
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2022-01-01:
I am going to need more info before I can help. I don't know what version you were upgrading from. What version of raspbian are you using?

You can log into the pi va ssh and and type:

     cat /etc/os-release

and copy and paste the output of the above command to this thread.

Just a wild guess, but if you were upgrading from any version before V1.18.0 I would look at these threads since V1.18.0 started with python 3 as the default and older versions of debian (Jessie) will need to be upgraded or a minor mod (listed in these threads) to keep using python 2.x:

https://github.com/jgyates/genmon/issues/598

https://github.com/jgyates/genmon/issues/603

Starz26, 2022-01-02:
I think I *was* on 1.15 something. I just updated buster to the latest oldstable release. I went through and updated all libraries and installed python3.

Now I get this when trying to start genmon

Warning: required library pyopenssl not installed. Attempting to install....
Error in InstallLibrary using pip : pyopenssl : b'  Failed building wheel for cryptography\n  Failed cleaning build dir for cryptography\n  Can\'t roll back cryptography; was not uninstalled\nCommand "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__=\'/tmp/pip-install-mrbfj78s/cryptography/setup.py\';f=getattr(tokenize, \'open\', open)(__file__);code=f.read().replace(\'\\r\\n\', \'\\n\');f.close();exec(compile(code, __file__, \'exec\'))" install --record /tmp/pip-record-n59s1pig/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-mrbfj78s/cryptography/\n'

Here is the output from your command

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

jgyates, 2022-01-02:
Looks like you have some conflicts in your python libraries. the pyopenssl library changed the library used for cryptography after python 3.5. Problems like these are tricky to find the root cause, but not impossible. I would suggest one of two paths.

1) Try uninstalling some of the libraries and reinstall them and see if that works

        sudo pip3 uninstall pyopenssl
        sudo pip3 uninstall cryptography
        sudo pip3 install pyopenssl

2) Save the contents of /etc/genmon and reinstall raspbian. Upgrading from one version of raspbian (or debian) to another is possible, but can cause issues like this.

You can try option one and if that does not work then move on to option 2. Let me know how it goes.

jgyates, 2022-01-02:
Another thing you may want to try is to run this command:

     sudo apt-get install build-essential libssl-dev libffi-dev python3-dev

Then try number one above. This makes sure you have the right environment to install some of the python libraries.

Starz26, 2022-01-02:
I had to change pip3 to pip to remove all those packages then I used the last command to install pyopenssl. Then it looks like pip needed updated, did that, then RUST was not installed so I installed it.

Still had issues so I ran

sudo apt-get install python3.6-dev

based on some stack overflow posts.

Still did not fix it. Then I ran

sudo apt-get install build-essential libssl-dev libffi-dev \
    python3-dev cargo

Still lots of errors. This is not an upgrade anyone should be doing when on buster and genmon 1.15.

If I backed up the genmon settings via the UI. What are the best steps to start fresh and get my settings, logs, and everything running again?

jgyates, 2022-01-03:
I am not exactly sure what steps you took when you did this: " I just updated buster to the latest oldstable release."

If you upgraded your pi from Jessie (Raspbian 8) to Buster (Raspbian 10) then I  completely agree. Upgrading the OS is not a recommended path. I have not seen a successful outcome to this approach and can leave your system with a lot of invalid library combinations.

For those who are running Jessie the recommended path to continue to use genmon is one of the following:

1) Don't upgrade to V1.18.x 
2) Upgrade to V1.18 but modify the crontab script to use 

Modify it from this:

       @reboot sleep 20 && /bin/bash /home/pi/genmon/startgenmon.sh start

To this:

       @reboot sleep 20 && /bin/bash /home/pi/genmon/startgenmon.sh -p 2 start

3) Make a backup of your configuration, then reinstall Raspbian Bullseye.

To backup in normal conditions, you can just click backup on the web interface about page. If genmon is not running you  can ssh into the pi and type:

        cd genmon
        ./genmonmaint.sh -b

This will create a file in the /home/pi folder named genmon_backup.tar.gz. This is the backup file that contains the contents of the /etc/genmon folder. Save this file to another system. 

If you already have a backup, then reinstall per the installation instructions. Once genmon is installed then copy the backup file to the pi. You can unzip the file like this:

       tar -xf ./genmon_backup.tar.gz

The files will be put in a folder named genmon_backup. Then copy the files to the /etc/genmon folder

        sudo cp ./genmon_backup/* /etc/genmon

Starz26, 2022-01-03:
I don't remember either but it was something simple like sudo apt-get upgrade or something. Nothing extreme.

Genmon is currently running but not sure the state of things at this point. The upgrade button is still there (not sure if it is supposed to be). Given the events I have taken a backup and......

I will just install bullseye and go from there.

closed 2022-01-03 by Starz26