jgyates/genmon #636: Multiple tanks configuration in gentankutil.conf

Mehrdadg, 2022-01-04:
Love this project as I also find the Generac IOS app + annual fee is useless and not to mention serious wifi network security hole as it always broadcasts MLGxxx which can be used as AP and from there you have access to all the internal lan devices! of course this can be solved via firewall rules.

now the issue.... I have two tanks which are both monitored remotely using generac tankutil sensor. 

the documentations says if you have multiple tanks, you need to name them each so I am doing this with the following syntax:

in gentankutil.conf:
....
tank_name = Primary Tank
tank_name = Secondary Tank

I have even tried using ' or " (e.g. "Primary Tank") but for some reason it only shows one tank level on the main page. I am sure I am doing something silly so any help is greatly appreciated or perhaps a bug?
thanks
MG

jgyates, 2022-01-04:
Hi @Mehrdadg,

There are a few types of tank support in this project, (built in tanks for Evolution and Industrial Diesel, External Tanks support from tankutility.com, and the DIY sensor support). Currently the DIY tank support is the only one that supports two tanks. Adding support for multiple tanks with tankutility is possible, I just know know the format of the data for the second tank that is read from the tankutility web API. I can look into adding this support, but I may need to get some data from you if possible since I do not have two tanks to test with. It would require you running the software and submitting logs. Let me know if this approach is suitable for you.

Mehrdadg, 2022-01-04:
Hi

Please let me know how I can do to support this effort and happy to provide some info as needed.

Regards

Mehrdad

jgyates, 2022-01-04:
I updated the code to support this. To test do the following:

1) Edit the file /etc/genmon/gentankutil.conf and add the following lines:

         tank_name_2 = "tank name"

 where "tank name" is the name of the 2nd tank you have setup in tankutility.com. 

2) Update genmon from the about page. the version has not been updated since this is just a test for now.

You *should* see "External Tank" and "External Tank 2"

Let me know the results. Also, if this works send me your logs on the about page so I can validate there are no errors. Let me know if you have any questions.

Mehrdadg, 2022-01-04:
Very impressive! 
It worked, and now it shows two separate tiles with External Tank 1 and External Tank 2 names on the status page. It also shows the proper readings of the tanks.
Many thanks and great work.
Mehrdad
PS. I also submitted the log for your review.

jgyates, 2022-01-04:
Thanks for testing. I am glad it is working.

closed 2022-01-04 by jgyates