jgyates/genmon #637: MQTT retain message

SethL, 2022-01-04:
I'm setting up mqtt with Home Assistant and it is working great.  The only issue I am having is that when the generator is not running all of my dashboards error out.  This is because the sensors are not reporting a value to HA via mqtt.  Is it possible to add the retain option to the mqtt module?  I think this would solve things as the broker would retain the last message which wold be zero.  

[image]

jgyates, 2022-01-04:
Have you tried using the flush interval option in the genmon MQTT add on? This will set a timeout that will send all values that are unchanged. If your HA setup has a default value that sets and error after a period of inactivity, set the flush internal to half of that value to guarantee that this error will not occur.

SethL, 2022-01-04:
Ah ha!  
That did it.  I had to uncomment some of the disabled items but I am good to go. thanks!

[image]

closed 2022-01-04 by jgyates