jgyates/genmon #638: Outage log and total consumption 461.55cf for 0:00 length outage

Starz26, 2022-01-08:
Not really looking to figure out why, just what I need to modify to remove the outage and update the Maintenance total consumption calculation.

I removed the line from the outage log but that did not cause a recalculation. Here is the current state:

outage.log

[image]

Maintenance screen

[image]

How can I remove the 461.55 outage and the 461.55 from the total consumption?

jgyates, 2022-01-08:
if you have removed that entry from the outage log in /etc/genmon/outage.txt, then the only other step should be to restart genmon. You can do this from the advanced page or via ssh. For ssh type:

      cd genmon
      ./startgenmon.sh restart

Genmon only reads the outage file on startup and keeps a copy in memory, but writes it when there is an outage.

jgyates, 2022-01-08:
I am going to close this issue. Let me know if you still have questions.

closed 2022-01-08 by jgyates

Starz26, 2022-01-10:
@jgyates I deleted the line from /etc/genmon/outage.txt and then did the /.startgenmon.sh restart

The maintenance screen still shows the 461.55

outage:

[image]

Maintenance screen

[image]

jgyates, 2022-01-10:
To delete the fuel consumption from the total you will need to delete the corresponding entries in /etc/genmon/kwlog.txt

To do this delete all of the entries with the date of the outage you deleted earlier (10-23-2020). Save the file, then restart genmon. There are several of these entries for that date, one every 10 seconds that the generator was running.

Starz26, 2022-01-11:
> To delete the fuel consumption from the total you will need to delete the corresponding entries in /etc/genmon/kwlog.txt
>
> To do this delete all of the entries with the date of the outage you deleted [...]

:( there are no entries for 10/23/20 - may just have to reset power log and fuel consumption.