jgyates/genmon #639: (Not An Issue) Adding Custom Gaugues and Start, Stop, Manual Buttons (Also Custom buttons)

Ntampata, 2022-01-08:
Hi @jgyates, happy New Year, I hope that the 2022 will be a good year for you Sir!
I have been editing the json files to be able to use my custom controller (Deep Sea 7320), and its working fine, thank you for your support.

1. How can I add more gauge to the "Status" Interface? Let’s say, Gauge for Coolant Temp (Address: 0x01D7), Oil Pressure (Address: 0x020C) and other custom gauges. 
1.1. I was checking the files "libraries.min.js" and "genmon.js" but I need some light or guide to move on.

2. Correct me if I am wrong, I believe that I saw at previous version of this project the Buttons for start, stop, auto mode button, manual mode buttons, at Maintenance Page there is a way to add these button? 
Example: The Deep Sea Controller that I am using, to select the Auto mode, One of the system control keys must be written into register 8 and its ones-compliment written into register 9 with a single function 16 (write multiple registers) to perform the specified system control function.
To Select Auto mode: 
Function Code: 16
System control keys to be written into register 8 is: 35701
Its ones-compliment written into register 9 is: 29834

I am trying to add these buttons, but for now I am stuck, because there is two values that must be written into register to select the Auto Mode on Deep sea controller.

Thank you for your support!

jgyates, 2022-01-08:
To add a new gauge you can add an entry to the "gauges" array in your custom json file. Here is the gauges array:

https://github.com/jgyates/genmon/blob/master/data/controller/evo_lc.json#L295

The modbus register listed in your entry must be in the "base_registers" array.

Regarding the buttons, at this time the button interface is not supported on custom controllers. Presently only reading is supported, not writing registers. Let me think on this feature. In general, setting up an interface to do generic writes to modbus registers from a json file is a bit more complicated as there are many ways to write bits to enact a change in a controller so there are a lot of variables to consider to make this a meaningful feature. 

If you send me your json file and submit your registes on the about page I can simulate your generator and add your json file to the repository.  You can attached the json file here or email it to generatormonitor.software@gmail.com

jgyates, 2022-01-26:
@Ntampata 

any chance I could get a copy of your json file to add to the repository for anyone else wanting to use the deep sea controller? Also, if you have out bound email working you can submit your registers which helps me maintain and test the functionality of the custom controller support.

Thanks,
Jason

Ntampata, 2022-01-29:
Hi @jgyates 
Sorry for the late reply, I took few days off!
I will send a copy of my json file to be added to the repository.

Best regards,
Haudálio JJ Portraite

jgyates, 2022-01-29:
Got it. Thanks, I made some updates and checked in your custom JSON file so now the Deep Sea Controller support is available to all.

I will put the custom button issue on the feature TODO list.

https://github.com/jgyates/genmon/blob/master/feature_requests.md

closed 2022-01-29 by jgyates

jgyates, 2022-02-08:
This feature is included in V1.18.06

T-C123, 2023-01-10:
I have a similar question. I am making an adaptation of the thermocouple pHAT done by Louis Mamakos at https://github.com/lmamakos/RPi-pHat-Thermocouple to use with my Generac 18kw, air cooled generator with Evo-2 controls. I have temperature sensors (thermocouple on a MAX31850K amplifier and a DS18B20 thermometer) working to read oil temperature and Pi case temperature which are displaying as text in the Genmon status page. I would like to have these temperatures displayed graphically as the other gauges on the status page. I’ve looked at the Evolution_Liquid_Cooled.json file but, as I understand it, this file takes data from the controller for display.

Can an add-on be created that will take the temperature readings and display them graphically or is there another way to do this? I’d like something that won’t change if the Genmon software is updated.

Beginner here so please be gentile.

Thanks for all you do!

andrewvanlagen, 2023-01-10:
I export my temperature data to grafana

jgyates, 2023-01-10:
@T-C123 

The [gentemp](https://github.com/jgyates/genmon/wiki/1----Software-Overview#gentemppy-optional) add on, which can be enabled on the Add On page will add  one wire temperature sensors that are detected by the raspberry pi OS to the genmon user interface. Both DS18B20 and MAX31855 have been tested. In short, if you design your thermocouple interface to be compatible with the built in linux driver, then genmon will recognize it and display the values.

This page has info in software compatibility 

https://github.com/lmamakos/RPi-pHat-Thermocouple#integration-with-other-software-packages-and-systems

Also, genmon shows the temperature imported by the add on in text, not as a gauge. If you want to show a guage then you may want to look at exporting your genmon data to some other platform via MQTT and display it graphically (grafana, home assistant, etc)

T-C123, 2023-01-10:
I am using gentemp to display the data as text. I was hoping that since there are options to display additional gauges for liquid cooled Generac, Briggs & Stratton, & Deepsea controllers there might be a way to add additional gauges the temperature data.

Could do it with other software (Grafana) but would like to keep it within Genmon.

Thank you both for your response!

jgyates, 2023-01-11:
Adding gauges for Briggs, Deepsea, etc is done via modbus defining modbus data formats. Adding gauges for external temp sensors requires modifying the code in /genmon/controller.py. I can add this to the list of requested features.

https://github.com/jgyates/genmon/blob/master/feature_requests.md

I will update this thread once this feature has been added.

lmamakos, 2023-01-13:
@T-C123 as I mentioned on the github page, beware if you using those Maxim 1-wire bus parts as it relies on the thermocouple junction to be electrically isolated from the power supply, ground.. and other thermocouple junctions connected to neighboring chips.   This caused me a few hours of debugging until I realized this.   (And as a side-effect, I got really good at SMD rework, removing and replacing this SMD QFN packages multiple times!)

My solution was to encapsulate the junction in high temperature epoxy to provide the isolation. 

If I were to do this again, I think I'd adopt high temperature thermistor sesnsors, like you'd find in an oven or grill thermometer.  It was a great learning experience, though..

T-C123, 2023-01-13:
Thank you @lmamakos ! I did read your suggestions and added the drop of epoxy just in case my thermocouple finds ground. Almost ready to install.

jgyates, 2023-01-14:
@T-C123 

I have added the code to display external temp sensors as gauges. To enable the gauges you need to fill in the new parameters in the gentemp add on (nominal value and max value for each sensor). The nominal and max data is needed to tell the software how to draw the gauge. Let me know if you have any issues.

T-C123, 2023-01-15:
Thank you, I now have gauges! Working at figuring out what the "proper" numbers are for nominal and max values. I'll have to install everything and see what the readings are when running. Will Genmon send an alert if the readings go out of nominal range or hit max?
Another feature that I'd like to include is an oil pressure gauge. I've searched Github but haven't found anyone who has tried this with Genmon. Do you know if it has been done? 

My Generac 18kw, Evo-2 doesn't seem to read oil pressure. It only has a switch to cut off at low (<10 psi) pressure or high (>310 deg) oil temperature. I'm looking at some standard, automotive sending units. Older ones seem to vary resistance from 240 - 33 ohms for 0 - 100 psi. My reading indicates that I'd need an analog to digital converter to interface with the Pi and then some software to use one of these. Not sure if newer sending units are analog or digital. Just thinking out loud, code would need to send an alarm only if the pressure dropped while the generator was supposed to be running, not before startup or when shutting down.
Any thoughts would be appreciated!
Thanks for all of your work!
Tom

jgyates, 2023-01-15:
The gauges are display only. They do not send any notices. This has been the case since the controller monitors most conditions and sends alarms when things are not functioning as expected. To add any functionality for sending notices I would need to justify this with a use case, i.e why is this needed in the software and it is a benefit to the project. you an measure a lot of things with temperature sensors, but is this redundant to what the controller is offering?

The red and yellow regions of the gauges are calculated at fixed intervals based on the nominal. These intervals are based on the range for industrial generators coolant temp sensor ranges.

The Evolution controller already has an alarm for low oil pressure. Just because the information is not exported over modbus by the controller, does not mean that the controller is not monitoring that parameter.

T-C123, 2023-01-16:
As I've said, I'm new at this but from what I can figure, my generator doesn't so much monitor oil temp and pressure, it will only shut down if the pressure is low or the temperature is high. I wanted to monitor these things so I could determine if a problem was developing before getting caught with a DOA generator. 

A lot of people seem to want custom features for their application and it would be unfair to expect you to customize Genmon for each individual. I don't know how it would work, but maybe some sort of custom configuration file that would contain boilerplate for a variety of sensors. It could be modified by the individual and read by Genmon on startup to provide data to the interface. Just a thought.
Thanks always for you fast responses and all you do to keep this working!
Tom

jgyates, 2023-01-17:
Let me think on this. It is my understanding that both Nexus and Evolution generators monitor low oil pressure and high temperature and set alarms for them if they are out of range and many other alarms that would handle other errors. I guess my question is what are you wanting to monitor specifically (e.g. high oil pressure?) and what would you be trying to prevent that an alarm condition does not already exist? I would think this would be a hard thing to get right (i.e. what temp or pressure would you set it to be the alarm threshold an not get false alarms?). The better I can see a solution that is workable the easier the case or adding this feature.

T-C123, 2023-01-17:
I can't speak for the Nexus but my Evolution controller only sets an alarm when the oil pressure gets too low (<10 psi) or the oil temperature gets too high (>310 deg. F.). This is when these readings get critical and the controller's response to the alarm is to shut down the generator in order to "save" the engine. I like being able to monitor these readings and be able to take action before the situation is critical. Knowing the oil pressure is dropping may indicate low oil level, wrong viscosity, engine wear, or clogged oil filter. Temperature rising above normal may indicate poor airflow or lean fuel mixture. Being alerted to these readings progressing towards the critical point will allow the user to investigate and rectify a problem before the generator becomes out of service.
The alarm level should be user selectable since it will depend on application and environment. I would set my low oil pressure alarm somewhere between "normal" operating temperature and the Generac critical pressure. That would let me know a problem was developing before shutdown. Same with the high oil temperature alarm. Others may want sensors to measure engine head, block, or exhaust temperature and would need to set alarms appropriate for them. Someone living in the north may have different alarm levels than someone living in the south.
I can think of a number of other user specific things someone might want to monitor and be alerted to. Most would not be needed for most users and would involve a lot of time spent developing. If Genmon had a custom configuration file with the boilerplate to access and display different sensors, a knowledgeable user could adapt it for his situation. I know someone could modify your software to do these things but an update might wipe that all out. A custom file could remain unchanged as new versions of Genmon were released.
I'm not trying to make more work for you! You've been great at developing this application and responding to user requests and I hope my suggestions might be of help to others. Thanks!
Tom