jgyates/genmon #640: intermittent log file history issue

effeng1, 2022-01-08:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

### Expected Behavior

Complete and up to date log and run history

### Actual Behavior
Intermittent graphic of test run and conditions,  it seems to have months filled in, and then they disappear, note January test run not showing which happens each Wed at 1pm.

### Steps to Reproduce (including precondition)

comes and goes, sometimes works, sometimes not as you can see on screenshot

### Screenshot or Pictures relating to the problem (if possible)
[image]

### Your Environment

Model : G0058871
    Generator Serial Number : xxx
    Controller Detected : Nexus, Air Cooled
    Nominal RPM : 3600
    Rated kW : 20 kW
    Nominal Frequency : 60 Hz
    Fuel Type : Propane
    Engine Displacement : 999 cc

jgyates, 2022-01-08:
I received your logs and loaded them in my emulator and I get the same results as you show above. I think this may be an issue interpreting the presentation.  With the exception of  the current month of January, all log entries retained by the controller are shown. The controller will maintain a fixed number of log entries (I think 50 if memory serves). As new entries are created the older ones are deleted. The graphic at the top will show a combined display of the Error log and Run log on Nexus (Evo has those to and maintenance log). The gap you are seeing from June to November is because the lack of entries in either of the logs in the controller.  If I am interpreting what your concern is incorrectly let me know.

jgyates, 2022-01-08:
also, the genmon software does not keep the run and error logs. Those are maintained by the controller. The outage log is maintained by the genmon software.

davehawthorneGitHub, 2022-01-08:
Mine appears to have stopped Logging in 2022

[image]

jgyates, 2022-01-08:
@davehawthorneGitHub 

The current month is not displayed, only full months. This is a know issue.

effeng1, 2022-01-08:
What is strange is i have had the months of October, November and December showing correctly up to about two weeks ago, and then I have lost that data and showing as you see now.  It never deletes the oldest months eg.  Feb, Mar, April21.   The  unit and the genmon software work flawlessly other than this issue.  Not a huge issue, but strange none the less. 
Appreciate any ideas.

effeng1, 2022-01-08:
I also have always had the current month showing and filling in as new exercise/test runs were completed.  It never needed to wait until the month was completed to show up.

jgyates, 2022-01-08:
I don't recall this being an issue in the past either, but I am aware of this issue. Other than this issue, are there any other issues with this interface that you are aware of? I want to make sure I have all of the issues identified before I attempt to address any issues with the javascript.

effeng1, 2022-01-08:
Losing October and November data and January not showing are the only issues i see. Thanks for the review. 
Ed

davehawthorneGitHub, 2022-01-08:
> I also have always had the current month showing and filling in as new exercise/test runs were completed. It never needed to wait until the month was completed to show up.

Same here, almost immediately the green square showed the updated information.

jgyates, 2022-01-08:
@effeng1 

I agree with the issue with January, but I do not see an issue with October and November. The graph shows the last 12 months. Your logs do not have any dates in the last 12 months in October and November so there is nothing to display. As I mentioned before the run logs and alarm logs are maintained by the controller. The controller only keeps 50 entries for each log and older entries are deleted. I can not do anything to address this issue as it is a function of the controller, not the genmon software.

effeng1, 2022-01-08:
got it and agree with January issue alone.   Note, we had a huge utility outage issue in December where the utility tripped on and off for about 4 minutes straight for under 10 second durations each,  obviously their protection scheme did not work correctly that day which you may also be able to see in the logs.  So I assume that is why October and November had data and I was use to seeing about 3 months worth of exercise tests but that would explain the over-write. of registers. thanks for letting me know about the 50 entries per log. I am working on a project to add a utility under/over voltage relay with return to utility delay so that can't happen again, i would rather have the generator start more often and carry the utility then have the utility cycle on and off and blow the appliances. It is too bad that this is not more settable in the nexus controller. 
Many thanks again
Ed

jgyates, 2022-01-09:
If you update to the latest version you should see the problem with the log heat map not showing up for the current month. Let me know if you have any issues. the version number has not changed.

effeng1, 2022-01-09:
Upgraded and January has returned, looks good and is a great software, thanks for the info and fix. 
Ed

closed 2022-01-09 by jgyates