jgyates/genmon #645: External CT Monitor shows double KW Output

TopBanana123, 2022-01-13:
### Expected Behavior

L1 amps + L2 amps should be x120 to get the KW output

### Actual Behavior

L1 amps + L2 amps are x240 to give double the KW output

### Steps to Reproduce (including precondition)

Use the PintSize.me Sensor hat, add CT's to leg 1 and leg 2 and look at the KW, its showing for example a 3.6KW load when there is 15a load on the generator

There is 10a on L1 (120v) and 5a on L2 (120v) meaning the total load would actually be 1800w

### Screenshot or Pictures relating to the problem (if possible)

[image]

### Your Environment

- Generator Model: Generac RG027
- Genmon Logs: Done
- Genmon Version: Version V1.18.04

jgyates, 2022-01-14:
Thanks for the feedback. I checked in an update that should fix this. Let me know how it goes. I also added CT1 and CT2 to the status text on the main page.

TopBanana123, 2022-01-14:
Thanks!

I love how you did that, text showing the two legs and then the total amps. Perfect!

I'll test it out tomorrow morning and report back how it goes. I was comparing to my Enphase monitoring which is pretty accurate, and lead to me seeing the double usage

TopBanana123, 2022-01-14:
It looks like the KW Calculation is now accurate, however the Leg1 and Leg2 numbers are out of whack

Correct KW here

[image]

But this shows all 30a at the time on leg1, which is not possible as I had multiple 240v loads running at the time (Pool Pump, Mini Split AC)

[image]

TopBanana123, 2022-01-14:
Another thing I noticed is that the chart for current in my case goes to 124a, with the red starting at 104a which is the max output for my generator on NG (25kw)

However, because this is adding up the two 120v legs into one chart, the actual max is 208a. But even that is not correct, as you can't go over 104a on one leg of the generator

In my opinion, it might be better to ditch the single external current chart, and replace it with "Line 1 Current" and "Line 2 Current" or something similar, then just have the KW the same as it currently stands

jgyates, 2022-01-14:
@TopBanana123 

First, make sure both of your CTs are the same model. Next, put this value in /etc/genmon/gencthat.conf

       debug = True

Restart genmon and send me your logs. The log file should show the values read from the both CTs. If these values are off then you may want to try a different CT as one may be mis-labeled. Looking at the code and in my testing I don't see leg one being off. Can you measure the leg with a clamp on multi-meter?

Regarding the gauge / display. Let me think on this. I see what you are saying, I just want to  process this as there are some issues in how models with built in CTs display things and changing this for the add on may effect the built in models so I want to make sure I don't break something else up by changing this.

jgyates, 2022-01-14:
The log file is /var/log/gencthat.log

TopBanana123, 2022-01-14:
Both CT's are the same model, and the total amperage does equal the correct total amps. I measured with my clamp meter and when it was showing 15a, I had 12 + 3a

[image]

They are the correct way around, and both on different legs

[image]

I'll get those logs on in a moment. 

Personally I would be fine with no current display as long as the text was there for each leg, and the KW displayed stayed

TopBanana123, 2022-01-14:
After enabling debug will it then show the details from the run I did earlier, or would you like me to run the generator again?

jgyates, 2022-01-14:
No, it will only show for the time when debug = true is enabled. It will not show past runs. Another data point would be swapping the plugs on the CT and see if the problem follows the CT or the plug (i.e. the ADC channel the plug is assigned to).

TopBanana123, 2022-01-14:
Thanks, I'll get the logs sent over to you in the next few hours when I have a moment free

I figure it must be getting a good reading from both CT's, as the total amps is correct

TopBanana123, 2022-01-14:
Just sent the logs, I did not swap the CT's yet as thats a bit of a pain on this model. But I will get to that at some point

jgyates, 2022-01-14:
The logs are showing this:

CT1: 0.65, CT2: 17.88

which matches the output on the Status page. This rules out most of the software (except gencthat.py). Swapping the CTs will show if it is dependant on the HAT or CT.

jgyates, 2022-01-14:
A thought on the max current output on the gauge. The max current is max power divided by voltage output. This is single phase (but split into two legs to support a two phase like setup so power factor is 1.0) with a voltage divider circuit to give the two leg outputs. The max current for both legs should be 25000/240. One leg can provided more than half but the total will never exceed 104A. The red line in the graph starts at 104A, which is the max. There is a valid argument to be made that the red line should start sooner. The reality is that you will likely get an over current alarm before you hit the redline on the graph.

jgyates, 2022-01-14:
after looking at the code the colors on the graph change like this:

Nominal is rated kW / Nominal output voltage(240)

Green is 0 - (Nominal * 0.8)
Yellow is (Nominal * 0.8) - (Nominal * 0.95)
Red is (Nominal * 0.95) -  (Nominal * 1.2)

TopBanana123, 2022-01-14:
Okay on the CT's, you were right, it was on my end. One of the CT's was not reading for some reason. But in the end, they both read high

This is what was throwing me, because it just so happened that the amount it was reading high was the amount of the other leg! Talk about unlucky

Here is a video of the high reading: https://youtu.be/8v9NoER0XNc

Any idea on that? 

I also had some of these CT's which have the exact same specs as the SCT1013's and they get identical readings

https://stuff.iotawatt.com/product/accuct-100a-x-16mm-split-core/?v=7516fd43adaa

skipfire, 2022-01-14:
The 25K output should be 120v x 104A x 2 legs.  Each leg should is capped at 104A and the breaker reflects that (my 22kw has a double 100A breaker).  The generator output is a pair of partially-independent voltage x amperage and the stated ratings and overall summary stuff loses accuracy as the demands on each leg drift apart, for best accuracy calculations and information would have to be done independently as the text does, but for the gauges where redline is important the single view of the higher value is what I would go with.  105A on one leg is too much even if the other leg is only 5A.

TopBanana123, 2022-01-14:
Let me verify the above numbers, I'm going to dig through my Enphase monitoring and see if I can figure if my meter is incorrect, or the CT's

Its very possible my meter is out of whack, its not a very high dollars meter

TopBanana123, 2022-01-14:
> The 25K output should be 120v x 104A x 2 legs. Each leg should is capped at 104A and the breaker reflects that (my 22kw has a double 100A breaker). The generator output is a pair of [...]

Would it just select the highest leg at that point? Because if my generator was reading 104a, it could be an even 52/52 load which would be just fine and only half capacity, but it could also be a very unbalanced load and be almost all 104a on one leg

It of course really couldn't be that in reality, but you get my point there

TopBanana123, 2022-01-14:
Okay, my meter is correct, which means the numbers in Genmon are incorrect

[image]

skipfire, 2022-01-14:
It then starts to get into user experience questions and "What is the purpose of this gauge?"  For my 22kW Evo2 genmon doesn't have any gauge for current output, just voltage and power so I haven't had it in my face to think about what I would want it to mean.  When I've been troubleshooting (total load and UPSes freaking out) I have used my Sense monitor (I don't recommend it though) to see the individual legs with voltage variations and such.  I don't know if it is possible with the toolsets in use, but a single gauge with 2 hands showing L1 and L2 independently would be my visualization choice for utility voltage and output voltage as well as external current.

TopBanana123, 2022-01-14:
I'm going to go throw a CT over this deathtrap contraption I made and see if I also get 9a through Genmon

jgyates, 2022-01-14:
@skipfire. 

FYI from genmon's perspective it never gets voltage readings on a per leg basis so any time I need a single leg voltage I just divide the total output by 2 (i.e. modbus only gives one output for voltage).

skipfire, 2022-01-14:
@jgyates when you looked at the logs and got the CT2 = 17.88A value, did it have the min/max from the GetCTReading method?  Starting from the readings from the ADC may be informative.  For a 9A load I believe you and I both observationally found it should be in the 40-44 range.  That may help determine if the discrepancy is hardware or software.

@TopBanana123 if you grab one of those CTs, can you post pictures of the CT face with it's details for reference?

jgyates, 2022-01-14:
@skipfire:

The log file captures this:

From channel 1 of ADC
 sample: 509, max: 512, min: 508
  ms elapsed: 80, num samples 419

From channel 2
 sample: 521, max: 594, min: 429
 ms elapsed: 80, num samples 652

After both channels have been probed:
 CT1: 0.65, CT2: 17.88

jgyates, 2022-01-14:
This is 

sample: 521, max: 594, min: 429
ms elapsed: 80, num samples 652

last sample is 521
max 594
min 429
80 millisecond sample time
total number of samples 652

TopBanana123, 2022-01-14:
Here are the CT's I've tried

[image]

[image]

And here is the video of the testing

https://youtu.be/L2oUcmcuX_Y

and (Note meter says 8.88, it almost looks like it never videoed the screen right)

https://youtu.be/HHuDtPk7utc

And here is with the other brand CT which gets the same results (Which are MUCH nicer and have longer cords, as an FYI)

https://youtu.be/UxyC4VxCUJs

TopBanana123, 2022-01-14:
Just submitted logs again if it helps

TopBanana123, 2022-01-14:
> It then starts to get into user experience questions and "What is the purpose of this gauge?" For my 22kW Evo2 genmon doesn't have any gauge for current output, just voltage and power so I haven't [...]

I will heavily use this feature of current, because I do a monthly loaded test where I try and get a good load on it. being able to monitor it right there on the generator is nice

skipfire, 2022-01-14:
Ok, so both of those CTs are current output rather than voltage output CTs which is probably why the values into the ADC are so far off the expected values.  I don't have a current output CT handy to test with (yet, I will be ordering one today).  It may be something where an alternate equation can be used to determine the amps, but I'm really not sure yet.

TopBanana123, 2022-01-14:
Are mine SCT013's the same as the recommended CT? I just went back and re-read the description for the Hat and I see it suggests the SCT-013-000 which it looks like I have

I notice that does then say the voltage output models, but I can't seem to even find those. I should have ordered some with the board. Doh!

TopBanana123, 2022-01-14:
Okay I just ordered some from PintSize, when I get those I'll report back the difference

dennis48755, 2022-01-14:
I misunderstood the need for voltage CT's when I placed my order for the sensor HAT earlier today. But because other items I ordered prompted questions, Jon at PintSize.me was kind enough to contact me for clarification. He was able to add the voltage CT's to my order before it was shipped.

TopBanana123, 2022-01-14:
Mine is already in my generator, so a bit late for that at this point. Hopefully I did no damage to the board with the other CT's

skipfire, 2022-01-14:
Given that your board is reading values that are within a normal expected range I would lean towards it not having damaged the Sensor HAT.  If it did not, then the circuit should be adequately isolated that if any damage did occur it should be limited to the Sensor HAT (most likely just the removable ADC chip).  When the new CTs arrive if it looks like the board was damaged contact me directly.

grsthegreat, 2022-01-20:
I have most current update, but i dont see individual ct readouts for output?

TopBanana123, 2022-01-20:
You have the sensor hat right? And You've followed the CLI instructions and enabled the addon?

grsthegreat, 2022-01-20:
Ive enabled nothing new. I thought the update took care of this. And, have no idea what a sensor hat is.

my ct’s do give reading for total output though, so guess ill stick with that.

skipfire, 2022-01-20:
@grsthegreat the Sensor HAT is another board from PintSize.Me LLC (disclosure: that is my company) that allows for 2 CTs and a temp probe to be connected to the Pi rather than the generator.

skipfire, 2022-01-20:
@TopBanana123 there will be another change to get the right values coming out in addition to needing the correct type of CTs.  I gave jgyates incorrect information on converting the ADC value to amps due to using the wrong clamps myself.  I hope to have the changes up this weekend.

TopBanana123, 2022-01-20:
Thanks for all the help guys

Should be pretty easy to test now I have my space heater test rig

grsthegreat, 2022-01-20:
Ah, i see now. Thanks.

jgyates, 2022-01-22:
@TopBanana123 

I checked in an update that should work well with the new CTs. Just update from the about page. The version number has not changed. Let us know how your testing goes.

TopBanana123, 2022-01-23:
Thanks! I'll update and test it out when I get the new CT's

Very excited to try them out, and happy that they work up to 120a (over 28kw)

TopBanana123, 2022-01-29:
Got the new CT's earlier this week and threw them in today as well as updating to the latest software. Not too sure what's going on though. Apologies for all the videos, I thought it was the easiest way to get the details in

First, it was dead on correct. I was about to close it up and call it a day

https://youtu.be/a2QFKP1YH8U

But then, they would drift off. The total power looked correct, but it would add some current to one, and then remove from the other, it would go between that and the correct readings

https://youtu.be/W4wUqdEtwpk

I turned off the heat gun, but then current was still being shown

https://youtu.be/q9INaLkHFzc

And even with the CT's off completely

https://youtu.be/xjnJt0rgtAs

And even with them completely removed

https://youtu.be/xncglz_FqUk

Then I noticed that it would randomly add 250a, if the CT's were connected or not, and attached to load or not

https://youtu.be/kI34dmm16r0

I thought at this point it could be the board damaged by the other CT's I had used, but even with it completely removed it does it

https://youtu.be/sEm5W5I39Gc

Any ideas?

I have the sensor hat off and the CT's out, I'm hoping I can just test them on a spare Pi, or do they require the Generator Hat too?

skipfire, 2022-01-29:
I am digging into this.  The hat can operate by itself, though I'm not sure if the software will show the data when it isn't attached.

TopBanana123, 2022-01-29:
Okay cool, later on this evening I'll try get it hooked up to a spare Pi, will be much easier to test without opening up the generator 

The fact it was pretty much correct at first makes me optimistic. The phantom amps at idle also doesn't really matter, as it seemed correct when under load

But it drifting off between the legs, and then adding 250a seems odd

jgyates, 2022-01-30:
I will think on the other issues you pointed out.

skipfire, 2022-01-30:
I haven't looked through the videos yet, but being off by 250A is a bit whacked.  I believe it's capturing inrush spikes and noise and presenting that.  I'm currently working on a Python script that will work outside genmon to work through the data issue and then merge updated code in once things are worked out.

TopBanana123, 2022-01-30:
It shows 250a when the CT's have no wire running through them, when the CT's are disconnected, and even what the hat is removed entirely

dennis48755, 2022-01-30:
Bit whacked is right. Mine has been showing 498.74 A steady all afternoon with or without the CT's connected
[image]
.

dennis48755, 2022-01-30:
> and even when the hat is removed entirely

I can confirm, these weird readings appear without the Sensor HAT attached.

skipfire, 2022-01-30:
Without the sensor HAT attached at all I'm not surprised to see it pegged like that.  My guess is that not having it there is defaulting the input value to 0, which would equal ~250A on the bottom of the sine wave.

jgyates, 2022-01-30:
@TopBanana123 

When you removed the HAT and restarted the pi, did you refresh the browser? Then browser loads it requests the gauges and they type of data expected and it only refreshes the expected data after that. If you change the config if genmon you will need to refresh the browsers.

Also, sending your log files form the about screen would be helpful.

TopBanana123, 2022-01-30:
Yes I refreshed the page, I must because I have it served through a reverse proxy (No caching)

I just submitted my logs, but I believe I turned off the CT debug logging we enabled before. Is that what you'd be looking for?

jgyates, 2022-01-30:
I checked in an update that should help the issue when the HAT shows up if the HAT is not present.  Since the SPI bus shows up as long as the changes are in /boot/config.txt, reading the device returns zeros if the HAT is not present. Since the zero current reading is 512 this makes an unusually large reading. The change makes the HAT add on ignore any readings if they are all zero.

One thing to keep in mind is that the can be up to a 20 second delay in the readings of the HAT until the time the web page is updated. 15 seconds for the default "Poll Time" in the CT Hat settings and another 5 seconds for how often the web page refreshes.

skipfire, 2022-01-30:
@TopBanana123 or @dennis48755 are either of you up for running a python script to see if I've got a possible solution?  The script checks the inputs from the CTs, and calculates amps, and then it prints 2 values per second, a "filtered" amps (new value) and a peak amps (current value).  The current code in genmon takes many samples (hundreds) to try and get one as close as possible to the crest or trough of the sine wave but only preserving the single greatest offset, this method includes inrushes and noise (I gave jgyates this code so my bad).  The new code does that 5 times and takes the lowest peak value.  I have a steady 3.7A load that was reading 3.42A to 4.64A, now it is 3.66A with a rare 3.42A or 3.91A, which are only a single step different (there's nothing between 3.42 & 3.66 or 3.66 & 3.91).  Hopefully it at least filters out basic noise, but if there is an inrush current that could be longer than the scan duration, my fridge has an inrush period of about 400ms.  This script is currently stand alone to try and confirm that it is a viable fix, which I would say is it being within 1A per leg, preferably 0.5A.  If you do this on the Pi that has GenMon I believe all the dependencies should be in place to run the script successfully.

dennis48755, 2022-01-30:
I will give it a try tomorrow.

Can you give me instructions how & when to run it, I'm not very Linux literate.

skipfire, 2022-01-30:
If you are on Mac I'm not sure if scp is present or has the same switches or not.  Download the file, open a command prompt and get to the folder you downloaded to.  If you use the default download folder then you will need to
`cd c:\users\{username}\downloads` 
and then to transfer the file
`scp test.txt pi@{Pi name or IP}:~`

To execute the script, ssh into the Pi
`ssh pi@{Pi name or IP}`
enter your password and then to execute the file,
`python3 test.txt`
You can stop the script with Ctrl+C.  It will output the readings once per second until you stop it and it will show data for both clamps.

jgyates, 2022-01-30:
scp is on a mac, to copy just open Terminal and type:

     cd Downlaods
     scp test.txt pi@{Pi name or IP}:~

then enter the password of the pi user on your raspberry pi.

jgyates, 2022-01-30:
I checked in a new version of the add on that incorporates @skipfire's new approach in the test app. Let us know how your testing goes.

[11 more replies]

closed 2022-02-25 by jgyates